espressif/esp-idf #3670: How do I solve /dev/ttyUSB0 disappearance?

giordano-cm, 2019-06-21:
## Problem Description

I have a **ESP32-DevKitC** and when i plug it into any USB, the command `ls /dev/tty*` should return (among others) this one: `/dev/ttyUSB0`. Sometimes, it returns. Some other times, it doesn't, no matter the USB port i plug it in. This problem happens specially after that i touch the cable and the serial connection stops. This is what logs in terminal when it stops:

```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/giordanocm/esp/esp-idf/tools/idf_monitor.py", line 136, in _run_outer
    self.run()
  File "/home/giordanocm/esp/esp-idf/tools/idf_monitor.py", line 229, in run
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/usr/lib/python2.7/dist-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
```
Cable is not the problem. I tried to change it and nothing happened. For some reason, if i wait for some minutes (i don't know how many minutes, that happened once only) and i plug the cable into some USB, `/dev/ttyUSB0` shows up. However, i don't know if it happens all the time (works after waiting for some time).

Neither `lsusb` -t or `lsblk` or `dmesg` commands show something usefull.

Does anyone know what is happening? How can i solve this problem?

PerMalmberg, 2019-06-22:
> This problem happens specially after that i touch the cable and the serial connection stops

Sounds very much like you have a bad connection, probably on the DevKitC side based on what you've written.

giordano-cm, 2019-06-22:
> Sounds very much like you have a bad connection, probably on the DevKitC side based on what you've written.

Yes, probably. But how do i solve this? Like i said, cable isn't the problem. Maybe the USB soldering on the DevKitC side? I don't think so, once i touch the cable closer to the computer than to the DevKitC.

giordano-cm, 2019-06-22:
Well, i guess the cable **extension** is the real problem. I took off the usb cable extenson that i was using and it looks like the problem is solved. I'm gonna make some more tests and let you know.

Alvin1Zhang, 2019-07-11:
@giordano-cm Hi, would you please help share whether your issue still exists? Thanks.

giordano-cm, 2019-07-11:
Sure. Sorry for the late answer.

> Well, i guess the cable extension is the real problem.

Definitely, that was the problem. Once i took off the USB extension cable, the problem stopped. I was trying in both USB ports: 2.0 and 3.0. Both were not responding well, but USB 2.0 port was responding better. I've read that USB 2.0 supports longer distance than USB 3.0. But i'm not really sure. Anyway, i took off the extension cable and both ports started working normally.

closed 2019-07-11 by giordano-cm