espressif/esp-idf #3671: Does ESP32 support seamless Wi-Fi roaming (IEEE 802.11v , r, k) in the STA mode?

flyoo, 2019-06-21:


sagb2015, 2019-06-21:
Currently ESP32 does not support these features. We will keep these requirements in mind, but it is unlikely that they will be available anytime soon. Among the three, we may prioritise 11r though.

MoffKalast, 2020-04-08:
It's been a year, any progress on this front?

sagb2015, 2020-04-08:
We have started working on 11r. However, it is still at preliminary stage.

g8ecj, 2020-06-22:
Is the 11r going to be available to resellers/packagers such as u-blox?

AxelLin, 2020-12-16:
@sagb2015 

I have noticed that 27101f94546bf054ae022fcb9c3acb673de0e270 added initial roaming support.
How about 802.11r support?

Also note, the comment of esp_wifi_set_rssi_threshold() says
This API needs to be called every time after WIFI_EVENT_STA_BSS_RSSI_LOW event is received.

But in the example code, it actually calls it when got WIFI_EVENT_STA_CONNECTED?

kapilkedawat, 2020-12-18:
Hi @AxelLin ,

11r support will be added in future.

'This API needs to be called every time after WIFI_EVENT_STA_BSS_RSSI_LOW event is received.'
This comment is to indicate that only one event will come after calling esp_wifi_set_rssi_threshold(), for getting a second event esp_wifi_set_rssi_threshold() needs to be called again.

sagb2015, 2021-01-25:
Hi @policeman0077, we had to put 11r in the backlog because of other higher priority action items. We will keep you updated when we restart 11r activity. We are targeting 2nd quarter of 2021.

policeman0077, 2021-01-27:
> Hi @policeman0077, we had to put 11r in the backlog because of other higher priority action items. We will keep you updated when we restart 11r activity. We are targeting 2nd quarter of 2021.

thank you :)

AxelLin, 2021-02-18:
Hi @kapilkedawat 

After checking the roaming_example.c code, it looks like esp_bss_rssi_low_handler can be a common implementation to me.
Why not provide a common implementation instead of adding the implementation in application (example) code?
Or do you expect people to implement different behavior in esp_bss_rssi_low_handler?

kapilkedawat, 2021-02-22:
Hi @AxelLin ,

esp_bss_rssi_low_handler can indeed be a common implementation, however it shouldn't be part of supplicant. People may want to do different kind of roaming/disconnection(normal AP switch, 11r based roam, 11kv based roam) based on this trigger.

This will only be added in a common implementation if we decide to have a separate roaming component, in that case it will act as one of the triggers to initiate roaming.

Currently, we are delaying these tasks in favor of other high-priority features.

AxelLin, 2021-06-28:
> Hi @policeman0077, we had to put 11r in the backlog because of other higher priority action items. We will keep you updated when we restart 11r activity. We are targeting 2nd quarter of 2021.

@sagb2015 
It has been a long wait for the 11r.
Hope it will be available by end of this month.

AxelLin, 2021-07-02:
@kapilkedawat 

We are working on adding romaing support for our project.
However, we are still waiting 11r support (have been waited for very long time).
Can you update the 11r support status? (I don't see **any** progress in recent esp-idf update.)

MoffKalast, 2021-07-02:
Excellent work guys, looking forward to seeing it work 👍

AxelLin, 2021-07-05:
> Hi @AxelLin ,
> 
> 11r support will be added in future.
> 
> 'This API needs to be called every time after WIFI_EVENT_STA_BSS_RSSI_LOW event is received.'

Hi @kapilkedawat 

If I have CONFIG_WPA_11KV_SUPPORT=y but didn't call esp_wifi_set_rssi_threshold() at all,
will this cuase some bad sid-effect (compare to CONFIG_WPA_11KV_SUPPORT=n)?
i.e. Is it safe to set CONFIG_WPA_11KV_SUPPORT=y but disable the romaing at by not calling esp_wifi_set_rssi_threshold() in application side?
I want to allow user to enable/disable roaming.

kapilkedawat, 2021-07-05:
Hi @AxelLin , Yes it's safe to enable only CONFIG_WPA_11KV_SUPPORT=y and not calling esp_wifi_set_rssi_threshold().

esp_wifi_set_rssi_threshold() allows users to write an algorithm for station-initiated roaming based on RSSI. Only enabling CONFIG_WPA_11KV_SUPPORT will enable the device to support network-initiated roaming which is sufficient in most cases.

AxelLin, 2021-07-05:
> Only enabling CONFIG_WPA_11KV_SUPPORT will enable the device to support network-initiated roaming which is sufficient in most cases.

I don't quite understand this..
Do you mean only enable CONFIG_WPA_11KV_SUPPORT=y will enable the device to support network-initiated roaming?
It seems nothing happen when I test such case,  what is the condition to trigger roaming in such case (without calling esp_wifi_set_rssi_threshold)?

negativekelvin, 2021-07-05:
> what is the condition to trigger roaming in such case

Triggered from AP side

> 802.11v BSS Transition Management allows an AP to request a station to transition to a specific AP, or suggest a set of preferred AP

AxelLin, 2021-07-05:
> > what is the condition to trigger roaming in such case
> 
> Triggered from AP side
> 
> > 802.11v BSS Transition Management allows an AP to request a station to transition to a specific AP, or suggest a set of preferred AP

Hi @kapilkedawat 
In my test, seems nothing happen if I didn't call esp_wifi_set_rssi_threshold().
However, it works if I call esp_wifi_set_rssi_threshold().

How to debug (just want to make sure roaming is triggered, etc) for the case without calling esp_wifi_set_rssi_threshold()?

@kapilkedawat  Does it work for you with the wifi roaming example code if CONFIG_EXAMPLE_WIFI_RSSI_THRESHOLD=0?

kapilkedawat, 2021-07-05:
Hi @AxelLin,

'In my test, seems nothing happen if I didn't call esp_wifi_set_rssi_threshold().'
Please make sure that your AP network support client steering(most of the mesh home routers does support it, you need to check in the router specification).

When 11KV is enabled, client essentially tells the AP that it support 11k and 11v. AP may use those features to steer client to a suitable AP in the network(how AP decides whether it needs to steer the client and on which AP will entirely depend on access point and client has no control over it).

We have verified it with hostapd as well as TP link Deco routers and some other mesh wifi systems and it works fine.

Enabling CONFIG_WPA_SCAN_CACHE adds the support of table scan capability. AP may request scan table using 11K and therefore the support was added. Max memory impact due to this is around 10KB heap memory(scan table cache size * assuming max beacon/probe size of 1KB). It normally has no impact on roaming since we have seen generally AP request for active scan during the steering but as I explained earlier its entirely depends on AP.

AxelLin, 2021-07-06:
Hi @kapilkedawat 

I don't quite understand the logic in below code block:
https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c#L286-#L298

My questions:
1. So it doesn't need to set any setting in params for esp_wifi_scan_start?
   e.g. WIFI_ALL_CHANNEL_SCAN / WIFI_CONNECT_AP_BY_SIGNAL / pmf_cfg setting are not required?
2. it calls esp_wifi_scan_get_ap_records but actually never use the ap_records at all
   What is the purpose of such code?

BTW, some of my device hitting OOM while testing.
My esp_alloc_failed_hook_t callback shows the size=1696, caps=0x1800.

AxelLin, 2021-07-06:
Hi @kapilkedawat 

I think there are some problem with CONFIG_WPA_11KV_SUPPORT=y.
I didn't call net_utils_set_wifi_roam_trigger() in my test devices.
So the configuration is simple, my application just enable CONFIG_WPA_11KV_SUPPORT=y
and set below flags in wifi_config.
                        .rm_enabled =1,
                        .btm_enabled =1,

I test with TP link Deco (mesh with 4 deco).
After running for a few hours, several esp32 devices hit OOM.
So I think there is memory leak somewhere in the roaming code.

The symptom is the same, the request size=1696, caps=0x1800.
I observe the symptom on both v4.3 and master branch.

kapilkedawat, 2021-07-06:
Hi @AxelLin,

Can you please provide capture logs and wifi + supplicant logs? We will take a look.

AxelLin, 2021-07-06:
Hi @kapilkedawat 
I cannot capture the log at the moment due to work from home.
(I just update my test devices in office but it's difficult to debug remotely).
I think it's easy to reproduce since there is basically not application code involved.

AxelLin, 2021-07-07:
> Hi @kapilkedawat
>
> I don't quite understand the logic in below code block:
> https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c#L286-#L298
>
> My questions:
>
> 1. [...]
Hi @kapilkedawat 
I think the reason to call esp_wifi_scan_get_ap_records here is to clean scan_list from net80211.
But I'm wondering if pass number=1 is correct when esp_wifi_scan_get_ap_num(&number) > 1? (possible leak?)
The implementation is in a binary library so I cannot check it.
In additional, please check the memory leak possibility in esp_wifi_scan_get_ap_records() error paths.
It's not clear to me if the caller needs to do additional action to clean scan_list if esp_wifi_scan_get_ap_records fails.

AxelLin, 2021-07-07:
Hi @kapilkedawat 

In my device's debug log I found when the device run into OOM:
I didn't find any log about roaming happen on the devie.
But the device has got WIFI_REASON_BEACON_TIMEOUT then STA disconnected and then reconnect for 12 times.

Note, if I re-compile the firmware with CONFIG_WPA_11KV_SUPPORT=n, the issue never happen.  (Just a hint)

AxelLin, 2021-07-08:
> 
> 'This API needs to be called every time after WIFI_EVENT_STA_BSS_RSSI_LOW event is received.'
> This comment is to indicate that only one event will come after calling esp_wifi_set_rssi_threshold(), for getting a second event esp_wifi_set_rssi_threshold() needs to be called again.

Hi @kapilkedawat 

In my test, the device will receive WIFI_EVENT_STA_BSS_RSSI_LOW event only 1 time after the device connected to AP.

In roaming_example.c, it calls esp_wifi_set_rssi_threshold() when got WIFI_EVENT_STA_CONNECTED.
But this seems problematic because roaming may fail for some reason.
When means if the device does not roam to a new AP, it will not receive WIFI_EVENT_STA_CONNECTED.
(i.e. the esp_wifi_set_rssi_threshold() is not called.)
After that the device no longer receive WIFI_EVENT_STA_BSS_RSSI_LOW event anymore.

However, if I call esp_wifi_set_rssi_threshold() after got WIFI_EVENT_STA_BSS_RSSI_LOW event.
It may receive too many WIFI_EVENT_STA_BSS_RSSI_LOW event. 
I'm not sure if it's correct to do so because it will call esp_rrm_send_neighbor_rep_request() many times.

negativekelvin, 2021-07-08:
Example may be too simplistic, but roaming algorithm can be as complex as you want, implement timers and dynamic rssi thresholds etc...

AxelLin, 2021-07-09:
> Example may be too simplistic, but roaming algorithm can be as complex as you want, implement timers and dynamic rssi thresholds etc...

The roaming_example.c is the only reference code for wifi roaming in esp-idf.
It's important to make it *correct*.
What I pointed out in my previous reply is not a special case.
It's very common use case that people running the example code could hit and
notice roaming no longer happen any more.

What I want to do is just with fixed rssi threshold so actually I want the code
as simple as possible. I even mentioned that such simple case should be implemented
in esp-idf as common code (https://github.com/espressif/esp-idf/issues/3671#issuecomment-781119043).
And it's easier to test and verify wifi roaming with common code.

AxelLin, 2021-07-30:
> > 'This API needs to be called every time after WIFI_EVENT_STA_BSS_RSSI_LOW event is received.'
> > This comment is to indicate that only one event will come after calling esp_wifi_set_rssi_threshold(), for getting a second event esp_wifi_set_rssi_threshold() needs to be called again.
> 
> Hi @kapilkedawat
> 
> In my test, the device will receive WIFI_EVENT_STA_BSS_RSSI_LOW event only 1 time after the device connected to AP.
> 
> In roaming_example.c, it calls esp_wifi_set_rssi_threshold() when got WIFI_EVENT_STA_CONNECTED.
> But this seems problematic because roaming may fail for some reason.
> Which means if the device does not roam to a new AP, it will not receive WIFI_EVENT_STA_CONNECTED.
> (i.e. the esp_wifi_set_rssi_threshold() is not called.)
> After that the device no longer receive WIFI_EVENT_STA_BSS_RSSI_LOW event anymore.
> 
> However, if I call esp_wifi_set_rssi_threshold() after got WIFI_EVENT_STA_BSS_RSSI_LOW event.
> It may receive too many WIFI_EVENT_STA_BSS_RSSI_LOW event.
> I'm not sure if it's correct to do so because it will call esp_rrm_send_neighbor_rep_request() many times.

@kapilkedawat @sagb2015 
Can you reply my above comments?
1. I don't think calling esp_wifi_set_rssi_threshold() when got WIFI_EVENT_STA_CONNECTED. is correct.
2. I doubt if it's fine to call esp_wifi_set_rssi_threshold() after got WIFI_EVENT_STA_BSS_RSSI_LOW event.
   (Because it may receive too many WIFI_EVENT_STA_BSS_RSSI_LOW event in short period).

BTW, I'd appreciate if you can make the roaming_example.c correct since it is the only reference code for wifi roaming.

AxelLin, 2021-08-04:
@sagb2015 

I'm asking very basic thing:
The esp_wifi_set_rssi_threshold() comment says: (https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi.h#L1156)
This API needs to be called every time after WIFI_EVENT_STA_BSS_RSSI_LOW event is received
**Why the example code does not follow the document says above?**
(If it's for demonstrative purpose it should follow what the document says, right?)
How do you expect developers to use the esp_wifi_set_rssi_threshold() api **correctly** if your example code does not follow the document says?

As I pointed out the example code is buggy if wifi connect fails, that needs fix for no excuse. (People read the example code to learn using the APIs).

AxelLin, 2021-08-04:
> @AxelLin Regarding [#3671 (comment)](https://github.com/espressif/esp-idf/issues/3671#issuecomment-781119043), we intend to refrain from making policy decisions in SDK to keep it flexible for different scenarios.

Actually, you can have a common implementation for simple use case as fixed RSSI threshold.
If the developer needs a more complex implementation, it is fine to write a customized implementation instead.
In additional, a basic common implementation will be a good reference code.

kapilkedawat, 2021-08-04:
> @sagb2015
>
> I'm asking very basic thing:
> The esp_wifi_set_rssi_threshold() comment says: (https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_wifi.h#L1156)
> This API needs [...]

This means user will not get another WIFI_EVENT_STA_BSS_RSSI_LOW event once they get one event after calling esp_wifi_set_rssi_threshold() , for getting another event of RSSI breach, they needs to call esp_wifi_set_rssi_threshold()  API again(As you correctly deduced otherwise it will flood the event).

In case the document is not clear enough, we will try to make it clearer in future.

> (If it's for demonstrative purpose it should follow what the document says, right?)
> How do you expect developers to use the esp_wifi_set_rssi_threshold() api **correctly** if your example code does [...]

Every developer may have different approaches when roaming fails such as
1. Lower the RSSI threshold by some offset.
2. Retry the roaming after some time(start a timer)
3. Stay with the same AP.
4. Put current AP in blacklist and retry with other candidates.

All the approaches are correct, it will depend on the user how they want their device to behave incase of a failure.
We are using 3rd approach, however user is free to implement any approaches they seems fit for their usecase.

AxelLin, 2021-08-04:
> Every developer may have different approaches when roaming fails such as
>
> 1. Lower the RSSI threshold by some offset.
> 2. Retry the roaming after some time(start a timer)
> 3. Stay with the same AP.
> 4. [...]

So you stay with the same AP and due to the fact of missing esp_wifi_set_rssi_threshold() call, the WIFI_EVENT_STA_BSS_RSSI_LOW will **no longer trigger**.
And you call it is **correct**?

AxelLin, 2021-08-09:
> Hi @AxelLin,
> 
> When 11KV is enabled, client essentially tells the AP that it support 11k and 11v. AP may use those features to steer client to a suitable AP in the network(how AP decides whether it needs to steer the client and on which AP will entirely depend on access point and client has no control over it).
> 
@kapilkedawat 

What if setting rm_enabled = 0 && btm_enabled = 0 with CONFIG_WPA_11KV_SUPPORT=y?
Does that mean disallow trigger from AP side?

I ask this because the CONFIG_WPA_11KV_SUPPORT=y is compile time determinated.
I'm wondering if setting rm_enabled = 0 && btm_enabled = 0 allows to disable roaming at runtime.

kapilkedawat, 2021-08-10:
> @kapilkedawat
>
> What if setting rm_enabled = 0 && btm_enabled = 0 with CONFIG_WPA_11KV_SUPPORT=y?
> Does that mean disallow trigger from AP side?

Station has no control whatsoever on AP trigger, how and when AP decides to roam a station is completely in AP's control. AP can even try to roam a station using legacy methods.

> 
> I ask this because the CONFIG_WPA_11KV_SUPPORT=y is compile time determinated.
> I'm wondering if setting rm_enabled = 0 && btm_enabled = 0 allows to disable roaming at runtime.

rm_enabled = 0 && btm_enabled = 0  disables station's rm and wnm capabilities and the corresponding caps won't show during association. Also station won't honour those specific packets.

AxelLin, 2021-10-21:
> Hi @policeman0077, we had to put 11r in the backlog because of other higher priority action items. We will keep you updated when we restart 11r activity. We are targeting 2nd quarter of 2021.

@sagb2015 

(I don't see any progress regarding 11r)
Will esp-idf v4.4 release include 11r support?

AxelLin, 2021-12-17:
@sagb2015 @kapilkedawat 

Could you help to explain a bit about below roaming example code?
I don't understand why such check can tell if the request was initiated by us.
https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c#L276-L279

It seems make more sense if the checking is "**if (val != &rrm_ctx) {**" instead. (By compare the address rather than the dereference value)

kapilkedawat, 2021-12-21:
@AxelLin Idea was here to send some magic number and compare them. However, yes you can use pointer validation or use to context to pass some data to handler if needed. 

How we want to use it completely depend on the user implementation.

AxelLin, 2021-12-22:
> @AxelLin Idea was here to send some magic number and compare them. However, yes you can use pointer validation or use 

That example code looks wrong to me.
Even with val != &rrm_ctx (different addresses), in theory it still has chance *val == rrm_ctx.

BTW, below comment says "send AP btm query, this will cause STA to roam as well"
https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c#L301-L302

But when I test it, it does not cause STA to roam.
i.e. Not every esp_wnm_send_bss_transition_mgmt_query call will cause STA to roam.
There must be some internal condition to decide if STA should roam.
Can you share the details?
I'm wondering how to know if STA roams to new AP because low RSSI rather than requested by AP.

kapilkedawat, 2021-12-22:
> That example code looks wrong to me. Even with val != &rrm_ctx (different addresses), in theory it still has chance *val == rrm_ctx.

As I said, we need to replace value to a cookie value. Example is just how we can do it.

> BTW, below comment says "send AP btm query, this will cause STA to roam as well" https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c#L301-L302
It's a network assisted roaming example, Station will roam given that it gets a valid BTM request in response of BTM query. AP needs to provide next candidate in the BTM request for station to roam. If AP doesn't provide next candidate, station will assume it is still connected to best candidate.

AxelLin, 2021-12-22:
> > BTW, below comment says "send AP btm query, this will cause STA to roam as well" https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c#L301-L302

> It's a network assisted roaming example, Station will roam given that it gets a valid BTM request in response of BTM query. AP needs to provide next candidate in the BTM request for station to roam. If AP doesn't provide next candidate, station will assume it is still connected to best candidate.

Figure out the reason why my device does not roam when rssi is low:

In compare_scan_neighbor_results, the wpa_s->wnm_num_neighbor_report is always 1
and the target bssid is always my current connected AP.
As you see in https://github.com/espressif/esp-idf/issues/3671#issuecomment-999218139 ,
there are multiple APs around my STA. so I have no idea why compare_scan_neighbor_results always
return my current AP only.

Do you have any idea what could be wrong?

Note:
If I move the device far away from the AP, in the end I got disconnect from AP (not roaming)
and then it joins a different AP when reconnect.

kapilkedawat, 2021-12-22:
Please share the sniffer capture and app_test code logic to debug this faster.

One thing you can check by yourself whether AP is providing any candidate besides itself in the BTM req. If it's not, station will not move.

AxelLin, 2021-12-26:
> Please share the sniffer capture and app_test code logic to debug this faster.
> 
> One thing you can check by yourself whether AP is providing any candidate besides itself in the BTM req. If it's not, station will not move.

well, I got it working now.
Just in some case (maybe environment related) the AP keeps report itself as best candidate, but this is probably not STA (esp32)'s issue.

AxelLin, 2021-12-26:
@kapilkedawat 

After STA (esp32) joined an AP, is there any API to know if the AP has 802.11kv support?
I'm wondering if it is possible to avoid repeating below errors when AP does not support 802.11kv.
https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c#L318-L324

BTW, Having API to tell if AP has 802.11kv support is helpful for debug issues.
(Especially when the customer who report issues does not know if the AP used has 802.11kv capability or not).

3DES, 2021-12-30:
@AxelLin: it would be great if you could share your solution! There are so many people out there (me included) that have the same problem.

AxelLin, 2022-02-13:
> @AxelLin: it would be great if you could share your solution! There are so many people out there (me included) that have the same problem.

I'm not allowed to post my code since I don't work for espressif.
If you have any issue you can ask it on esp32 forum or report issues here.

jack0c, 2022-05-17:
Just update 11r support is merged to master, you can get it since 805b5c9115cb4785e45901290c9e680a88485ae9

AxelLin, 2022-05-25:
> Just update 11r support is merged to master, you can get it since [805b5c9](https://github.com/espressif/esp-idf/commit/805b5c9115cb4785e45901290c9e680a88485ae9)

A question about testing 802.11r: What's the user visible changes with 802.11r enabled v.s. disabled?

I got below messages with 802.11r enabled, is this normal behavior? (The message/behavior seems the same as 802.11r disabled)

I (125300) networking: RMM neigbor report bssid=ac:84:c6:32:1a:27 info=0x1897 op_class=0 chan=44 phy_type=9
I (125310) networking: RMM neigbor report bssid=ac:84:c6:32:35:b6 info=0x1897 op_class=0 chan=44 phy_type=9
I (125320) networking: RMM neigbor report bssid=ac:84:c6:32:14:78 info=0x1897 op_class=0 chan=44 phy_type=9
I (125331) networking: RMM neigbor report bssid=d8:47:32:7c:34:bb info=0x1897 op_class=0 chan=44 phy_type=14
I (125341) networking: RMM neigbor report bssid=d8:47:32:7c:37:af info=0x97 op_class=0 chan=44 phy_type=14
I (125351) networking: RMM neigbor report bssid=ac:84:c6:32:1a:26 info=0x1887 op_class=0 chan=7 phy_type=7
I (125362) networking: RMM neigbor report bssid=ac:84:c6:32:35:b5 info=0x1887 op_class=0 chan=7 phy_type=7
I (125372) networking: RMM neigbor report bssid=ac:84:c6:32:14:77 info=0x1887 op_class=0 chan=7 phy_type=7
I (125382) networking: RMM neigbor report bssid=d8:47:32:7c:34:ba info=0x1887 op_class=0 chan=7 phy_type=14
I (125393) networking: Low RSSI, Send BTM query cand_list=0
I (135396) wifi:state: run -> init (cc0)
I (135397) wifi:pm stop, total sleep time: 63717216 us / 130269526 us

W (135397) wifi:<ba-del>idx
W (135398) wifi:<ba-del>idx
I (135401) wifi:new:<7,0>, old:<7,1>, ap:<255,255>, sta:<7,1>, prof:1
I (135411) networking: Disconnect reason: 207
I (137820) wifi:new:<7,1>, old:<7,0>, ap:<255,255>, sta:<7,1>, prof:1
I (138323) wifi:state: init -> auth (b0)
I (138328) wifi:state: auth -> assoc (0)
I (138338) wifi:state: assoc -> run (30)
I (138382) wifi:connected with IGS-MESH, aid = 8, channel 7, 40U, bssid = ac:84:c6:32:1a:26
I (138382) wifi:security: WPA2-PSK, phy: bgn, rssi: -14
I (138385) wifi:pm start, type: 1

chegewara, 2023-02-22:
Hi,
can this be considered as working feature?
What i mean, its working to me more or less, esp32 is roaming between APs in wifi mesh:
```
03:33:51.437 > 36:de:4b:68:3a:ee => rssi: -83
03:33:51.559 > station got disconnected reason=207
03:33:51.559 > station roaming, do nothing
03:33:52.187 > setting rssi threshold as -70
03:33:52.438 > 36:de:4b:68:1c:72 => rssi: -58
03:33:53.201 > got ip: 192.168.68.112
03:33:53.201 > RRM supported
03:33:53.201 > BTM supported
03:33:53.440 > 36:de:4b:68:1c:72 => rssi: -52
```

Thanks

kapilkedawat, 2023-02-22:
@chegewara do you see any issues?

chegewara, 2023-02-22:
> @chegewara do you see any issues?

Im not sure, as it was a quick test so far. I dont know how fast this event should be triggered when threshold is detected:
`WIFI_EVENT_STA_BSS_RSSI_LOW`
It also takes about 2 seconds to get IP from new AP, but its probably normal.
Other than that it seems to be working to me. Of course i did not test it fully, just basic functionality.

chegewara, 2023-03-13:
Hi,
is there any reason for this parameter not being `REASON_RSSI`?
https://github.com/espressif/esp-idf/blob/master/examples/wifi/roaming/main/roaming_example.c#L314

Thanks

kapilkedawat, 2023-03-13:
There is no specific reason, you can use REASON_RSSI as well.

chegewara, 2023-03-20:
Hi,
is it expected that esp32 can detect all wifi APs withing mesh in 2.4 and 5GHz, which prevents it to roam?
```
01:03:40.913 > [ 55040][I][WiFi_Task.cpp:199] neighbor_report_recv_cb(): [] rrm: neighbor report len=70
01:03:40.913 > [ 55041][I][WiFi_Task.cpp:155] get_btm_neighbor_list(): [] RMM neigbor report bssid=30:de:xx:xx:1c:72 info=0x2887 op_class=0 chan=1 phy_type=7
01:03:40.913 > [ 55050][I][WiFi_Task.cpp:155] get_btm_neighbor_list(): [] RMM neigbor report bssid=36:de:xx:xx:1c:73 info=0x1897 op_class=0 chan=44 phy_type=9
01:03:40.913 > [ 55062][I][WiFi_Task.cpp:155] get_btm_neighbor_list(): [] RMM neigbor report bssid=36:de:xx:xx:3a:ef info=0x1897 op_class=0 chan=44 phy_type=9
01:03:42.992 > wifi_event_handler:bss rssi is=-74
01:03:42.992 > 
01:03:42.992 > 
01:03:43.024 > [ 57190][I][WiFi_Task.cpp:199] neighbor_report_recv_cb(): [] rrm: neighbor report len=70
01:03:43.077 > [ 57191][I][WiFi_Task.cpp:155] get_btm_neighbor_list(): [] RMM neigbor report bssid=30:de:xx:xx:1c:72 info=0x2887 op_class=0 chan=1 phy_type=7
01:03:43.077 > [ 57200][I][WiFi_Task.cpp:155] get_btm_neighbor_list(): [] RMM neigbor report bssid=36:de:xx:xx:1c:73 info=0x1897 op_class=0 chan=44 phy_type=9
01:03:43.077 > [ 57212][I][WiFi_Task.cpp:155] get_btm_neighbor_list(): [] RMM neigbor report bssid=36:de:xx:xx:3a:ef info=0x1897 op_class=0 chan=44 phy_type=9
01:03:45.045 > wifi_event_handler:bss rssi is=-74
```
When i enable only 2.4Ghz on that wifi mesh then esp32 can easy roam.

Thanks

jack0c, 2023-04-12:
@chegewara ESP32 do not support 5GHz, can't scan 5GHz Wi-Fi. @kapilkedawat is MBO necessary in this case?

chegewara, 2023-04-12:
Hi @jack0c 
I know that esp32 does not support 5GHz, but this is what i have observed with deco mesh:
- when i enable only 2.4GHz on deco mesh, then roaming works perfectly fine
- when i enable 2.4 and 5GHz, then esp32 no longer roam, and as you can see from above logs, some APs with roaming found by esp32 are reporting channels 44, which are 5GHz channels (which does not mean they are reported on 5Ghz, its just means that base is reporting them as roaming capable in that mesh)
```
RMM neigbor report bssid=30:de:xx:xx:1c:72 info=0x2887 op_class=0 chan=1 phy_type=7
RMM neigbor report bssid=36:de:xx:xx:1c:73 info=0x1897 op_class=0 chan=44 phy_type=9
RMM neigbor report bssid=36:de:xx:xx:3a:ef info=0x1897 op_class=0 chan=44 phy_type=9
```

I have no idea what is preventing esp32 to roam, maybe its somehow related to other issues reported when esp32 cant connect wifi when 2.4 and 5GHz SSIDs are the same (i tried this option and got no issues here).
For now i only can say that 2.4GHz alone works to me and 2.4 + 5GHz combined on AP prevents esp32 to roam.

Thanks

chegewara, 2023-04-12:
@kapilkedawat I believe ive been trying it by myself already and i think it didnt help.

Thanks

kapilkedawat, 2023-04-12:
If possible, can you share the sniffer capture so that we can take a look?

chegewara, 2023-04-15:
I have many different tools, but ididnt learn to use wireshark with external packets sniffing yet, sorry.