espressif/esp-idf #3673: How to close the ESP_log and printf

Herwey, 2019-06-21:
Hi :
Now, I want to close all ESP_log/printf/sprintf and etc, I don't need the print info any more. I want to use "#if", but there are so many print functions in my "main.c". I think maybe I can change the "ESP_Log" and "printf" functions, but I can't find they in esp_log.c
Thanks!

InfiniteYuan1, 2019-06-21:
Yeah, Please choose No Output item by make menuconfig command to close ESP_LOG* .

Herwey, 2019-06-21:
@InfiniteYuan1 Hi ， I can't find the  No Output item, I search it in sdkconfig. as below:
# Log output
CONFIG_LOG_DEFAULT_LEVEL_NONE=
CONFIG_LOG_DEFAULT_LEVEL_ERROR=
CONFIG_LOG_DEFAULT_LEVEL_WARN=
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=
CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=
CONFIG_LOG_DEFAULT_LEVEL=3
CONFIG_LOG_COLORS=y

Which one should I select ?

Herwey, 2019-06-21:
@InfiniteYuan1 Hi, I select the "No output", but it still print the ESP_LOG*

kifantidis, 2019-06-22:
Good morning there!

I believe that this post will solve your problems.    [ESP32 How to turn off automatic log of system](https://www.esp32.com/viewtopic.php?t=4249)  

Regards, Kostas.

Alvin1Zhang, 2019-07-10:
Hi, @Herwey Thanks for reporting the issue, would you please help check whether your issue has bene resolved by the post, [ESP32 How to turn off automatic log of system output](https://www.esp32.com/viewtopic.php?t=4249)? Thanks.

Herwey, 2019-07-17:
I find a nice method. The Log was printed from PIN 1 and 3(UART), so when I changed the 1 and 3 for IIC, so that the uart not worked any more, no log.       :D

closed 2019-07-17 by Herwey