espressif/esp-idf #3674: Where can I find the printf function? I want to edit the "printf" function

Herwey, 2019-06-21:
Hi All:
Where can I find the printf function? I want to edit the "printf" function  :D
Thanks !

Dazza0, 2019-06-21:
@Herwey 
**Please do not ignore the GitHub issue posting guidelines.**
> If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

If you have a "how to" question and not a suspected/confirmed bug, please post it to the [forum ](https://esp32.com/)instead.

With regards to `printf()`,  this is a C standard library function which is implemented as a ROM function (function address provided [here](https://github.com/espressif/esp-idf/blob/58df1d93bc17c74499d58e05390af9c309192a5c/components/esp_rom/esp32/ld/esp32.rom.newlib-nano.ld#L38)). To change it would mean removing the symbol declaration in the linker file and implementing your own version.

A better to question is ask is why you need to change the `printf()` funciton? What are you hoping to achieve?

negativekelvin, 2019-06-21:
Maybe this helps you
https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/log.html?highlight=redirection#_CPPv419esp_log_set_vprintf14vprintf_like_t

Alvin1Zhang, 2019-06-21:
@Herwey Thanks for reporting the issue. If you have further such questions, please help post on our [forum](https://esp32.com), thanks.

closed 2019-06-21 by Alvin1Zhang

Herwey, 2019-06-24:
@negativekelvin Hi,Thanks . I found as below:
1. Disabling ROM bootloader output: pull down GPIO15.
 2. Disabling 2nd stage bootloader output: make menuconfig, go to "Bootloader config" and set "Bootloader log verbosity" to "No output"
 3. Application: make menuconfig, go to "Component config", "Log output", set "Default log verbosity" to "No output".
The first step, how to do. That means connect GPIO15 with GND ? Right?