espressif/esp-idf #3675: advanced_https_ota - mutual auth issue (IDFGH-1397)

kifantidis, 2019-06-21:
**Environment**

-Module or chip used: ESP-WROOM-32D
-IDF version: ESP-IDF v4.0-dev-837-g58df1d93b-dirty
-Build System: Make
-Toolchain version: esp32-2019r1
-Compiler version: 8.2.0
-Operating System: Windows 10
-Power Supply: USB
-example used: advanced_https_ota_example

**Problem Description**

I'm trying to update OTA via mutual authentication but I get a fatal alert message (40) after receiving handshake message. I've talked with the tech that has made the server for my tests and after reading the logs it seems that ESP32 completes the first handshake (device verifies the server's crt) but the device never starts the re-negotiation procedure in order for the server to verify the clients crt and key. And as it seems the connection between the server and the device never succeeds. 

I have tested the server with wget program and I've downloaded successfully the OTA.bin file. Also I've tested many things with wget in order to ensure that the server is working. Also server is working fine with firefox browser ++ mutual auth. 

In the advanced_https_ota_example I've added the srvs.crt, cli_.crt and cli_.key as constant char arrays and added them also in the esp_http_client_config_t config structure in order to enable mutual authentication:

```
esp_err_t ota_finish_err = ESP_OK;
     esp_http_client_config_t config = {
        .url = "https://www.xxxxxxxx.com/updates/pol.bin",
        .cert_pem = server_crt,
        .client_cert_pem= cli_crt,
        .client_key_pem= cli_key
    };
    esp_https_ota_config_t ota_config = {
        .http_config = &config,
    };

    esp_https_ota_handle_t https_ota_handle = NULL;
    esp_err_t err = esp_https_ota_begin(&ota_config, &https_ota_handle);
    if (err != ESP_OK) {
        ESP_LOGE(TAG, "ESP HTTPS OTA Begin failed");
        vTaskDelete(NULL);
    }

```

But as far I had no success in making this example to work.

I'm posting also a small log file from the above. If needed I can provide a full (verbose mbedtls) log. 

**Terminal Debugging Log**
```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6504
load:0x40078000,len:11628
load:0x40080400,len:6684
entry 0x4008076c
I (28) boot: ESP-IDF v4.0-dev-837-g58df1d93b-dirty 2nd stage bootloader
I (28) boot: compile time 16:05:05
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 0000a000 00004000
I (65) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (73) boot:  2 phy_init         RF data          01 01 00010000 00001000
I (80) boot:  3 factory          factory app      00 00 00020000 00100000
I (88) boot:  4 ota_0            OTA app          00 10 00120000 00100000
I (95) boot:  5 ota_1            OTA app          00 11 00220000 00100000
I (103) boot: End of partition table
I (107) boot: Defaulting to factory image
I (112) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x20a0c (133644) map   
I (168) esp_image: segment 1: paddr=0x00040a34 vaddr=0x3ffb0000 size=0x02f90 ( 12176) load  
I (173) esp_image: segment 2: paddr=0x000439cc vaddr=0x40080000 size=0x00400 (  1024) load  
0x40080000: _WindowOverflow4 at C:/esp/esp-idf/components/freertos/xtensa_vectors.S:1778    

I (175) esp_image: segment 3: paddr=0x00043dd4 vaddr=0x40080400 size=0x0c23c ( 49724) load  
I (204) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018 size=0x9163c (595516) map   
0x400d0018: _stext at ??:?

I (415) esp_image: segment 5: paddr=0x000e165c vaddr=0x4008c63c size=0x03a90 ( 14992) load  
0x4008c63c: xTaskCreatePinnedToCore at C:/esp/esp-idf/components/freertos/tasks.c:816       

I (432) boot: Loaded app from partition at offset 0x20000
I (432) boot: Disabling RNG early entropy source...
I (432) cpu_start: Pro cpu up.
I (436) cpu_start: Application information:
I (441) cpu_start: Project name:     advanced_https_ota
I (447) cpu_start: App version:      v4.0-dev-837-g58df1d93b-dirty
I (454) cpu_start: Compile time:     Jun 21 2019 16:04:39
I (460) cpu_start: ELF file SHA256:  5d221cac2a03c351...
I (466) cpu_start: ESP-IDF:          v4.0-dev-837-g58df1d93b-dirty
I (472) cpu_start: Starting app cpu, entry point is 0x40081088
0x40081088: call_start_cpu1 at C:/esp/esp-idf/components/esp32/cpu_start.c:265

I (0) cpu_start: App cpu up.
I (483) heap_init: Initializing. RAM available for dynamic allocation:
I (490) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (496) heap_init: At 3FFB9698 len 00026968 (154 KiB): DRAM
I (502) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (508) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (515) heap_init: At 400900CC len 0000FF34 (63 KiB): IRAM
I (521) cpu_start: Pro cpu start user code
I (539) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (600) wifi: wifi driver task: 3ffc0f70, prio:23, stack:3584, core=0
I (600) system_api: Base MAC address is not set, read default base MAC address from BLK0 of 
EFUSE
I (600) system_api: Base MAC address is not set, read default base MAC address from BLK0 of 
EFUSE
I (620) wifi: wifi firmware version: ec61a20
I (620) wifi: config NVS flash: enabled
I (620) wifi: config nano formating: disabled
I (620) wifi: Init dynamic tx buffer num: 32
I (630) wifi: Init data frame dynamic rx buffer num: 32
I (630) wifi: Init management frame dynamic rx buffer num: 32
I (640) wifi: Init management short buffer num: 32
I (640) wifi: Init static rx buffer size: 1600
I (650) wifi: Init static rx buffer num: 10
I (650) wifi: Init dynamic rx buffer num: 32
I (660) example_connect: Connecting to XXXXXXXXXXX
I (740) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (740) wifi: mode : sta (24:0a:c4:15:7b:fc)
I (1960) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (2610) wifi: state: init -> auth (b0)
I (2610) wifi: state: auth -> assoc (0)
I (2620) wifi: state: assoc -> run (10)
I (2630) wifi: connected with XXXXXXXXXXX, channel 11, bssid = XXXXXXXXXXXX
I (2630) wifi: pm start, type: 1

I (3590) tcpip_adapter: sta ip: 10.0.0.163, mask: 255.255.255.0, gw: 10.0.0.2
I (4590) example_connect: Connected to XXXXXXXXXXXX
I (4590) example_connect: IPv4 address: 10.0.0.163
I (4590) example_connect: IPv6 address: fe80:0000:0000:0000:260a:c4ff:fe15:7bfc
I (4600) advanced_https_ota_example: Starting Advanced OTA example
W (5840) mbedtls: ssl_tls.c:8294 received handshake message

W (5940) mbedtls: ssl_tls.c:5172 is a fatal alert message (msg 40)

W (5950) mbedtls: ssl_tls.c:4363 mbedtls_ssl_handle_message_type() returned -30592 (-0x7780)

W (5950) mbedtls: ssl_tls.c:8272 mbedtls_ssl_read_record() returned -30592 (-0x7780)        

E (5960) esp-tls: read error :-30592:
E (5960) TRANS_SSL: esp_tls_conn_read error, errno=Success
I (5970) esp_https_ota: Starting OTA...
I (5970) esp_https_ota: Writing to partition subtype 16 at offset 0x120000
E (5980) advanced_https_ota_example: esp_https_ota_read_img_desc failed
E (5990) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed...
```

kifantidis, 2019-06-23:
Dear @mahavirj,
I've tried to achieve mutual auth OTA with both versions of ESP-IDF (v4 - master and v3.2 ) but with no success. Both codes seems to have ended the same mbedtls debug message. And as I mention above from the servers side, logs show that re-negotiation is never achieved because the device (ESP32) seems that never does that. I don't know how to debug this in order to find where the problem may be located. Any ideas on this would be nice. Today I installed the toolchain with compiler 5.2 and v3.2 esp-idf in a new machine and I've tested the code in ESP-WROVER-KIT v3. I've tested hello-world example to test the ESP32's and compiler's functionality. That was a success and after this I've changed my esp_http_client .c/.h files as AndersKaloer recommends [LINK](https://github.com/espressif/esp-idf/commit/d81c79d79ff3af59c26fc1bb091b6ef9cce61b48)

I'm posting below the mbedTLS logs: 

```
I (3589) wifi: state: auth -> assoc (0)
I (3609) wifi: state: assoc -> run (10)
I (3709) wifi: connected with XXXXXXXXX, channel 1
I (3709) wifi: pm start, type: 1

I (7599) event: sta ip: 192.168.1.9, mask: 255.255.255.0, gw: 192.168.1.1
I (7599) simple_ota_example: Connect to Wifi ! Start to Connect to Server....
I (7899) mbedtls: ssl_tls.c:8021 => handshake

I (7899) mbedtls: ssl_cli.c:3405 client state: 0

I (7899) mbedtls: ssl_tls.c:2751 => flush output

I (7899) mbedtls: ssl_tls.c:2763 <= flush output

I (7909) mbedtls: ssl_cli.c:3405 client state: 1

I (7909) mbedtls: ssl_tls.c:2751 => flush output

I (7919) mbedtls: ssl_tls.c:2763 <= flush output

I (7919) mbedtls: ssl_cli.c:774 => write client hello

I (7929) mbedtls: ssl_tls.c:3180 => write handshake message

I (7939) mbedtls: ssl_tls.c:3337 => write record

I (7939) mbedtls: ssl_tls.c:2751 => flush output

I (7949) mbedtls: ssl_tls.c:2770 message length: 241, out_left: 241

I (7949) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 241 (-0xffffff0f)

I (7959) mbedtls: ssl_tls.c:2803 <= flush output

I (7969) mbedtls: ssl_tls.c:3470 <= write record

I (7969) mbedtls: ssl_tls.c:3314 <= write handshake message

I (7979) mbedtls: ssl_cli.c:1106 <= write client hello

I (7979) mbedtls: ssl_cli.c:3405 client state: 2

I (7989) mbedtls: ssl_tls.c:2751 => flush output

I (7989) mbedtls: ssl_tls.c:2763 <= flush output

I (7999) mbedtls: ssl_cli.c:1499 => parse server hello

I (8009) mbedtls: ssl_tls.c:4305 => read record

I (8009) mbedtls: ssl_tls.c:2532 => fetch input

I (8019) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (8099) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (8099) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (8109) mbedtls: ssl_tls.c:2738 <= fetch input

I (8109) mbedtls: ssl_tls.c:2532 => fetch input

I (8119) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 74
[... 504 lines cut ...]
    */
    xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
                        false, true, portMAX_DELAY);
    ESP_LOGI(TAG, "Connect to Wifi ! Start to Connect to Server....");
    
    esp_http_client_config_t config = {
        .url =            "https://www.XXXXXXXXXX.com/updates/pol.bin",
        .cert_pem =       server_crt    ,      // <-- const char array  //(char *)server_cert_pem_start,
		.client_cert_pem= cli_crt  ,   // <-- const char array
		.client_key_pem=  cli_key  , // <-- const char array
        .event_handler = _http_event_handler,
    };
    esp_err_t ret = esp_https_ota(&config);
    if (ret == ESP_OK) {
        esp_restart();
    } else {
        ESP_LOGE(TAG, "Firmware Upgrades Failed");
    }
    while (1) {
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```

Regards, Kostas

KonstantinKondrashov, 2019-06-23:
Hi @kifantidis! These weekend I also tried to build OTA Example with mutual SSL authentication base on advanced_https_ota (for ESP-IDF: v4.0-dev-917-gd1da76e36). It works for me you can see how I did it [here](
https://github.com/KonstantinKondrashov/advanced_https_ota). It may be useful.

kifantidis, 2019-06-24:
Good morning @KonstantinKondrashov, thank you for your response. I'm going to check your link and let you know later my results !!

kifantidis, 2019-06-24:
Hello @KonstantinKondrashov , 
I've been testing since morning your example and tried to recreate it but with no success. When I finished with steps 1-8 the server was up and running I've created two of advanced_https_ota projects. The only difference between those is the FW version. The first has version 2 in the txt file and the other version 3. 

The first attempt ended up with the ESP32 failing to verify the peer's certificate. (Common name does not match with the expected CN error). I've tested the server with wget programm and I've got the same error: CN does not match with the expected CN.
The second attempt was done after disabling the certificate check from ssl_tls.c file in order to see if the handshake goes on after the first negotiation. This worked and the device did the re-negotiation with the server successfully. 
But it seems that the length of the .bin file downloaded from the device is not the same as the original. I figured this out by using wget to download the file in order to see what is happening. I've forgot to mention that I've disabled certification check in wget in order to be able to download the .bin file. 
I'm posting below the logs of the second try below: 

```
I (4587) example_connect: IPv4 address: 10.0.0.173
I (4597) example_connect: IPv6 address: fe80:0000:0000:0000:260a:c4ff:fe15:7bfc
I (4597) advanced_https_ota_example: Starting Advanced OTA example
I (4957) mbedtls: ssl_tls.c:8022 => handshake

I (4957) mbedtls: ssl_cli.c:3405 client state: 0

I (4957) mbedtls: ssl_tls.c:2751 => flush output

I (4967) mbedtls: ssl_tls.c:2763 <= flush output

I (4967) mbedtls: ssl_cli.c:3405 client state: 1

I (4977) mbedtls: ssl_tls.c:2751 => flush output

I (4977) mbedtls: ssl_tls.c:2763 <= flush output

I (4987) mbedtls: ssl_cli.c:774 => write client hello

I (4997) mbedtls: ssl_tls.c:3180 => write handshake message

I (4997) mbedtls: ssl_tls.c:3337 => write record

I (5007) mbedtls: ssl_tls.c:2751 => flush output

I (5007) mbedtls: ssl_tls.c:2770 message length: 239, out_left: 239

I (5017) mbedtls: ssl_tls.c:2775 ssl->f_send() returned 239 (-0xffffff11)

I (5027) mbedtls: ssl_tls.c:2803 <= flush output

I (5027) mbedtls: ssl_tls.c:3470 <= write record

I (5037) mbedtls: ssl_tls.c:3314 <= write handshake message

I (5037) mbedtls: ssl_cli.c:1106 <= write client hello

I (5047) mbedtls: ssl_cli.c:3405 client state: 2

I (5057) mbedtls: ssl_tls.c:2751 => flush output

I (5057) mbedtls: ssl_tls.c:2763 <= flush output

I (5067) mbedtls: ssl_cli.c:1499 => parse server hello

I (5067) mbedtls: ssl_tls.c:4305 => read record

I (5077) mbedtls: ssl_tls.c:2532 => fetch input

I (5077) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (5087) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (5097) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (5097) mbedtls: ssl_tls.c:2738 <= fetch input

I (5107) mbedtls: ssl_tls.c:2532 => fetch input

I (5107) mbedtls: ssl_tls.c:2693 in_left: 5, nb_want: 70

[... 618 lines cut ...]

I (9507) mbedtls: ssl_tls.c:4305 => read record

I (9517) mbedtls: ssl_tls.c:2532 => fetch input

I (9517) mbedtls: ssl_tls.c:2693 in_left: 0, nb_want: 5

I (9527) mbedtls: ssl_tls.c:2717 in_left: 0, nb_want: 5

I (9527) mbedtls: ssl_tls.c:2718 ssl->f_recv(_timeout)() returned 0 (-0x0000)

W (9537) mbedtls: ssl_tls.c:4967 mbedtls_ssl_fetch_input() returned -29312 (-0x7280)

W (9547) mbedtls: ssl_tls.c:4338 ssl_get_next_record() returned -29312 (-0x7280)

I (9557) esp_https_ota: Connection closed, all data received
I (9557) esp_image: segment 0: paddr=0x00120020 vaddr=0x3f400020 size=0x20b2c (133932) map
E (9647) esp_image: invalid segment length 0xffffffff
E (9647) advanced_https_ota_example: ESP_HTTPS_OTA upgrade failed...
```
I've tried this with 2-3 devices to check that there wasn't any problem with the particular device. If I can provide anything else to pinpoint the root of the problem please let me know. 

I know that I'm not far away from making this work. I'll look into it again tomorow. 

Best regards, Kostas.

kifantidis, 2019-06-25:
Hello @KonstantinKondrashov . If it's not a problem for you I would like to discuss something off the forum. If it is possible let me know in my email k.ifantidis@outlook.com. 
Best regards, Kostas.

kifantidis, 2019-06-25:
Thank you @KonstantinKondrashov  for the openssl server and the example that you have provided. I've figured out the problem and my device is now working. 

Best regards, Kostas.

closed 2019-06-25 by kifantidis