espressif/esp-idf #3676: Can't write or read the eMMC

khorinaj, 2019-06-21:
I am currently using the ESP32 to read/write sectors for the eMMC devices. The circuit set up is as below.
[image]
However, I can get the information from the chip, but the read/write operations are always fail with the error message  E (6505) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (6505) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
`#include <stdio.h>
#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_vfs_fat.h"
#include "driver/sdmmc_host.h"
#include "driver/sdspi_host.h"
#include "sdmmc_cmd.h"
#include "driver/gpio.h"

void setup(){
  Serial.begin(112500);
  
  sdmmc_host_init();

  sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
  slot_config.width = 4;

  sdmmc_host_init_slot(SDMMC_HOST_SLOT_1, &slot_config);

  sdmmc_host_t host = SDMMC_HOST_DEFAULT();
  //host.max_freq_khz = SDMMC_FREQ_HIGHSPEED;
  //host.max_freq_khz = SDMMC_FREQ_PROBING;

  sdmmc_card_t card;
  sdmmc_card_init(&host,&card);
  Serial.println("\n ready to get information");
  sdmmc_card_print_info(stdout, &card);

    pinMode(15, INPUT_PULLUP);
    pinMode(2, INPUT_PULLUP);
    pinMode(4, INPUT_PULLUP);
    pinMode(12, INPUT_PULLUP);
    pinMode(13, INPUT_PULLUP);
   
  sdmmc_write_sectors(&card, data, 1000, 1);
  Serial.println("check write");
  ESP_ERROR_CHECK( sdmmc_write_sectors(&card, data, 1000, 1) );
  sdmmc_read_sectors(&card, buff, 1000, 1);
  ESP_ERROR_CHECK( sdmmc_read_sectors(&card, buff, 1000, 1) );

}

void loop() {
  // put your main code here, to run repeatedly:

}`

the output is shown below
`ready to get information
Name: Q2J55L
Type: MMC
Speed: 20 MHz
Size: 7264MB
CSD: ver=3, sector_size=512, capacity=14876672 read_bl_len=9
E (6505) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (6505) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
check write
`

igrr, 2019-06-22:
Can you set log level to verbose in menuconfig, and attach the output you get?

khorinaj, 2019-06-22:
My project is Arduino based project, I don't think I can change the contents in menuconfig
[image]
Can I simply change like this? but nothing change. The error message is the same
[image]

igrr, 2019-06-22:
No, indeed the log level needs to be set when compiling with the esp-idf. Since your test code is mostly independent from Arduino functionality, can you switch to IDF for the purpose of debugging this? You can also use Arduino library as IDF component: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
This will allow log level to be changed, revealing more information about the cause of this issue.

khorinaj, 2019-06-23:
> No, indeed the log level needs to be set when compiling with the esp-idf. Since your test code is mostly independent from Arduino functionality, can you switch to IDF for the purpose of debugging [...]
I just need to check one thing first. I found that in this paper https://cdn.hackaday.io/files/269911154782944/esp32_datasheet_en.pdf
It said that the SD/SDIO/MMC host controller is available on ESP32 which supports Multimedia Cards (MMC Version 4.41, eMMC Version 4.5 and Version 4.51). Is there any updated eMMC version that the ESP32 supports? Since I am using the latest version eMMC 5.0 and 5.1 for testing.

omnitex, 2026-11-02:
Hi @khorinaj,

The read/write timeouts are caused by the `pinMode(..., INPUT_PULLUP)` calls after `sdmmc_card_init()`. On ESP32, SDMMC slot 1 uses fixed IO_MUX pins (CLK 14, CMD 15, D0 2, D1 4, D2 12, D3 13). 

Arduino's `pinMode()` switches these pins back to plain GPIO, which disconnects CMD and D0-D3 from the SDMMC controller. Card init succeeds because it runs before `pinMode()`, and `sdmmc_card_print_info()` only prints data that was already read, so the first command that fails is the write.

To fix it:
- Remove the `pinMode()` calls. Use external 10k pull-ups on CMD and D0-D3 (see the [SD pull-up requirements](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/sd_pullup_requirements.html)), or set `slot_config.flags |= SDMMC_SLOT_FLAG_INTERNAL_PULLUP` before `sdmmc_host_init_slot()` for testing only.
- Check the return value of `sdmmc_card_init()`.
- Note that GPIO12 (D2) is a strapping pin. A pull-up on it can break booting on modules with 3.3V flash. The pull-up requirements page explains how to handle this.

eMMC 5.0/5.1 devices are backward compatible and work with the ESP32 SDMMC host. Your card was identified correctly, so the protocol side is fine.

Closing this as it's not an IDF bug. Feel free to reopen if the problem persists after these changes.

closed 2026-11-02 by omnitex