espressif/esp-idf #3677: CMake: Possibility to specify location of mconf-idf when using IDF as library

PerMalmberg, 2019-06-22:
@renzbagaporo Some futher feedback on the CMake build system when using it together with an interface library between the main app and IDF.

Project structure:

- Main App ([G3](https://github.com/PerMalmberg/IO-Card-G3))
  - externals/smooth ([Interface library](https://github.com/PerMalmberg/Smooth))
 - IDF v4.0-dev-837-g58df1d93b located in /home/permal/esp/esp-idf

As the top-level projects needs to set its own settings, I need to pass the correct `sdkconfig` file down to the build system of IDF via the interface library, which I do like this;

[Top `CMakeLists.txt`](https://github.com/PerMalmberg/IO-Card-G3/blob/master/software/CMakeLists.txt):
```
set(project_specific_sdkconfig ${CMAKE_CURRENT_LIST_DIR}/sdkconfig)
add_subdirectory(externals/smooth/lib/smooth)
add_subdirectory(main)
```

Excerpt from interface library `CMakeLists.txt`, via [`idf_utils.cmake`](https://github.com/PerMalmberg/Smooth/blob/master/lib/smooth/idf_utils.cmake#L43)

```
# When a project specifies a specific sdkconfig, use it
if("${project_specific_sdkconfig}" EQUAL "")
    set(smooth_sdkconfig ${CMAKE_CURRENT_LIST_DIR}/sdkconfig)
else()
    set(smooth_sdkconfig ${project_specific_sdkconfig})
endif()

idf_build_process(esp32
    COMPONENTS ${smooth_req_comps} smooth_idf_component
    SDKCONFIG ${smooth_sdkconfig}
    BUILD_DIR ${CMAKE_BINARY_DIR})
```

This gets me so far that I running `cmake .. -G Ninja -DESP_PLATFORM=1 -DCMAKE_TOOLCHAIN_FILE=$IDF_PATH/tools/cmake/toolchain-esp32.cmake -DTARGET=esp32 && ninja` builds IDF, interfrace library and the main app using the correct sdkconfig. So far so good.

The problem is that when running `ninja menuconfig`, it emits the following error:

> No such file or directory

Dissecting the output of the command being run shows that the missing file it is `/home/permal/electronics/IO-Card-G3/software/build/kconfig_bin/mconf-idf`.

Not entirely unexpected, `mconf-idf` is placed here: `/home/permal/electronics/IO-Card-G3/software/build/externals/smooth/lib/smooth/kconfig_bin/mconf-idf`, i.e. within `${CMAKE_BINARY_DIR}/kconfig_bin` of IDF, as specified in [`kconfig.cmake`](https://github.com/espressif/esp-idf/blob/master/tools/cmake/kconfig.cmake#L41).

I thought that by specifying `BUILD_DIR ${CMAKE_BINARY_DIR}` in the call to `idf_build_process`, all IDF build artifacts would use that instead of `CMAKE_BINARY_DIR`, but alas that is not the case.

Apart from the [issue generating project_desctiption.json](https://github.com/espressif/esp-idf/issues/3640#issuecomment-504466980), this seems to be the last item that is preventing a pleasant user experience when using IDF as a library. Yes, the file can be modified by had, but its structure is rather unsuited for that and you loose all the help you get from the tool itself in terms of range limits etc.

As such, would it be possible to change the idf build system so that it uses the `BUILD_DIR` specified in the call to `idf_build_process` also when building mconf-idf`? I think this would make even `idf.py` function properly.

PerMalmberg, 2019-07-06:
@renzbagaporo If you get a moment to spare, I'd really appreciate some feedback on the above idea as not being able to run menuconfig without manually editing the command line each time is a real pain.

PerMalmberg, 2019-08-10:
@renzbagaporo Are you still working on changes to the CMake build system? This is still relevant.

renzbagaporo, 2019-08-16:
> I thought that by specifying BUILD_DIR ${CMAKE_BINARY_DIR} in the call to idf_build_process, all IDF build artifacts would use that, but alas that is not the case.

Yes, ESP-IDF has some artifacts placed in the build directory. In the case of mconf-idf, its because the targets for those are created before BUILD_DIR gets set via argument to `idf_build_process`.

This is certainly a bug however, and can be solved by changing https://github.com/espressif/esp-idf/blob/master/tools/cmake/kconfig.cmake#L50 to

```
            BINARY_DIR "${CMAKE_BINARY_DIR}/kconfig_bin"
```

Is this how you workaround this problem?

PerMalmberg, 2019-08-16:
>  Is this how you workaround this problem?

No, I edited the command line.

Good see that it's another bug found, I assume it'll be in master sometime soon.

ESP-Marius, 2025-08-13:
Since mconf was removed from IDF a long time ago, it seems like we can close this one.