espressif/esp-idf #3680: a2dp source bluetooth example can not connecting to bluetooth headset

yunling-cn, 2019-06-23:
https://github.com/espressif/esp-idf/issues/2116#issue-336413889

blueMoodBHD, 2019-07-08:
Hi @yunling-cn， I have fixed a bug about a2dp source connect fail with some speakers,  Refer to commit: a831fe596be6e176ab25f8dcf489dd5aa804b98f. Could you please update and test again?

Alvin1Zhang, 2019-07-19:
@yunling-cn Hi, thanks for reporting the issue, would you please share if any updates for the issue? Has your issue been resolved? Thanks.

Nathan-ma, 2019-08-01:
@Alvin1Zhang I can report the results, I just used the a2dp_source code with esp-idf v4.0 and the mentioned fix. 
It works 100%, I had a few BT speakers with the connection problem described now working great.

I just wanna know when I will be able to use this fix on arduino-esp32, seems to me this lib makes part of the esp-idf components. @blueMoodBHD can you tell me if I can use this fix with the v3.2? Can I simply update the submodule and place the .a file inside the folder or do I have to recompile something in order to generate the .a and make it work with the v3.2 version?

Anyway thank you.

blueMoodBHD, 2019-08-02:
No, you cann't replace the .a file into v3.2. I will fix it on release/v3.2, too. Please wait patiently, Thanks.

Nathan-ma, 2019-08-02:
Much Appreciated @blueMoodBHD , you are the best

Nathan-ma, 2019-08-20:
@blueMoodBHD Any news about this? can I use it on v3.2 already?

blueMoodBHD, 2019-08-21:
Sorry for later reply,  Yes, it have fixed in release/v3.2. Refer to commit  f023bf52c9c243efd15b02c3bb5397649ff89989

Alvin1Zhang, 2020-01-16:
@Nathan-ma @yunling-cn Thanks for reporting and updates. Feel free to reopen if the issue still happens. Thanks.

closed 2020-01-16 by Alvin1Zhang