espressif/esp-idf #3683: esp_camera_fb_get()  failed

Herwey, 2019-06-25:
Hi :
As we know, esp_camera_fb_get() is a api to get the camera buffer. But this api stopped in my main.c. I have check the function. "xQueueReceive(s_state->fb_out, &fb, portMAX_DELAY); " failed. because here is portMAX_DELAY. Can you help me ? Thanks !

camera_fb_t* esp_camera_fb_get()
{
    if (s_state == NULL) {
        return NULL;
    }
    if(!I2S0.conf.rx_start) {
        if(s_state->config.fb_count > 1) {
            ESP_LOGD(TAG, "i2s_run");
        }
        i2s_run();
    }
    if(s_state->config.fb_count == 1) {
        xSemaphoreTake(s_state->frame_ready, portMAX_DELAY);
    }
    if(s_state->config.fb_count == 1) {
        return (camera_fb_t*)s_state->fb;
    }
    camera_fb_int_t * fb = NULL;
    if(s_state->fb_out) {
        printf("s_state->fb_out\r\n" );
        xQueueReceive(s_state->fb_out, &fb, portMAX_DELAY);
        printf("s_state->fb_out complete\r\n" );
    }
    return (camera_fb_t*)fb;
}

me-no-dev, 2019-06-25:
maybe your camera is not running? If it times out, then something else is wrong. Please follow this in the esp32-camera repository and not here. The camera driver is not part of IDF :)

Herwey, 2019-06-25:
@me-no-dev Hi, I thinke the camera works fine. it prints" I (812) camera: i2s_run". And the same camera in another esp32 works fine.

me-no-dev, 2019-06-25:
moved to esp32-camera repo. closing here

closed 2019-06-25 by me-no-dev