espressif/esp-idf #3684: Gatt security and iOS, disconnects after 30 sec.

korstiaanS, 2019-06-25:
Hi,

I'm using idf version 3.2, ios 12.3.1.
App on iphone = LightBlue 
(BLE Scanner and nRF Connect does not work? After pressing on connect the login box appears BUT disappears automatically after 1 sec. Therefore I cannot enter the pincode)

First I tried the gatts table example from idf and that works without any timeout.

```
I (899) GATTS_TABLE_DEMO: advertising start successfully
I (10939) GATTS_TABLE_DEMO: ESP_GATTS_CONNECT_EVT, conn_id = 0
I (10939) GATTS_TABLE_DEMO: 76 9f 81 0e c7 67 
I (11089) GATTS_TABLE_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (11309) GATTS_TABLE_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 24,latency = 0, timeout = 400
I (11789) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (13469) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (15329) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (15779) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (15959) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (16139) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (18359) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (19259) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT

```

Then I added the security because I want to protect the variables via a pin code.
Therefore I added some code from the idf security server example.
It all works fine but after approx. 30 seconds I get the following messaging and have to reconnect again:

```
I (69908) GATTS_TABLE_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 24,latency = 0, timeout = 400
E (74768) BT_BTM: BTM_GetSecurityFlags false

E (74778) BT_GATT: GATT_INSUF_AUTHENTICATION

```

I also noticed that, after added the security, I get: 
`BT_L2CAP: l2cble_start_conn_update, the last connection update command still pending.`
The complete output is:

```
I (928) GATTS_TABLE_DEMO: advertising start successfully
I (25448) GATTS_TABLE_DEMO: ESP_GATTS_CONNECT_EVT, conn_id = 0
I (25448) GATTS_TABLE_DEMO: 45 59 29 20 d8 a3 
W (25458) BT_L2CAP: l2cble_start_conn_update, the last connection update command still pending.
I (25568) GATTS_TABLE_DEMO: ESP_GATTS_MTU_EVT, MTU 500
I (26438) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (26448) GATTS_TABLE_DEMO: update connection params status = 16, min_int = 16, max_int = 32,conn_int = 24,latency = 0, timeout = 72
I (43358) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (45248) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (47378) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (47858) GATTS_TABLE_DEMO: ESP_GATTS_READ_EVT
I (69908) GATTS_TABLE_DEMO: update connection params status = 0, min_int = 16, max_int = 32,conn_int = 24,latency = 0, timeout = 400
E (74768) BT_BTM: BTM_GetSecurityFlags false

E (74778) BT_GATT: GATT_INSUF_AUTHENTICATION
```

I suppose the timeout is coming from this `BT_L2CAP: l2cble_start_conn_update, the last connection update command still pending.`?

What did I do wrong?

This is the complete code:

```
/*
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
[... 561 lines cut ...]
    }

    /* set the security iocap & auth_req & key size & init key response key parameters to the stack*/
    esp_ble_auth_req_t auth_req = ESP_LE_AUTH_BOND;  //bonding with peer device after authentication
    esp_ble_io_cap_t   iocap    = ESP_IO_CAP_OUT;    //ESP_IO_CAP_NONE;           //set the IO capability to No output No input
    uint8_t            key_size = 16;                //the key size should be 7~16 bytes
    uint8_t            init_key = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    uint8_t            rsp_key  = ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK;
    //set static passkey
    uint32_t passkey     = 123456;
    uint8_t  auth_option = ESP_BLE_ONLY_ACCEPT_SPECIFIED_AUTH_DISABLE;
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_STATIC_PASSKEY, &passkey, sizeof(uint32_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_AUTHEN_REQ_MODE, &auth_req, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_IOCAP_MODE, &iocap, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_MAX_KEY_SIZE, &key_size, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_ONLY_ACCEPT_SPECIFIED_SEC_AUTH, &auth_option, sizeof(uint8_t));
    /* If your BLE device act as a Slave, the init_key means you hope which types of key of the master should distribut to you,
    and the response key means which key you can distribut to the Master;
    If your BLE device act as a master, the response key means you hope which types of key of the slave should distribut to you,
    and the init key means which key you can distribut to the slave. */
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_INIT_KEY, &init_key, sizeof(uint8_t));
    esp_ble_gap_set_security_param(ESP_BLE_SM_SET_RSP_KEY, &rsp_key, sizeof(uint8_t));
}

```

Weijian-Espressif, 2019-06-26:
@korstiaanS , Bluetooth disconnected after pairing failed. You set `ocap    = ESP_IO_CAP_OUT`, example will print passkey, your phone need input the passkey, but your `gap_event_handler ()` do nothing, So  pairing failed. please refer to our gatt_security_server example.

korstiaanS, 2019-06-26:
I saw that too, added some code but same result.
If I try the gatt_security_server example I get the same result. (disconnect after 30 sec.)
This is the result:

```
I (829) SEC_GATTS_DEMO: app_main init bluetooth
I (979) SEC_GATTS_DEMO: The number handle = 8
I (999) SEC_GATTS_DEMO: advertising start success
I (28279) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (28399) SEC_GATTS_DEMO: The passkey Notify number:123456
I (32369) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC
I (32369) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID
I (62479) SEC_GATTS_DEMO: remote BD_ADDR: 42230d487f68
I (62479) SEC_GATTS_DEMO: address type = 1
I (62479) SEC_GATTS_DEMO: pair status = fail
I (62489) SEC_GATTS_DEMO: fail reason = 0x63
I (62489) SEC_GATTS_DEMO: Bonded devices number : 0

I (62499) SEC_GATTS_DEMO: Bonded devices list : 0

W (65479) BT_BTM: btm_sec_clr_temp_auth_service() - no dev CB

I (65479) SEC_GATTS_DEMO: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x16
I (65489) SEC_GATTS_DEMO: advertising start success

```

After 30 sec. I get a fail?
Why?

Weijian-Espressif, 2019-06-26:
@korstiaanS, I use iPhone6s test gatt_security_server example, ESP32 works well.
please ` make erase_flash `   , Clear the pairing record in  your phone's Bluetooth and try, thanks.

here is the log:
I (8375) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (8445) SEC_GATTS_DEMO: The passkey Notify number:123456
I (12385) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LENC
I (12385) SEC_GATTS_DEMO: key type = ESP_LE_KEY_LID
I (12535) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PENC
I (12535) SEC_GATTS_DEMO: key type = ESP_LE_KEY_PID
I (12535) SEC_GATTS_DEMO: remote BD_ADDR: 75a92ebb3abc
I (12545) SEC_GATTS_DEMO: address type = 1
I (12545) SEC_GATTS_DEMO: pair status = success
I (12555) SEC_GATTS_DEMO: auth mode = INVALID BLE AUTH REQ
I (12555) SEC_GATTS_DEMO: Bonded devices number : 1

I (12565) SEC_GATTS_DEMO: Bonded devices list : 1

I (12565) SEC_GATTS_DEMO: 75 a9 2e bb 3a bc

korstiaanS, 2019-06-26:
Hi,

erase_flash did not help.

> Clear the pairing record in your phone's Bluetooth

Where is that? Under Settings Bluetooth is no ESP32 pairing visible?

korstiaanS, 2019-06-26:
Hi,

Tried on an ipad pro and there it worked.
A setting somewhere?

korstiaanS, 2019-06-26:
I found it.
It seems to be that the ESP was under Bluetooth devices BUT with a generic name and therefore I didn't found it directly. I deleted this and now it works fine.
THANKS!

I also have a small question:
If I In the sp_gatts_attr_db_t use a NULL for a characteristic value. Where will the received value be stored? Will the ESP library allocate memory for this? I noticed if I don't point to a variable, the ESP keeps the value so that at a read or write this value is saved. Where?

closed 2019-06-29 by korstiaanS