espressif/esp-idf #3685: OTA example handshake failure

morozovars, 2019-06-25:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version: [v4]
- IDF version:  3.2
- Build System:         [CMake]
- Compiler version:   1.22.0-80-g6c4433a
- Operating System:     [Windows]
- Power Supply:         [USB]

## Problem Description

I want to run OTA (native_ota example), but I meet failure during handshake procedure of connection. I do everything from this guide: https://github.com/espressif/esp-idf/tree/master/examples/system/ota 

I obtain these debug logs from ESP and from console:

## Debug Logs ESP32

I (4342) native_ota_example: Connect to Wifi ! Start to Connect to Server....
E (9012) esp-tls: mbedtls_ssl_handshake returned -0x2700
I (9012) esp-tls: Failed to verify peer certificate!
I (9012) esp-tls: verification info:   ! The certificate is signed with an unacceptable hash.

E (9022) esp-tls: Failed to open new connection
E (9032) TRANS_SSL: Failed to open a new connection
E (9032) HTTP_CLIENT: Connection failed, sock < 0
E (9042) native_ota_example: Failed to open HTTP connection: ERROR
E (9042) native_ota_example: Exiting task due to fatal error...

## Debug Logs in console

bad gethostbyaddr
12196:error:14094418:SSL routines:SSL3_READ_BYTES:tlsv1 alert unknown ca:.\ssl\s3_pkt.c:1060:SSL alert number 48
12196:error:140780E5:SSL routines:SSL23_READ:ssl handshake failure:.\ssl\s23_lib.c:142:
ACCEPT

## My steps

-1) Check my IP address by command "ipconfig"
0) Check by command "netstat -a" that port 8070 is free
1) Generate key and certificate by this command: "winpty openssl req -x509 -newkey rsa:2048 -keyout ca_key.pem -out ca_cert.pem -days 365 -nodes"
2) I write the same certificate parameters as on the animation in guide (beside CN)
3) Run server by this command "winpty openssl s_server -WWW -key ca_key.pem -cert ca_cert.pem -port 8070"
4) In menuconfig set settings of my wifi access point
5) In menuconfig set host myip:8070/hello-world.bin
6) Put hello-world.bin in folder where I run server in 3)
7) Put ca_cert.pem in directory server_certs/
8) Compile and flash OTA_Native example.

## Other items if possible
I also find this issue : https://github.com/espressif/esp-idf/issues/2296
But as I see in the guide you already fix RSA to 2048, but I still have this issue.

mastertheknife, 2019-08-09:
I also have this issue, also using 2048bit RSA.

morozovars, 2019-08-09:
> I also have this issue, also using 2048bit RSA.

Try to start cmd with administator rules. It help me. If not, try different versions of openssl.

mastertheknife, 2019-08-09:
I am using Linux, the SSL certificate was created like this:
```/usr/bin/openssl req -newkey rsa:2048 -keyout 'new.key' -nodes -x509 -days 3650 -out 'new.crt'```

OTA upgrades fail like this:
```␛[0;31mE (590636) esp-tls: mbedtls_ssl_handshake returned -0x2700␛[0m
␛[0;32mI (590636) esp-tls: Failed to verify peer certificate!␛[0m
␛[0;32mI (590636) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA␛[0m
␛[0;31mE (590646) esp-tls: Failed to open new connection␛[0m
␛[0;31mE (590656) TRANS_SSL: Failed to open a new connection␛[0m
␛[0;31mE (590656) HTTP_CLIENT: Connection failed, sock < 0␛[0m
␛[0;31mE (590666) esp_https_ota: Failed to open HTTP connection: ERROR␛[0m
```
It seems like an issue in ESP-IDF version 3.2, older versions are probably not affected.
The same code works perfectly fine on ESP8266 with RTOS SDK v3.2

sachin0x18, 2019-08-27:
Hi @morozovars,

I tried it on a Windows machine using MSYS and openssl (version: 1.1.1c 28 May 2019) to generate the certificate and key and I was able to successfully execute the OTA example.

Are you still facing this issue ? Can you try it once using MSYS and openssl ?

mastertheknife, 2019-09-30:
> I am using Linux, the SSL certificate was created like this:
> `/usr/bin/openssl req -newkey rsa:2048 -keyout 'new.key' -nodes -x509 -days 3650 -out 'new.crt'`
> 
> OTA upgrades fail like this:
> 
> ```
> ␛[0;32mI (590636) esp-tls: Failed to verify peer certificate!␛[0m
> ␛[0;32mI (590636) esp-tls: verification info:   ! The certificate is not correctly signed by the trusted CA␛[0m
> ␛[0;31mE (590646) esp-tls: Failed to open new connection␛[0m
> ␛[0;31mE (590656) TRANS_SSL: Failed to open a new connection␛[0m
> ␛[0;31mE (590656) HTTP_CLIENT: Connection failed, sock < 0␛[0m
> ␛[0;31mE (590666) esp_https_ota: Failed to open HTTP connection: ERROR␛[0m
> ```
> 
> It seems like an issue in ESP-IDF version 3.2, older versions are probably not affected.
> The same code works perfectly fine on ESP8266 with RTOS SDK v3.2

Hi, just a small update,
After updating to ESP-IDF v3.3, everything works as expected. No code modifications, just re-compile and flash.

mahavirj, 2019-10-01:
Thanks for confirmation, I will go ahead and close the issue.

closed 2019-10-01 by mahavirj