espressif/esp-idf #3687: collect2.exe: error: ld returned 1 exit status

meteorego, 2019-06-26:
Environment

    Development Kit: ESP32-DevKitC
    Kit version (for DevKitC): v1
    Module or chip used: ESP32-WROOM-32
    IDF version: v4.0-dev-71-g5beb2802e
    Build System: CMake
    Compiler version: 1.22.0-80-g6c4433a5 - 5.2.0
    Operating System: Windows 10
    Power Supply: USB

Problem Description

F:\Work\esp\hello_world>idf.py menuconfig
Checking Python dependencies...
Python requirements from D:\esp\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory f:\work\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized f:\work\esp\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.18.0.windows.1")
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
CMake Warning at D:/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Xtensa toolchain C:/Program Files/Espressif/ESP-IDF
  Tools/tools/bin/xtensa-esp32-elf-gcc.exe version 5.2.0 is not the supported
  version 8.2.0.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  D:/esp/esp-idf/components/esp32/project_include.cmake:20 (gcc_version_check)
  D:/esp/esp-idf/tools/cmake/build.cmake:287 (include)
  D:/esp/esp-idf/tools/cmake/build.cmake:440 (__build_process_project_includes)
  D:/esp/esp-idf/tools/cmake/project.cmake:315 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Warning at D:/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain C:/Program Files/Espressif/ESP-IDF
  Tools/tools/bin/xtensa-esp32-elf-gcc.exe crosstool-ng version
  crosstool-ng-1.22.0-80-g6c4433a5 doesn't match supported version
  esp32-2019r1.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  D:/esp/esp-idf/components/esp32/project_include.cmake:21 (crosstool_version_check)
  D:/esp/esp-idf/tools/cmake/build.cmake:287 (include)
  D:/esp/esp-idf/tools/cmake/build.cmake:440 (__build_process_project_includes)
  D:/esp/esp-idf/tools/cmake/project.cmake:315 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Found PythonInterp: C:/Python27/python.exe (found version "2.7.14")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script F:/Work/esp/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script D:/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script D:/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script D:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
[... 51 lines cut ...]
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
1
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lsim
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhandlers-sim
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhal
collect2.exe: error: ld returned 1 exit status

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe 
Build flags: -mlongcalls;-Wno-frame-address
Id flags:  

The output was:
1
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find crt1-sim.o: No such file or directory
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find _vectors.o: No such file or directory
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lsim
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhandlers-sim
c:/program files/espressif/esp-idf tools/tools/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lhal
collect2.exe: error: ld returned 1 exit status

meteorego, 2019-06-26:
There have not a directory on your system named C:\esp32\?

renzbagaporo, 2019-06-26:
Yes, this is an issue with a fix on its way. This was a regression in e8582e9a, in the meantime, you can revert your branch prior to this commit.

meteorego, 2019-06-27:
@renzbagaporo
The version of esp-idf is the latest, but it is still the above problem, how to solve it?

closed 2019-06-30 by igrr