espressif/esp-idf #3688: ble_mesh_fast_provision client compile error

tatsutaigu, 2019-06-26:
Hi,

I try ble_mesh_fast_provision client.
however, I got error as below.

> 
> $ make flash
> Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
> Compiler version: 5.2.0
> Python requirements from C:/msys32/home/tatsutaigu/esp/esp-idf-ble_mesh_release-esp-ble-mesh-v0.6.1/requirements.txt are satisfied.
> Building partitions from /home/tatsutaigu/esp/esp-idf-ble_mesh_release-esp-ble-mesh-v0.6.1/components/partition_table/partitions_singleapp.csv...
> 
> App "ble_mesh_fast_prov_client" version: v4.0-dev-297-g54ecdca32
> CC build/libsodium/libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.o
> C:/msys32/home/tatsutaigu/esp/esp-idf-ble_mesh_release-esp-ble-mesh-v0.6.1/components/libsodium/libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.c:375:1: fatal error: opening dependency file libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.d: No such file or directory
>  }
>  ^
> compilation terminated.
> make[1]: *** [/home/tatsutaigu/esp/esp-idf-ble_mesh_release-esp-ble-mesh-v0.6.1/make/component_wrapper.mk:290: libsodium/src/libsodium/crypto_pwhash/scryptsalsa208sha256/nosse/pwhash_scryptsalsa208sha256_nosse.o] Error 1
> make: *** [/home/tatsutaigu/esp/esp-idf-ble_mesh_release-esp-ble-mesh-v0.6.1/make/project.mk:578: component-libsodium-build] Error 2
> 

esp-idf version is here.

> $ git log -2
> commit 54ecdca3207ae540714dc986de723b49fbf5fd91 (HEAD -> ble_mesh_release/esp-ble-mesh-v0.6.1, origin/ble_mesh_release/esp-ble-mesh-v0.6.1)
> Author: lly <lly@espressif.com>
> Date:   Thu Apr 25 12:07:36 2019 +0800
> 
>     component/ble_mesh: fix sending GATT service change indication
> 
> commit c38be0f81b97cce97f5d3ee860ae98151c8bc265
> Author: wangcheng <wangcheng@espressif.com>
> Date:   Wed Apr 10 15:40:40 2019 +0800
> 
>     component/ble_mesh: Add user manuals for ESP BLE Mesh examples
>

WCCWCC, 2019-06-27:
Hi tatsutaigu，
     
     You should execute the command in your IDF_PATH: git submodule update - - init - - recursive.

tatsutaigu, 2019-06-28:
Hi WCCWCC,

I tried to execute `git submodule update --init --recursive`.
But, it could not solve.

WCCWCC, 2019-06-28:
Hi tatsutaigu，

* Can your other `ble mesh` projects compile normally?  
    * I would like to ask you which document to compile？ I can see if this problem can be reproduced.

I am currently using a linux environment.

tatsutaigu, 2019-06-28:
Hi WCCWCC,

My PC's OS is windows 7.
I guess folder name is so long.
I will try to make it by short folder name.
Currently,  I do `git clone`.

When I finish it, I will update it.

WCCWCC, 2019-06-28:
Hi tatsutaigu，

  You may refer to this document：

  https://docs.espressif.com/projects/esp-idf/en/latest/get-started/windows-setup.html

tatsutaigu, 2019-06-28:
Hi WCCWCC,

I could compile it.
Root cause is too long folder name of ESP-IDF for ble_mesh_v0.6.1.

BTW, I found below message when I did menuconig.
I will install latest toolcahin for ble_mesh_v0.6.1.

> Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
> WARNING: Toolchain version is not supported: crosstool-ng-1.22.0-80-g6c4433a5
> Expected to see version: esp32-2019r1
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> WARNING: Compiler version is not supported: 5.2.0
> Expected to see version(s): 8.2.0
> Please check ESP-IDF setup instructions and update the toolchain, or proceed at your own risk.
> Python requirements from C:/msys32/home/m-muto/esp/esp-idf-ble_mesh_v0.6.1/requirements.txt are satisfied.
> MENUCONFIG
>

Alvin1Zhang, 2019-07-17:
@tatsutaigu Thanks for reporting the issue, has your issue been resolved? Thanks.

closed 2019-10-11 by Alvin1Zhang