espressif/esp-idf #3689: Error make monitor for hello world

aadnesd, 2019-06-26:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version DevKitC: v4]
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it): 
    // v3.2
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         [USB]

## Problem Description

//Detailed problem description goes here.
When trying to "make monitor" while following the get started example (Hello world), I get an error. However if i run make simple_monitor it works as expected

### Expected Behavior
Wanted it to return Hello World as it said it should

### Actual Behavior

## Debug Logs

# make monitor
Toolchain path: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433a5
Compiler version: 5.2.0
Python requirements from C:/msys32/home/▒dne/esp/esp-idf/requirements.txt are satisfied.
MONITOR
--- idf_monitor on COM7 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6180
load:0x40078000,len:10180

Traceback (most recent call last):
  File "C:/msys32/home/
make: *** [/home/Ådne/esp/esp-idf/components/esptool_py/Makefile.projbuild:106: monitor] Error 1

igrr, 2019-06-26:
Can you please check which Python version is being used?

Also, can you please try to move the esp directory to some path which doesn't contain non-ASCII characters, delete the build directory, and try `make monitor` again?

aadnesd, 2019-06-26:
The python version is 2.7.16.
I will try this tomorow, however, when I tried on a new kit just before leaving today I wasn't even able to flash hello world to the chip. I got the error:

No rule to make target 'C:/msys32/home/▒dne/esp/esp-idf/components/ bootloader_support/src/bootloader_random.c', needed by 'src/bootloader_random.o'

aadnesd, 2019-06-27:
I deleted everything and went through the get started documentation again, this time for the latest version.
I created a new folder and moved the esp in this one, and now it works OK. Both flashing and make monitor. However I can't run cd ~/esp because this folder still looks for /home/Ådne/esp which doesn't exists. I don't know if that will be a problem in the future?

closed 2019-06-27 by aadnesd