espressif/esp-idf #3690: Controller mode config (BTDM/BLE) question in newer idf version

kvbulusu, 2019-06-26:
Environment
Development Kit: ESP32-LyraT
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-837-g58df1d93b
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: Linux
Power Supply: USB

Problem Description
It seems like CONFIG_BTDM_CONTROLLER_MODE_BTDM 
 & CONFIG_BTDM_CONTROLLER_MODE_BLE_ONLY  are deprecated. 

What is right config to select the mode now ?

dobairoland, 2019-06-26:
Hi @kvbulusu. Here are the new options: https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-btdm-ctrl-mode (`CONFIG_BTDM_CTRL_MODE_BTDM` and `CONFIG_BTDM_CTRL_MODE_BLE_ONLY`).

kvbulusu, 2019-06-26:
Thanks @dobairoland

closed 2019-06-26 by kvbulusu