espressif/esp-idf #3691: error in BT function bta_dm_config_eir  ( argument 1 null where non-null expected)

kvbulusu, 2019-06-26:
Environment
Development Kit: ESP32-LyraT
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v4.0-dev-837-g58df1d93b
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
Operating System: Linux
Power Supply: USB

Problem Description: 
Error while compiling  . 
esp-idf/components/bt/bluedroid/bta/dm/bta_dm_act.c: In function 'bta_dm_config_eir':
esp-idf/components/bt/bluedroid/bta/dm/bta_dm_act.c:687:13: error: argument 1 null where non-null expected [-Werror=nonnull]
             memcpy(p_bta_dm_eir_cfg->bta_dm_eir_url, config_eir->eir_url, config_eir->eir_url_len);
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from esp-idf/components/bt/bluedroid/bta/dm/bta_dm_act.c:40:
tools/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/string.h:31:9: note: in a call to function 'memcpy' declared here
 void *  memcpy (void *__restrict, const void *__restrict, size_t);

blueMoodBHD, 2019-06-27:
Thanks for reporting this. I will fix it.

blueMoodBHD, 2019-07-03:
Hi @kvbulusu, This have been fixed, refer to commit 45364fe54ea97807f2ffdcbf99129d6870292454

blueMoodBHD, 2019-07-03:
Could you please update and test. And close the issue is there is no problem.

Alvin1Zhang, 2019-07-03:
@kvbulusu Would you please help verify the fix and help provide feedback? Thanks.

kvbulusu, 2019-07-03:
@blueMoodBHD @Alvin1Zhang  ..Yes that error is gone and I'm able to compile.. I see other errors now. will open new issues. But just FYI. 

esp-idf/components/bt/common/osi/config.c:446:49: error: '%d' directive output may be truncated writing between 1 and 10 bytes into a region of size 7 [-Werror=format-truncation=]
             snprintf(keyname, keyname_bufsz, "%s%d", CONFIG_KEY, i);
                                                 ^~
esp-idf/components/bt/common/osi/alarm.c:85:1: error: 'xyz' undeclared (first use in this function)
 xyz
 ^~~
esp-idf/components/bt/common/osi/alarm.c:85:1: note: each undeclared identifier is reported only once for each function it appears in
esp-idf/components/bt/common/osi/alarm.c:85:4: error: expected ';' before 'if'
 xyz
    ^
    ;
 if ((alarm_cbs = (osi_alarm_t *)osi_malloc(sizeof(osi_alarm_t) * ALARM_CBS_NUM)) == NULL) {

closed 2019-07-03 by kvbulusu