espressif/esp-idf #3692: I2S sample rate is halved

detly, 2019-06-26:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at esp32.com instead.

## Environment

- Development Kit: Wemos Lolin D32 
- Module or chip used:  ESP32-WROOM-32
- IDF version: `master` (`58df1d93`)
- Build System: Cmake
- Compiler version: `1.22.0-80-g6c4433a`
- Operating System: Linux (Ubuntu 18.04)
- Power Supply: USB

## Problem Description

When I configure I2S to use a sample rate of 24kHz, it displays 12kHz.

### Expected Behavior

The I2S sample rate is confirmed to be 24kHz.

### Actual Behavior

The I2S info messages tell me it's 12kHz. The actual sample rate seems to be correct though — if I run a 1kHz tone generator into a microphone connected to the ADC, I get peaks 24 samples apart, which is correct.

### Steps to repropduce

See the code in [this Gitlab project](https://gitlab.com/detly/i2s_test/tree/i2s_rate_bug). Compile and flash it with `idf.py build` and `idf.py flash monitor`. You will see the following I2S-related messages:

```text
I (284) I2S: DMA Malloc info, datalen=blocksize=4092, dma_buf_count=2
I (284) I2S: APLL: Req RATE: 24000, real rate: 11999.990, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 3071997.500, SCLK: 383999.687500, diva: 1, divb: 0
I (304) I2S: APLL: Req RATE: 24000, real rate: 11999.990, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 3071997.500, SCLK: 383999.687500, diva: 1, divb: 0
```

It's even worse if you have `use_apll = 0`:

```text
I (283) I2S: DMA Malloc info, datalen=blocksize=4092, dma_buf_count=2
I (283) I2S: PLL_D2: Req RATE: 24000, real rate: 1515.000, BITS: 16, CLKM: 55, BCK: 60, MCLK: 55.556, SCLK: 48480.000000, diva: 64, divb: 35
I (303) I2S: PLL_D2: Req RATE: 24000, real rate: 1515.000, BITS: 16, CLKM: 55, BCK: 60, MCLK: 55.556, SCLK: 48480.000000, diva: 64, divb: 35
```
### Code to reproduce this issue

```c
void app_main()
{
    int i2s_num = I2S_NUM_0;

    i2s_config_t i2s_config = {
        .mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN,
        .sample_rate =  24000,
        .bits_per_sample = 16,
        .communication_format = I2S_COMM_FORMAT_I2S_MSB,
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
        .intr_alloc_flags = 0,
        .dma_buf_count = 2,
        .dma_buf_len = 1024,
        .use_apll = 1,
        .fixed_mclk = 0
    };

    i2s_driver_install(i2s_num, &i2s_config, 0, NULL);
    i2s_set_adc_mode(ADC_UNIT_1, ADC1_CHANNEL_6);

    ESP_ERROR_CHECK(i2s_adc_enable(I2S_NUM_0));
}
```

But see the Gitlab project I linked to.

## Debug Logs

<details>
<summary>Debug logs</summary>

```text
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6584
load:0x40078000,len:10748
load:0x40080400,len:6676
entry 0x40080768
I (29) boot: ESP-IDF v4.0-dev-837-g58df1d93b 2nd stage bootloader
I (29) boot: compile time 09:08:09
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 2MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x09d38 ( 40248) map
I (108) esp_image: segment 1: paddr=0x00019d60 vaddr=0x3ffb0000 size=0x01ec8 (  7880) load
I (111) esp_image: segment 2: paddr=0x0001bc30 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/[...]/Code/esp-libs/esp-idf/components/freertos/xtensa_vectors.S:1778

I (115) esp_image: segment 3: paddr=0x0001c038 vaddr=0x40080400 size=0x03fd8 ( 16344) load
I (130) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x15818 ( 88088) map
0x400d0018: _flash_cache_start at ??:?

I (164) esp_image: segment 5: paddr=0x00035838 vaddr=0x400843d8 size=0x03f64 ( 16228) load
0x400843d8: vTaskSwitchContext at /home/[...]/Code/esp-libs/esp-idf/components/freertos/tasks.c:4561

I (176) boot: Loaded app from partition at offset 0x10000
I (176) boot: Disabling RNG early entropy source...
I (176) cpu_start: Pro cpu up.
I (180) cpu_start: Application information:
I (185) cpu_start: Project name:     i2s-test
I (190) cpu_start: App version:      i2s_rate_bug-dirty
I (196) cpu_start: Compile time:     Jun 27 2019 09:18:14
I (202) cpu_start: ELF file SHA256:  0259c48bdcfca3b0...
I (208) cpu_start: ESP-IDF:          v4.0-dev-837-g58df1d93b
I (214) cpu_start: Starting app cpu, entry point is 0x40080e68
0x40080e68: call_start_cpu1 at /home/[...]/Code/esp-libs/esp-idf/components/esp32/cpu_start.c:265

I (0) cpu_start: App cpu up.
I (225) heap_init: Initializing. RAM available for dynamic allocation:
I (232) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (238) heap_init: At 3FFB2EF8 len 0002D108 (180 KiB): DRAM
I (244) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (250) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (257) heap_init: At 4008833C len 00017CC4 (95 KiB): IRAM
I (263) cpu_start: Pro cpu start user code
I (281) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (283) I2S: DMA Malloc info, datalen=blocksize=4092, dma_buf_count=2
I (283) I2S: APLL: Req RATE: 24000, real rate: 11999.990, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 3071997.500, SCLK: 383999.687500, diva: 1, divb: 0
I (303) I2S: APLL: Req RATE: 24000, real rate: 11999.990, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 3071997.500, SCLK: 383999.687500, diva: 1, divb: 0
```
</details>

negativekelvin, 2019-06-27:
Related to https://github.com/espressif/esp-idf/issues/3380 ?

detly, 2019-06-27:
Git bisect gives:

```text
4b34ae1258c03e1c67cf51348855ff6e74dd21fd is the first bad commit
commit 4b34ae1258c03e1c67cf51348855ff6e74dd21fd
Author: Wangjialin <wangjialin@espressif.com>
Date:   Tue Nov 27 14:10:33 2018 +0800

    bugfix(i2s): allow to use apll in pdm/adc/dac mode
    
    1. Allow to use apll in pdm mode
    2. Add clock frequency configuration for PDM mode
    3. Allow to use apll in ADC/DAC mode

 components/driver/i2s.c                          | 41 ++++++++++++++++--------
 components/driver/include/driver/i2s.h           | 27 ++++++++++++++++
 examples/peripherals/i2s_adc_dac/main/app_main.c |  3 +-
 3 files changed, 56 insertions(+), 15 deletions(-)
```

detly, 2019-06-27:
The problem with `I (137) I2S: PLL_D2: Req RATE: 24000, real rate: 1515.000` is not fixed before this commit, I could not find a commit where that works as expected.

detly, 2019-06-27:
I was wrong about the physical sample rate being correct. Enabling the APLL in this code (either in `master` or the last "good" commit) results in a signal that is stretched by about a factor of 8. Leaving the APLL off works fine, even though the reported sample rate is 1515Hz instead of 24kHz.

detly, 2019-06-27:
Okay, with the help of the example I2S code and an oscilloscope, I figured out that this only happens when `mode = I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN`.

I took the [example code here](https://github.com/espressif/esp-idf/blob/4fc411e744a87efb68993ff4d7560750f7f96dc9/examples/peripherals/i2s/main/i2s_example_main.c) and changed the `mode` to `I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN` and the `sample_rate` to `24000`. I used an oscilloscope to measure the frequency on GPIO 25. But I'm not sure how seriously I should take the I2S CLK signal when it's in ADC built-in mode.

On `master`:

  * with APLL enabled: reported rate is 11999.990 Hz. If I use the pin configuration from the example code (with I2S CLK on GPIO 25), I see the frequency switch between 192kHz, 288kHz and 384kHz. 
  * with APLL disabled: reported rate is 1515 Hz or lower (I've seen down to 757 Hz reported), but the oscilloscope shows 24kHz with no variation.

On `90a6c9aad`:

  * with APLL enabled: reported rate is 24kHz, but oscilloscope shows frequencies of 384kHz, 576kHz, 764kHz.
  * with APLL disabled: same as above.

detly, 2019-06-28:
I noticed that without the in-built ADC mode enabled, I see:

```text
D (326) I2S: sdm0=2, sdm1=154, sdm2=4, odir=12
```

Which is correct. With it enabled, I see:

```text
D (326) I2S: sdm0=2, sdm1=154, sdm2=4, odir=26
```

Which effectively halves the rate.

negativekelvin, 2019-06-28:
https://github.com/espressif/esp-idf/blob/bbafd506624b2a3d1e58f2dd3154286b79a8632b/components/driver/i2s.c#L409-L411

detly, 2019-06-28:
@negativekelvin Yeah I just saw that (see also [this comment](https://github.com/espressif/esp-idf/issues/3686#issuecomment-506603145)). But it's not doing the right thing as implemented — for built-in ADC operation, WS should be the sample rate requested, but comes out much higher.

detly, 2019-06-28:
Okay, in 12.5 it says:

> In the LCD mode, the frequency of WS is half of f BCK. In the ADC/DAC mode, use PLL_D2_CLK as the clock source.

So using APLL in ADC mode is not supported.

detly, 2019-06-28:
So this is invalid, but my other issue re. channel doubling up still is (I think).

closed 2019-06-28 by detly