espressif/esp-idf #3693: Heavy logs at boot

jilchab, 2019-06-27:
## Environment

- Development Kit:  none
- Module or chip used: ESP32
- IDF version: v4.0-dev-764-g9a128e19c
- Build System:    Make
- Compiler version: 1.22.0-80-g6c4433a
- Operating System: macOS
- Power Supply: external 3.3V

## Problem Description

Logs from "memory_layout", "heap_init", "intr_alloc" and "efuse" can't be removed from compilation by log level without removing all other logs from the program. These logs slow down the startup time.

### Expected Behavior

These logs should be disabled with the bootloader log level, or another log level, to be able to discard them while keeping light verbose logs in the program.

### Actual Behavior

These logs share the same log level that all others logs.

## Debug Logs

Logs before entering `main()`, taking 1.3s:

```
I (0) cpu_start: App cpu up.
V (408) memory_layout: reserved range is 0x3f422a84 - 0x3f422aac
D (414) memory_layout: Checking 7 reserved memory ranges:
D (419) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (425) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffba998
D (432) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (438) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (445) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (451) memory_layout: Reserved memory range 0x40078000 - 0x4008000

D (458) memory_layout: Reserved memory range 0x40080000 - 0x40090e2

D (464) memory_layout: Building list of available memory regions:
V (470) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
V (477) memory_layout: Start of region 0x3ffae000 - 0x3ffb0000 overlaps reserved 0x3ffae000 - 0x3ffae6e0
D (486) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
V (493) memory_layout: Examining memory region 0x3ffb0000 - 0x3ffb8000
V (500) memory_layout: Region 0x3ffb0000 - 0x3ffb8000 inside of reserved 0x3ffb0000 - 0x3ffba998
V (508) memory_layout: Examining memory region 0x3ffb8000 - 0x3ffc0000
V (515) memory_layout: Start of region 0x3ffb8000 - 0x3ffc0000 overlaps reserved 0x3ffb0000 - 0x3ffba998
D (525) memory_layout: Available memory region 0x3ffba998 - 0x3ffc0000
V (531) memory_layout: Examining memory region 0x3ffc0000 - 0x3ffc2000
D (538) memory_layout: Available memory region 0x3ffc0000 - 0x3ffc2000
V (544) memory_layout: Examining memory region 0x3ffc2000 - 0x3ffc4000
D (551) memory_layout: Available memory region 0x3ffc2000 - 0x3ffc4000
V (557) memory_layout: Examining memory region 0x3ffc4000 - 0x3ffc6000
D (564) memory_layout: Available memory region 0x3ffc4000 - 0x3ffc6000
V (571) memory_layout: Examining memory region 0x3ffc6000 - 0x3ffc8000
D (577) memory_layout: Available memory region 0x3ffc6000 - 0x3ffc8000
V (584) memory_layout: Examining memory region 0x3ffc8000 - 0x3ffca000
D (590) memory_layout: Available memory region 0x3ffc8000 - 0x3ffca000
V (597) memory_layout: Examining memory region 0x3ffca000 - 0x3ffcc000
D (604) memory_layout: Available memory region 0x3ffca000 - 0x3ffcc000
V (610) memory_layout: Examining memory region 0x3ffcc000 - 0x3ffce000
D (617) memory_layout: Available memory region 0x3ffcc000 - 0x3ffce000
V (623) memory_layout: Examining memory region 0x3ffce000 - 0x3ffd0000
D (630) memory_layout: Available memory region 0x3ffce000 - 0x3ffd0000
V (637) memory_layout: Examining memory region 0x3ffd0000 - 0x3ffd2000
D (643) memory_layout: Available memory region 0x3ffd0000 - 0x3ffd2000
V (650) memory_layout: Examining memory region 0x3ffd2000 - 0x3ffd4000
D (656) memory_layout: Available memory region 0x3ffd2000 - 0x3ffd4000
V (663) memory_layout: Examining memory region 0x3ffd4000 - 0x3ffd6000
D (670) memory_layout: Available memory region 0x3ffd4000 - 0x3ffd6000
[... 101 lines cut ...]
D (1164) heap_init: New heap initialised at 0x40090e2c
I (1169) heap_init: At 40090E2C len 0000F1D4 (60 KiB): IRAM
I (1176) cpu_start: Pro cpu start user code
D (1188) clk: RTC_SLOW_CLK calibration value: 3259290
V (1197) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1197) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1203) intr_alloc: Connected src 46 to int 2 (cpu 0)
V (1208) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1214) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xC0E
D (1222) intr_alloc: Connected src 57 to int 3 (cpu 0)
V (1227) esp_dbg_stubs: esp_dbg_stubs_init stubs 3ffb35c4
V (1234) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1239) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (1248) intr_alloc: Connected src 24 to int 9 (cpu 0)
I (1253) cpu_start: Starting scheduler on PRO CPU.
V (0) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): checking args
V (6) intr_alloc: esp_intr_alloc_intrstatus (cpu 1): Args okay. Resulting flags 0x40E
D (14) intr_alloc: Connected src 25 to int 2 (cpu 1)
I (18) cpu_start: Starting scheduler on APP CPU.
D (1307) heap_init: New heap initialised at 0x3ffe0440
D (1311) heap_init: New heap initialised at 0x3ffe4350
V (1317) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1317) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1317) intr_alloc: Connected src 16 to int 12 (cpu 0)
```

negativekelvin, 2019-06-27:
Perhaps a way to set the LOG_LOCAL_LEVEL individually for each component at the project level?

jilchab, 2019-06-28:
@negativekelvin Actually this should do the trick. Being able to set a general log level (like today), and then override the level of some tags in order to remove them from compilation would be great!

Allineer, 2019-08-29:
Is it possible to solve this problem without decreasing the value of CONFIG_LOG_DEFAULT_LEVEL?

Dazza0, 2021-09-23:
- To configure the logging level used by the bootloader, configure the [CONFIG_BOOTLOADER_LOG_LEVEL](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/api-reference/kconfig.html#config-bootloader-log-level) option found under the `Bootloader config` in `menuconfig`
- To configure the logging level of the application, configure the [CONFIG_LOG_DEFAULT_LEVEL](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/api-reference/kconfig.html#config-bootloader-log-level) option.
    - @Allineer Note that the application log level can be lowered (but not raised) at runtime using `esp_log_level_set()`
- Both the bootloader log level and application log level can be overridden locally in a source file by defining `LOG_LOCAL_LEVEL` in that source file.
- See the [logging library documentation](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/api-reference/system/log.html#logging-library) for more details

I'm closing this issue. Please feel free to reopen if you have further inquires.

closed 2021-09-23 by Dazza0