espressif/esp-idf #3696: Bluetooth Host = NimBLE won't compile references to BLE headers

mws-rmain, 2019-06-27:
Setting 'Bluetooth Host = NimBLE' (i.e., CONFIG_BT_NIMBLE_ENABLED=y in sdkconfig) will cause builds to fail if BLE headers are included.

For instance, copy the standard 'bleprph' example project & ensure it compiles. 
At the top of 'main.c', add:
`#include "esp_gatt_defs.h"`
and recompile.
```
/home/mainr/workspace/bleprph/main/main.c:23:10: fatal error: esp_gatt_defs.h: No such file or directory
 #include "esp_gatt_defs.h"
          ^~~~~~~~~~~~~~~~~
compilation terminated.
```
Compiling older BLE example projects with 'Bluetooth Host = Bluedroid - Dual-mode' is able to properly include 'esp_gatt_defs.h'.

I believe BLE header files such as 'esp_gatt_defs.h' should be available for inclusion in NimBLE projects, but there appears to be an 'include path' issue when building this configuration.

dhrishi, 2019-06-28:
@mws-rmain You should not include `esp_gatt_defs.h` in NimBLE code. It is from the Bluedroid host code and not used/applicable in NimBLE

If you see, NimBLE porting layer is kept cleaner by not adding any abstraction APIs (esp_ble_*) and maintaining all the existing APIs of NimBLE along with a couple ESP-NimBLE API for init/de-init, making it simpler for the application developers.

mws-rmain, 2019-06-28:
Of course, you're right. I took a look this morning, and found (for instance) references to the GATT attribute 0x2A29 (Manufacturer Name) in 'components/nimble/host/services/dis/include/services/dis/ble_svc_dis.h'.

Clearly porting an existing ESP bluedroid BLE app to NimBLE won't be without pain.
Best Regards.

closed 2019-06-28 by mws-rmain