espressif/esp-idf #3697: ppp client - cannot enter ppp state twice

Sagigamil, 2019-06-27:
I am working with the esp idf: beaefd3359973dcf8bba4e1d4462aa7f5d67be6a - last master, with the ppp client example.
My setup contain sim800 gsm/gprs module and an esp32 sparkfun.
I am tring to recieve SMS message with the sim800 and send it to my gmail account.
The problem I have is that I cannot connect to the pppoe twice, I have changed the main to connect to call `esp_modem_setup_ppp` again after `esp_modem_exit_ppp` and it's look like the function never return.
I need help to figure this out and fix it.

Thank you,
Sagi

suda-morris, 2019-06-28:
Hi @Sagigamil To help reproduce the issue, could you please give us more detailed information, including the log, sdkconfig, code snippets?

Sagigamil, 2019-06-28:
> Hi @Sagigamil To help reproduce the issue, could you please give us more detailed information, including the log, sdkconfig, code snippets?

I updated the issue content, Hope it's clear.
Thank you,
Sagi

Sagigamil, 2019-06-30:
any help?

suda-morris, 2019-07-01:
@Sagigamil Sorry for the delay. IIRC, to reconnect a context, we should use the `ATO` command. The `esp_modem_setup_ppp` should be invoked on the first time sim800L setup a PPP context I think.
[image]

Sagigamil, 2019-07-02:
Any help?

Alvin1Zhang, 2019-10-11:
@Sagigamil Thanks for reporting, feel free to reopen if you still have the issue, thanks.

closed 2019-10-11 by Alvin1Zhang

fromeijn, 2019-10-28:
got the same problem. I want to really close the connection. after calling **esp_modem_exit_ppp** **esp_modem_setup_ppp** seems to freeze. It seems like the code stops on the **pppapi_pppos_create** call, but no idea why. As if the PPP_PHASE_ESTABLISH is never called the second time.