espressif/esp-idf #3698: Including <mutex> causes ctime "not been declared" errors

ssilverman, 2019-06-27:
Including `<mutex>` in a program causes the compiler to complain that a bunch of `ctime` things have not been declared. Specifically, the errors look like this:

```
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:64:11: error: '::clock' has not been declared
using ::clock;
^
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:65:11: error: '::difftime' has not been declared
using ::difftime;
^
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:66:11: error: '::mktime' has not been declared
using ::mktime;
^
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:67:11: error: '::time' has not been declared
using ::time;
^
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:68:11: error: '::asctime' has not been declared
using ::asctime;
^
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:69:11: error: '::ctime' has not been declared
using ::ctime;
^
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:70:11: error: '::gmtime' has not been declared
using ::gmtime;
^
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:71:11: error: '::localtime' has not been declared
using ::localtime;
^
.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/ctime:72:11: error: '::strftime' has not been declared
using ::strftime;
```

Maybe this has been fixed here? https://www.esp32.com/viewtopic.php?f=10&t=7400&p=31257#p41667
(I haven't figured out how to use this in my toolchain yet, so haven't tested.)

igrr, 2019-06-28:
Could you please include the information requested in the issue template? (IDF version, compiler version, build system, etc)

ssilverman, 2019-06-30:
Yes, sorry about that.

## Environment

* Development Kit: Adafruit ESP32 Feather
- Module or chip used: Adafruit ESP32 Feather
- IDF version: v1.9.0-4-g839ae90 (_PlatformIO version below_)
- Build System: PlatformIO
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 5.2.0
- Operating System: macOS
- Power Supply: USB via externally-powered USB 3.0 hub

PlatformIO version:
```
Platform Espressif 32 (Stage)
--------
Updating espressif32                     @ 839ae90        [Up-to-date]
Updating tool-esptoolpy                  @ 1.20600.0      [Up-to-date]
Updating toolchain-xtensa32              @ 2.50200.80     [Up-to-date]
Updating framework-arduinoespressif32    @ 7d78247        [Up-to-date]
Updating framework-espidf                @ 3.30202.190627 [Up-to-date]
Updating tool-mkspiffs                   @ 2.230.0        [Up-to-date]
```

igrr, 2019-07-01:
Sorry, I don't know how to relate PlatformIo versions to the versions in this repository. For example, which version of ESP-IDF corresponds to `v1.9.0-4-g839ae90` in PlatformIo terminology.
It would have been easier if the major.minor version tags were at least the same as in this repository...
Without some example which fails to compile with IDF build system (GNU Make based one or CMake based one) I'm afraid it's going to be hard to troubleshoot this.

ssilverman, 2019-07-01:
Fair point. I'll dig around with PlatformIO and find something.

ssilverman, 2019-07-10:
@igrr here's the version info, per PlatformIO's response:

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/featheresp32.html
PLATFORM: Espressif 32 1.9.0 > Adafruit ESP32 Feather
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10002.190628 (1.0.2), tool-esptoolpy 1.20600.0 (2.6.0)
```

igrr, 2019-07-10:
Thanks @ssilverman, this clarifies the version of the toolchain. I am still confused about IDF version since we don't have a 1.9.0 tag in this repository. Any chance the version of IDF you are using can be matched again one of the releases? (https://github.com/espressif/esp-idf/releases)

My speculation at the moment is that PlatformIo is using one of the released versions (so v3.2.x or v3.1.x), therefore there is no mismatch between the version of IDF and the version of the toolchain. This is good.

The next step would be to locate the full compiler command line, including all the flags, which leads to this error. Then we can compare it to the command line produced by the ESP-IDF build system, and see what the difference is.

Can you please check if you can get verbose output (including the full compiler command line) from PlatformIo build system?

ssilverman, 2019-07-11:
```
xtensa-esp32-elf-g++ -o .pio/build/featheresp32/src/main.cpp.o -c -fno-rtti -fno-exceptions -std=gnu++11 -Wall -fno-single-precision-constant -Os -Wall -nostdlib -Wpointer-arith -Wno-error=unused-but-set-variable -Wno-error=unused-variable -mlongcalls -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wno-error=deprecated-declarations -Wno-error=unused-function -Wno-unused-parameter -Wno-sign-compare -fstack-protector -fexceptions -DPLATFORMIO=40000 -DARDUINO_FEATHER_ESP32 -D_GLIBCXX_USE_C99 -DCORE_DEBUG_LEVEL=3 -DESP32 -DESP_PLATFORM -DF_CPU=240000000L -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DARDUINO=10805 -DARDUINO_ARCH_ESP32 -DARDUINO_VARIANT=\"feather_esp32\" "-DARDUINO_BOARD=\"Adafruit ESP32 Feather\"" -Iinclude -Isrc -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS/src -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/libraries/AsyncUDP/src -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/libraries/Preferences/src -I.pio/libdeps/featheresp32/libCBOR_ID2614/src -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/libraries/EEPROM/src -I.pio/libdeps/featheresp32/LiteOSCParser_ID2600/src -I.pio/libdeps/featheresp32/FastLED_ID126 "-I.pio/libdeps/featheresp32/ESP Async WebServer_ID306/src" -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/libraries/FS/src -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src -I.pio/libdeps/featheresp32/AsyncTCP_ID1826/src -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/config -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/app_trace -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/app_update -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/asio -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bootloader_support -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/bt -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/coap -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/console -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/driver -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp-tls -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp32 -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp_adc_cal -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp_event -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp_http_client -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp_http_server -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp_https_ota -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp_ringbuf -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/ethernet -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/expat -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/fatfs -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freemodbus -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/freertos -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/heap -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/idf_test -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/jsmn -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/json -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/libsodium -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/log -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/lwip -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/mbedtls -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/mdns -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/micro-ecc -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/mqtt -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/newlib -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/nghttp -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/nvs_flash -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/openssl -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/protobuf-c -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/protocomm -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/pthread -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/sdmmc -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/smartconfig_ack -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/soc -I/Users/shawn/.plat [... cut ...] 32/tools/sdk/include/xtensa-debug-module -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp32-camera -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/esp-face -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/fb_gfx -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/cores/esp32 -I/Users/shawn/.platformio/packages/framework-arduinoespressif32/variants/feather_esp32 src/main.cpp
```

valeros, 2019-07-11:
@ssilverman Are you sure you are using IDF framework? According to your logs, it looks like `Arduino` is selected as the framework in your project.

ssilverman, 2019-07-11:
Maybe I'm incorrectly assuming that this is an esp-idf issue. I thought this because that `mutex` include is part of the core. I am indeed using Arduino, and perhaps the Arduino core (arduino-esp32) is a better place to report this... Thanks for that suggestion. :)

igrr, 2019-07-11:
It might indeed be an Arduino-esp32 issue related to the include paths. For example, when building with IDF, the following include paths for the newlib component are added:
```
-I/idf/components/newlib/platform_include
-I/idf/components/newlib/include
```

However in the command line @ssilverman has attached, there only one newlib include directory:
```
-I/<paths>/tools/sdk/include/newlib
```

Looking at the [tools/sdk/include/newlib directory](https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/include/newlib) it contains the header files from https://github.com/espressif/esp-idf/tree/release/v3.2/components/newlib/include, but the header files from https://github.com/espressif/esp-idf/tree/release/v3.2/components/newlib/platform_include are missing. The latter files act as overrides for some of the newlib headers, and include the original files via `#include_next`, adding or correcting necessary defines.

ssilverman, 2019-08-01:
Turns out that I had `Time.h` and `Time.cpp` files in my root source directory. I tried both renaming these and moving them to a subdirectory without renaming, and both approaches solved the problem.

closed 2019-08-01 by ssilverman