espressif/esp-idf #3699: [SPI Master] the spi_transaction_t structure problem

imliubo, 2019-06-28:
## Environment 
- Development Kit:      None
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v3.0.8
- Build System:         Make
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5
- Operating System:     Windows
- Power Supply:         external 3.3V

## Problem Description

```C
size_t length;                  ///< Total data length, in bits
size_t rxlength;                ///< Total data length received, should be not greater than ``length`` in full-duplex mode (0 defaults this to the value of ``length``).
```

I don't quite understand the relationship between these two length. The "rxlength" I think it is the data I receviced lebgth,but the "length" is?Why the "rxlength" should be not greater than "length"?
E.g: I need ask the SPI slave device's data,so I send a command data package,it's just few bytes,But the data response from SPI slave is more than the command data package length.So I set the "rxlength" big.

Then I got this warning~
```bash
spi_master: spi_device_queue_trans(658): rx length > tx length in full duplex mode
```

So I am a bit confused, is it that I understand it incorrectly?
Thanks!

ginkgm, 2019-06-28:
Hi @imliubo ,

It's a hardware behavior that in the fullduplex mode, the data phase length is determined by the out side, in in length is only used to stop receiving more data than expected.

Please set the length to MAX(out_len, in_len) and set rxlength to in_len.

imliubo, 2019-06-28:
Hi @ginkgm ,
    Got it,thanks for help me.
    But I also have some problem with the spi master commuication to the spi slave device.
    I have a IR camera device and I need take data from it use SPI.I work this for few days,but it didn't work well.
    The datasheet description like below picture:
[image]
    And the test protocol package like below picture:
[image]
So I send the test protocol package use **spi_device_transmit()** function.

```C
    uint8_t TEST_PACKAGE[9]={0x02,0x00,0x04,0x00,0x01,0x00,0x00,0x03,0x05};
    uint8_t rxdata[9]={0};
    

    spi_transaction_t transaction;
    memset(&transaction, 0, sizeof(transaction));

    transaction.cmd = 0;
    transaction.addr = 0;
    transaction.length = 9* 8;
    transaction.rxlength = 9 * 8;
    transaction.user = NULL;
    transaction.tx_buffer = TEST_PACKAGE;
    transaction.rx_buffer = rxdata;
    esp_err_t err = spi_device_transmit(handle, &transaction);
```
The SPI device init just like below:
```C
    #define PIN_NUM_MISO 12
    #define PIN_NUM_MOSI 13
    #define PIN_NUM_CLK  14
    #define PIN_NUM_CS   15

    spi_device_handle_t handle;
    spi_bus_config_t buscfg={
        .miso_io_num=PIN_NUM_MISO,
        .mosi_io_num=PIN_NUM_MOSI,
        .sclk_io_num=PIN_NUM_CLK,
        .quadwp_io_num=-1,
        .quadhd_io_num=-1
    };
    spi_device_interface_config_t devcfg={
        .command_bits=0,
        .address_bits=0,
        .dummy_bits=0,                          //need flags=SPI_DEVICE_HALFDUPLEX
        .clock_speed_hz=10*1000*1000,           //Clock out at 10 MHz
        .mode=2,                                //SPI mode 2
        .spics_io_num=PIN_NUM_CS,               //CS pin
        .queue_size=1,                          //We want to be able to queue 7 transactions at a time
    };

    //Initialize the SPI bus and add the device we want to send stuff to.
    ret=spi_bus_initialize(HSPI_HOST, &buscfg, 0);
    assert(ret==ESP_OK);
    ret=spi_bus_add_device(HSPI_HOST, &devcfg, &handle);
    assert(ret==ESP_OK);
```

In theory, can I get the data I need from **rxdata**?

Thank you again for your help.
IAMLIUBO

ginkgm, 2019-06-28:
This device is not ideal: AFAIK, ESP32 doesn't support this kind of protocol. Each transaction is start from a CS(your SS) pulling low and ends with a CS pulling high.

For your case, the throughput will be quite low, since you need to send byte-by-byte, between the bytes, you need to control the GPIO manually (by the SW). set the cs gpio num to -1 in the spi driver, and manually control the GPIO by the gpio driver.

Set length=8 (1byte), rxlength=8, flags to USE_TX_DATA |USE_RX_DATA, and  send /receive data by the data buf inside the descriptor.

If you still want to do, strongly recommend to use the polling mode, which keeps the CPU polling until all bytes are sent and received. See https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#polling-transactions. 

BTW, if you are using the interrupt mode, you can use pretrans_cb and posttrans_cb to control the GPIO before and after a byte is sent.

ginkgm, 2019-06-28:
wrt your code, I can't find any problem with it, except the CS issue I mentioned above.

You should be able to read all 9 bytes from the rxdata.

Maybe you need a Logic Analyzer to help you, and you will see the reading from the driver is exactly same as on the bus. It's incorrect just because the CS requirement of your device is a bit wired.

imliubo, 2019-06-29:
Hi @ginkgm ,
   Thanks for help me. I tested it again according to the method you said.Unfortunately the problem still exists.But I use STM32 test it ,everything is OK.Please check if there is a problem with the logic of the code I am writing.
```C
#define PIN_NUM_MISO 12
#define PIN_NUM_MOSI 13
#define PIN_NUM_CLK  14

#define PIN_NUM_CS   15
#define PIN_NUM_RST  32
#define GPIO_OUTPUT_PIN_SEL  ((1ULL<<PIN_NUM_CS) | (1ULL<<PIN_NUM_RST))

//                     |<-               COMMAND DATA             ->|<-             DUMMY DATA            ->|-<           JUST FOR RECV DATA           ->|
uint8_t Command_buf[26]={0x02,0x00,0x04,0x00,0x01,0x00,0x00,0x03,0x05,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00};
uint8_t Data_buf[26]={0};

esp_err_t spi_write_bytes(spi_device_handle_t handle, const uint8_t *txdata, size_t txlength){

    if(txlength == 0 ) return ESP_ERR_INVALID_SIZE;

    esp_err_t ret=ESP_OK;

    spi_transaction_t transaction;
    memset(&transaction, 0, sizeof(transaction));

    transaction.flags = SPI_TRANS_USE_TXDATA | SPI_TRANS_USE_RXDATA; //Can write like this?
    transaction.cmd = 0;
    transaction.addr = 0;
    transaction.length = 8;                         // 1 byte
    transaction.rxlength = 8;                       // 1 byte

    //printf("\nsend data: ");
    for (int i = 0; i < txlength; ++i)              //txlength = 26,Command buf
    {   
        transaction.tx_data[0] = Command_buf[i];
        //printf(" %#x", transaction.tx_data[0]);   //just for log the data of send!
        gpio_set_level(PIN_NUM_CS,0);               // PULL DOWN the SS signal pin in manual.
        ret=spi_device_polling_transmit(handle, &transaction);
        gpio_set_level(PIN_NUM_CS,1);               // PULL UP the SS signal pin in manual.
        Data_buf[i] = transaction.rx_data[0];       //storage the data from SPI slave device. Am I right?
        assert(ret == ESP_OK); 
    }

    printf("\nrecv data: ");                        //log the recevice data.
    for (int i = 17; i < 26; ++i)
    {
        printf(" %#x", Data_buf[i]);
    }

    return ret;
}

void app_main()
{
    esp_err_t ret;
    spi_device_handle_t handle;
    spi_bus_config_t buscfg={                   
        .miso_io_num=PIN_NUM_MISO,
        .mosi_io_num=PIN_NUM_MOSI,
        .sclk_io_num=PIN_NUM_CLK,
        .quadwp_io_num=-1,
        .quadhd_io_num=-1
    };
    spi_device_interface_config_t devcfg={
        .command_bits=0,
        .address_bits=0,
        .dummy_bits=0,                          //need flags=SPI_DEVICE_HALFDUPLEX
        .clock_speed_hz=10*1000*1000,           //Clock out at 10 MHz
        .mode=2,                                //SPI mode 2
        .spics_io_num=-1,                       //CS pin
        .queue_size=1,                          //We want to be able to queue 7 transactions at a time
    };

    gpio_config_t io_conf;
    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;
    io_conf.mode = GPIO_MODE_OUTPUT;
    io_conf.pin_bit_mask = GPIO_OUTPUT_PIN_SEL;
    io_conf.pull_down_en = 0;
    io_conf.pull_up_en = 1;
    gpio_config(&io_conf);

    gpio_set_level(PIN_NUM_RST, 0);   //RESET the TW8035
    vTaskDelay(100 / portTICK_RATE_MS);
    gpio_set_level(PIN_NUM_RST, 1);

    //Initialize the SPI bus and add the device we want to send stuff to.
    ret=spi_bus_initialize(HSPI_HOST, &buscfg, 0);
    assert(ret==ESP_OK);
    ret=spi_bus_add_device(HSPI_HOST, &devcfg, &handle);
    assert(ret==ESP_OK);

    while(1) {
        vTaskDelay(1000 / portTICK_RATE_MS);

        spi_write_bytes(handle,Command_buf,26);
    }

    //Never reached.
    ret=spi_bus_remove_device(handle);
    assert(ret==ESP_OK);
}
```

**In order to make a better comparison, I also posted some of the code of STM32.**

```C
  while (1)
  {
		for(int i = 0; i < 29; i++){
			recv_buffer[i] = SPI1SendByte(sendbuf[i]);
		}
		printf("\r\nRecv: ");
		for(int i = 17; i < 26; i++){
			printf(" %#x",recv_buffer[i]);
		}
		HAL_Delay(100);
		
  }

uint8_t SPI1SendByte(uint8_t data)
{
	unsigned char writeCommand[1];
	unsigned char readValue[1];

	writeCommand[0] = data;
	HAL_GPIO_WritePin(SPI_SS_GPIO_Port,SPI_SS_Pin,GPIO_PIN_RESET);
	HAL_SPI_TransmitReceive(&hspi1, (uint8_t*)&writeCommand, (uint8_t*)&readValue, 1, 1);
	HAL_GPIO_WritePin(SPI_SS_GPIO_Port,SPI_SS_Pin,GPIO_PIN_SET);
	return readValue[0];
}
```
And the picture of the serial log.
**ESP32**
[image]
**STM32**
[image]

*For better comparison, I used to manually pull down or pull up the SS signal line.Is that OK?*

Thank you again for your help.I am looking forward to your relpy.
IAMLIUBO

imliubo, 2019-07-01:
Hi @ginkgm ,
I used the oscilloscope to look at the data and found that the data sent by ESP32 is very different from that sent by STM32.Why is this so, is the method I tested wrong or the code I wrote has a problem.
**STM32**
[image]
[image]

**ESP32**
[image]
[image]

It can be seen that the data waveform is completely different.

Thanks,
IAMLIUBO

ginkgm, 2019-07-01:
Maybe you should gpio_set_level(cs,1) when you initialize it.

And, the SPI is a 4-line synchronized protocol, what matters is the relationship between the clock and the data (also related with the CS). Suggest use the LA to check the timing among 4 signals. It looks good only from the MOSI side.

imliubo, 2019-07-02:
Hi @ginkgm ,
Thanks!This problem is solved.When I tried various methods, it didn't work. I realized that it might be a problem with this test PCB board.So I use the ESP32_DevKitc_V4 replace the test PCB board.Then it finally work well.
**Before I changed the board:**
[image]
[image]
**After I changed the board:**
[image]
[image]

Finally found the cause of the problem, I need to communicate with the hardware engineer.Obviously I ignored the hardware problem.

Thank you for your help these days.
Have a nice day!
IAMLIUBO

ginkgm, 2019-07-03:
@imliubo you are welcome :)

closed 2019-07-03 by ginkgm