espressif/esp-idf #3705: unknown reference to "hwcrypto/aes.h"

SunilDullolli, 2019-06-28:
I see that latest master code doesn't reference this code file but still while compiling, I get this error:
[image]

Am I missing something?

ghemal, 2019-07-01:
@SunilDullolli Would you please provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)?

SunilDullolli, 2019-07-01:
@ghemal I am trying to build latest ESP-IDF and facing this problem. 

**Environment**
Module or chip used: ESP32-WROOM-32
IDF version :  v4.0-dev-1018-gbbafd5066
Build System: Make
Compiler version :  (crosstool-NG esp32-2019r1) 8.2.0
Operating System: [Windows]
Power Supply:  external 3.3V

**Problem Description**
library is trying to access a header file ("hwcrytpto/aes.h") which is not referenced in the latest ESP-IDF and build fails. (though I can see that it actually references "esp32/aes.h" which is present).

**Expected Behavior**
Should compile without any issues.

**Actual Behavior**
build fails

**Steps to reproduce**
git clone the latest master branch and build it

**Debug Logs**
CC build/esp-tls/esp_tls.o
In file included from C:/Users/dullollis/Documents/msys32/home/dullollis/esp/esp-idf/components/mbedtls/mbedtls/include/mbedtls/aes.h:115,
                 from C:/Users/dullollis/Documents/msys32/home/dullollis/esp/esp-idf/components/mbedtls/mbedtls/include/mbedtls/ctr_drbg.h:45,
                 from C:/Users/dullollis/Documents/msys32/home/dullollis/esp/esp-idf/components/esp-tls/esp_tls.h:26,
                 from C:/Users/dullollis/Documents/msys32/home/dullollis/esp/esp-idf/components/esp-tls/esp_tls.c:24:
C:/users/dullollis/workspace/IOT/components/mbedtls/port/include/aes_alt.h:31:10: fatal error: hwcrypto/aes.h: No such file or directory
 #include "hwcrypto/aes.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [/home/dullollis/esp/esp-idf/make/component_wrapper.mk:290: esp_tls.o] Error 1
make: *** [C:/Users/dullollis/Documents/msys32/home/dullollis/esp/esp-idf/make/project.mk:582: component-esp-tls-build] Error 2

Any idea?

ghemal, 2019-07-02:
Thanks for the details however build/debug log seems incomplete. Was it compiling for you previously with any specific commit id?

Please try to run git submodule update --init --recursive from your IDF_PATH directory. Another suggestion is to remove your build directory and do fresh build again

I pulled the latest master and enabled AES hardware acceleration (make menuconfig -> component config -> mbedtls -> Enable AES acceleration). I was able to compile mbedtls particularly like below

cd $IDF_PATH/tools/unit-test-app
make -j4 TEST_COMPONENTS=mbedtls

Please provide how you are building & complete build log and also your sdkconfig file

SunilDullolli, 2019-07-02:
@ghemal Thanks for the reply. I did see that it was conflicting with the older version of mbedtls component and replacing that fixed the issue. Thanks

closed 2019-07-02 by SunilDullolli