espressif/esp-idf #3706: AES software decrypt broken

chegewara, 2019-06-29:
I am running my [aes-test example](https://github.com/chegewara/esp32-mbedtls-aes-test/blob/master/main/main.c) on master branch (today pulled) and i found issue with AES hardware acceleration disabled in menuconfig. Encryption seems to work for all cipher text types, but decryption is not working for few. Here is log with and without AES hardware enabled:
(1st line is encoded buffer, 2nd line decoded buffer, 3rd line plain decoded text)

without AES hardware disabled:
```
I (661) ecb: ff 6f a1 f9 cb 46 4e 89 29 52 0a e7 6c 40 b3 1d
I (671) ecb: c6 7f 54 10 37 eb ba 67 80 84 b5 e1 63 de d9 be
I (671) ecb: T7gcپ╔╔
I (701) cbc: a6 5a bc 09 3d a8 b9 f3 3d eb 69 e3 30 7d 80 f4
I (701) cbc: 51 e6 20 3f b0 9a 5c e3 9d 35 47 8a 35 72 b4 f8
I (711) cbc: Q ?\5G5r╔╔
I (721) cfb8: 40 ce aa a9 db e0 21 e8 fc 96 fc aa c5 7c d7 90
I (721) cfb8: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (731) cfb8: test string
I (731) cfb128: 40 5d 94 9a 20 f5 58 98 d0 1a b1 79 c9 4b 9e da
I (741) cfb128: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (751) cfb128: test string
I (751) ctr: 55 49 5c 10 64 22 28 d0 ac 5b e3 84 44 6b 6c 47
I (761) ctr: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (761) ctr: test string
I (771) xts: 6b f0 d6 78 14 d3 12 6e 30 12 c4 3c 94 4c 9f 56
I (771) xts: 14 02 4b e5 28 df 29 68 28 33 e5 35 9c 65 29 5a
I (781) xts: ╗K()h(35e)Z
```
with AES hardware enabled:
```
I (658) ecb: ff 6f a1 f9 cb 46 4e 89 29 52 0a e7 6c 40 b3 1d
I (668) ecb: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (668) ecb: test string
I (698) cbc: a6 5a bc 09 3d a8 b9 f3 3d eb 69 e3 30 7d 80 f4
I (698) cbc: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (708) cbc: test string
I (708) cfb8: 40 ce aa a9 db e0 21 e8 fc 96 fc aa c5 7c d7 90
I (718) cfb8: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (728) cfb8: test string
I (728) cfb128: 40 5d 94 9a 20 f5 58 98 d0 1a b1 79 c9 4b 9e da
I (738) cfb128: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (738) cfb128: test string
I (748) ctr: 55 49 5c 10 64 22 28 d0 ac 5b e3 84 44 6b 6c 47
I (748) ctr: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (758) ctr: test string
I (758) xts: 6b f0 d6 78 14 d3 12 6e 30 12 c4 3c 94 4c 9f 56
I (768) xts: 74 65 73 74 20 73 74 72 69 6e 67 00 00 00 00 00
I (778) xts: test string
```
Platform: `windows`
board: `m5stack`
commit: https://github.com/espressif/esp-idf/commit/bbafd506624b2a3d1e58f2dd3154286b79a8632b
gcc: `gcc version 8.2.0 (crosstool-NG esp32-2019r1)`

PerMalmberg, 2019-06-29:
@chegewara For the sake of clarity:

The sentence

> "without AES hardware disabled"

really means the same thing as

> with AES hardware enabled

I assume you meant to say "with AES hardware disabled" in the first section?

Are you running with PSRAM enabled for malloc? If so, can you try to disable it? I'm thinking you might be affected by #2892

chegewara, 2019-06-29:
> I assume you meant to say "with AES hardware disabled" in the first section?

Yes, thats correct.

> Are you running with PSRAM enabled for malloc? If so, can you try to disable it? I'm thinking you might be affected by #2892

No, PSRAM is not enabled, also m5stack does not have PSRAM.

PerMalmberg, 2019-06-29:
> No, PSRAM is not enabled, also m5stack does not have PSRAM.

Ok.

I ran your code on a ESP32 Wrover (devkit 3.x) with the same results;
AES with h/w disabled: failure.
AES h/w enabled : success.

`v4.0-dev-1018-gbbafd5066`
`xtensa-esp32-elf-g++ (crosstool-NG esp32-2019r1) 8.2.0`

chegewara, 2019-06-29:
The weird part is that unity test included in esp-idf mbedtls pass all tests.

ghemal, 2019-07-01:
@chegewara For pure software crypto operation in mbedtls AES there are two separate APIs to set the encryption and decryption key. In your case I see that you only call mbedtls_aes_setkey_enc() & mbedtls_aes_xts_setkey_enc(). The AES decryption key must be set using mbedtls_aes_setkey_dec() & mbedtls_aes_xts_setkey_dec(). Could you set the AES decryption key with these missing API and try?

chegewara, 2019-07-01:
Thanks @ghemal 
I remember i was confused few months ago when ive been playing with this code and all was working, even with decoding where mbedtls exclusively mentioned about using different function for encoding and decoding key initialization. Now i understand why, only weird/inconsistent is that with hardware decoding it does not matter.

closed 2019-07-01 by chegewara