espressif/esp-idf #3711: I can't build esp-idf/examples/ethernet/iperf

nopnop2002, 2019-06-30:
#3401  Environment

	freeRTOS version:V8.2.0
	NEWLIB version:2.2.0
	lwIP version:2-0-3-255
	ESP-IDF version:v4.0-dev-1018-gbbafd5066

## Problem Description

I can't build esp-idf/examples/ethernet/iperf 

### Steps to repropduce

My config is:

[image]
[image]
[image]

	/home/nop/esp-idf/examples/ethernet/iperf/main/cmd_ethernet.c: In function 'phy_device_power_enable_via_gpio':
	/home/nop/esp-idf/examples/ethernet/iperf/main/cmd_ethernet.c:52:5: error: implicit declaration of function 'gpio_pad_select_gpio' [-Werror=implicit-function-declaration]
	     gpio_pad_select_gpio(PIN_PHY_POWER);
	     ^
	/home/nop/esp-idf/examples/ethernet/iperf/main/cmd_ethernet.c:53:5: error: implicit declaration of function 'gpio_set_direction' [-Werror=implicit-function-declaration]
	     gpio_set_direction(PIN_PHY_POWER, GPIO_MODE_OUTPUT);
	     ^
	/home/nop/esp-idf/examples/ethernet/iperf/main/cmd_ethernet.c:53:39: error: 'GPIO_MODE_OUTPUT' undeclared (first use in this function)
	     gpio_set_direction(PIN_PHY_POWER, GPIO_MODE_OUTPUT);
	                                       ^
	/home/nop/esp-idf/examples/ethernet/iperf/main/cmd_ethernet.c:53:39: note: each undeclared identifier is reported only once for each function it appears in
	/home/nop/esp-idf/examples/ethernet/iperf/main/cmd_ethernet.c:55:9: error: implicit declaration of function 'gpio_set_level' [-Werror=implicit-function-declaration]
	         gpio_set_level(PIN_PHY_POWER, 1);
	         ^
	/home/nop/esp-idf/examples/ethernet/iperf/main/cmd_ethernet.c: At top level:
	cc1: warning: unrecognized command line option '-Wno-frame-address'
	cc1: some warnings being treated as errors
	/home/nop/esp-idf/make/component_wrapper.mk:289: recipe for target 'cmd_ethernet.o' failed
	make[1]: *** [cmd_ethernet.o] Error 1
	/home/nop/esp-idf/make/project.mk:582: recipe for target 'component-main-build' failed
	make: *** [component-main-build] Error 2

suda-morris, 2019-07-01:
Thanks @nopnop2002 for your effort in the example, we will look into it ASAP.
BTW, the Ethernet driver has been refactored these days, and the build error should be fixed there.

nopnop2002, 2019-07-01:
Hello.
I canged .max_cmdline_args = 16.
It's OK.
When you fixed example code, please close this isusues.
Thank you for your support.

closed 2019-07-05 by igrr