espressif/esp-idf #3712: Cannot build hello_world example

LaG1924, 2019-06-30:
CMake build failed because of `CMake Error: Could not open file for write in copy operation C:/Users/LaG1924/esp/hello_world/build/esp-idf/bootloader/C:/Users/LaG1924/esp/esp-idf/components/bootloader/flash_bootloader_args.in.in2.tmp`

I have followed `Get Started (CMake)` tutorial and stopped at step 7.
I tried to manually change owner of directory `%userprofile%\esp` but that didn't help.

IDF version is `v4.0-dev-1018-gbbafd5066`
Toolchain version is `(crosstool-NG crosstool-ng-1.22.0-80-g6c4433a5) 5.2.0`

Detailed build log is here:
```
PS C:\Users\LaG1924\esp\hello_world> idf.py menuconfig
Checking Python dependencies...
Python requirements from C:\Users\LaG1924\esp\esp-idf\requirements.txt are satisfied.
Executing action: menuconfig
Running cmake in directory c:\users\lag1924\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 --warn-uninitialized c:\users\lag1924\esp\hello_world"...
Warn about uninitialized values.
-- Found Git: C:/Program Files/Git/cmd/git.exe (found version "2.17.1.windows.2")
-- IDF_TARGET not set, using default target: esp32
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe
-- Check for working C compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe
-- Check for working CXX compiler: C:/Program Files/Espressif/ESP-IDF Tools/tools/bin/xtensa-esp32-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository,                         will not use 'git describe' to determine PROJECT_VER.
-- Project version: 1
-- Building ESP-IDF components for target esp32
CMake Warning at C:/Users/LaG1924/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:8 (message):
  Xtensa toolchain C:/Program Files/Espressif/ESP-IDF
  Tools/tools/bin/xtensa-esp32-elf-gcc.exe version 5.2.0 is not the supported
  version 8.2.0.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  C:/Users/LaG1924/esp/esp-idf/components/esp32/project_include.cmake:20 (gcc_version_check)
  C:/Users/LaG1924/esp/esp-idf/tools/cmake/build.cmake:287 (include)
  C:/Users/LaG1924/esp/esp-idf/tools/cmake/build.cmake:440 (__build_process_project_includes)
  C:/Users/LaG1924/esp/esp-idf/tools/cmake/project.cmake:315 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Warning at C:/Users/LaG1924/esp/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain C:/Program Files/Espressif/ESP-IDF
  Tools/tools/bin/xtensa-esp32-elf-gcc.exe crosstool-ng version
  crosstool-ng-1.22.0-80-g6c4433a5 doesn't match supported version
  esp32-2019r1.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  C:/Users/LaG1924/esp/esp-idf/components/esp32/project_include.cmake:21 (crosstool_version_check)
  C:/Users/LaG1924/esp/esp-idf/tools/cmake/build.cmake:287 (include)
  C:/Users/LaG1924/esp/esp-idf/tools/cmake/build.cmake:440 (__build_process_project_includes)
  C:/Users/LaG1924/esp/esp-idf/tools/cmake/project.cmake:315 (idf_build_process)
  CMakeLists.txt:6 (project)

ᮥ������� ᮧ���� ��� C:\Users\LaG1924\esp\hello_world\build\esp-idf\mbedtls\mbedtls\include\mbedtls <<===>> C:\Users\LaG1924\esp\esp-idf\components\mbedtls\mbedtls\include\mbedtls
ᮥ������� ᮧ���� ��� C:\Users\LaG1924\esp\hello_world\build\esp-idf\mbedtls\mbedtls\scripts <<===>> C:\Users\LaG1924\esp\esp-idf\components\mbedtls\mbedtls\scripts
-- Found PythonInterp: C:/Python27/python.exe (found version "2.7.14")
-- Could NOT find Perl (missing: PERL_EXECUTABLE)
-- Adding linker script C:/Users/LaG1924/esp/hello_world/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script C:/Users/LaG1924/esp/esp-idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script C:/Users/LaG1924/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/Users/LaG1924/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/Users/LaG1924/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/Users/LaG1924/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/Users/LaG1924/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/Users/LaG1924/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/Users/LaG1924/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-locale.ld
CMake Error: Could not open file for write in copy operation C:/Users/LaG1924/esp/hello_world/build/esp-idf/bootloader/C:/Users/LaG1924/esp/esp-idf/components/bootloader/flash_bootloader_args.in.in2.tmp
CMake Error: : System Error: Invalid argument
CMake Error at C:/Users/LaG1924/esp/esp-idf/components/esptool_py/project_include.cmake:187 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  C:/Users/LaG1924/esp/esp-idf/components/bootloader/CMakeLists.txt:10 (esptool_py_flash_project_args)

-- Components: app_trace app_update asio bootloader bootloader_support bt coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_event esp_http_client esp_http_server esp_https_ota esp_https_server esp_ringbuf esp_rom esp_websocket_client esp_wifi espcoredump esptool_py ethernet expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nimble nvs_flash openssl partition_table protobuf-c protocomm pthread sdmmc smartconfig_ack soc spi_flash spiffs tcp_transport tcpip_adapter ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/Users/LaG1924/esp/esp-idf/components/app_trace C:/Users/LaG1924/esp/esp-idf/components/app_update C:/Users/LaG1924/esp/esp-idf/components/asio C:/Users/LaG1924/esp/esp-idf/components/bootloader C:/Users/LaG1924/esp/esp-idf/components/bootloader_support C:/Users/LaG1924/esp/esp-idf/components/bt C:/Users/LaG1924/esp/esp-idf/components/coap C:/Users/LaG1924/esp/esp-idf/components/console C:/Users/LaG1924/esp/esp-idf/components/cxx C:/Users/LaG1924/esp/esp-idf/components/driver C:/Users/LaG1924/esp/esp-idf/components/efuse C:/Users/LaG1924/esp/esp-idf/components/esp-tls C:/Users/LaG1924/esp/esp-idf/components/esp32 C:/Users/LaG1924/esp/esp-idf/components/esp_adc_cal C:/Users/LaG1924/esp/esp-idf/components/esp_common C:/Users/LaG1924/esp/esp-idf/components/esp_event C:/Users/LaG1924/esp/esp-idf/components/esp_http_client C:/Users/LaG1924/esp/esp-idf/components/esp_http_server C:/Users/LaG1924/esp/esp-idf/components/esp_https_ota C:/Users/LaG1924/esp/esp-idf/components/esp_https_server C:/Users/LaG1924/esp/esp-idf/components/esp_ringbuf C:/Users/LaG1924/esp/esp-idf/components/esp_rom C:/Users/LaG1924/esp/esp-idf/components/esp_websocket_client C:/Users/LaG1924/esp/esp-idf/components/esp_wifi C:/Users/LaG1924/esp/esp-idf/components/espcoredump C:/Users/LaG1924/esp/esp-idf/components/esptool_py C:/Users/LaG1924/esp/esp-idf/components/ethernet C:/Users/LaG1924/esp/esp-idf/components/expat C:/Users/LaG1924/esp/esp-idf/components/fatfs C:/Users/LaG1924/esp/esp-idf/components/freemodbus C:/Users/LaG1924/esp/esp-idf/components/freertos C:/Users/LaG1924/esp/esp-idf/components/heap C:/Users/LaG1924/esp/esp-idf/components/idf_test C:/Users/LaG1924/esp/esp-idf/components/jsmn C:/Users/LaG1924/esp/esp-idf/components/json C:/Users/LaG1924/esp/esp-idf/components/libsodium C:/Users/LaG1924/esp/esp-idf/components/log C:/Users/LaG1924/esp/esp-idf/components/lwip C:/Users/LaG1924/esp/hello_world/main C:/Users/LaG1924/esp/esp-idf/components/mbedtls C:/Users/LaG1924/esp/esp-idf/components/mdns C:/Users/LaG1924/esp/esp-idf/components/mqtt C:/Users/LaG1924/esp/esp-idf/components/newlib C:/Users/LaG1924/esp/esp-idf/components/nghttp C:/Users/LaG1924/esp/esp-idf/components/nimbl [... cut ...] ts/unity C:/Users/LaG1924/esp/esp-idf/components/vfs C:/Users/LaG1924/esp/esp-idf/components/wear_levelling C:/Users/LaG1924/esp/esp-idf/components/wifi_provisioning C:/Users/LaG1924/esp/esp-idf/components/wpa_supplicant C:/Users/LaG1924/esp/esp-idf/components/xtensa
-- Configuring incomplete, errors occurred!
See also "C:/Users/LaG1924/esp/hello_world/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/LaG1924/esp/hello_world/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1
```

renzbagaporo, 2019-06-30:
Duplicate of https://github.com/espressif/esp-idf/issues/3687

igrr, 2019-06-30:
Has been fixed in https://github.com/espressif/esp-idf/commit/83ba2e082cfbcd75674862ebcec395771d634b1b.

closed 2019-06-30 by igrr