espressif/esp-idf #3713: CMake build system: IDF components missing for third-party projects

huming2207, 2019-06-30:
## Environment

- Development Kit:      Homemade (based on ESP-WROVER)
- Kit version (for WroverKit/PicoKit/DevKitC): N/A
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1018-gbbafd5066
- Build System: CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0
- Operating System:  Linux, Manjaro 18.1
- Power Supply: USB

## Problem Description

Some of the third-party projects like `libwebsocket` includes ESP-IDF components' header files. It works with the old GNU Make build system but it doesn't work (couldn't find the header) for new CMake build system.

### Expected Behavior

For example, `libwebsocket` includes `nvs.h` in their source file: https://github.com/huming2207/libwebsockets/blob/feature/esp32-cmake/include/libwebsockets/lws-esp32.h#L64 

Based on my understanding, it should be compiled successfully if the user required that in `idf_component_register(REQUIRES nvs_flash mdns mbedtls lwip nghttp)`

### Actual Behavior

But for now, the CMake build system does not apply the required IDF component to the include directories, which will cause these errors:

```
[ 53%] Building C object esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/lib/core/alloc.c.obj
cd /home/hu/Projects/lws-test/build/esp-idf/main/lib/libwebsockets && ccache /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -I/home/hu/Projects/lws-test/build/config -I/home/hu/Projects/lws-test/main/lib/libwebsockets/include -I/home/hu/Projects/lws-test/main/lib/libwebsockets/plugins -I/home/hu/Projects/lws-test/build/esp-idf/main/lib/libwebsockets -I/home/hu/Projects/lws-test/main/lib/libwebsockets/lib -I/home/hu/Projects/lws-test/main/lib/libwebsockets/lib/tls/mbedtls/wrapper/include -I/home/hu/Projects/lws-test/main/lib/libwebsockets/lib/tls/mbedtls/wrapper/include/platform -I/home/hu/Projects/lws-test/main/lib/libwebsockets/lib/tls/mbedtls/wrapper/include/internal -I/home/hu/Projects/lws-test/main/lib/libwebsockets/lib/tls/mbedtls/wrapper/include/openssl -I/home/hu/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/hu/esp/esp-idf/components/mbedtls/mbedtls/include/mbedtls -I/home/hu/esp/esp-idf/components/mbedtls/port/include -I/home/hu/esp/esp-idf/components/mbedtls/port/include/mbedtls -I/home/hu/esp/esp-idf/components/newlib/platform_include -I/home/hu/esp/esp-idf/components/freertos/include -I/home/hu/esp/esp-idf/components/heap/include -I/home/hu/esp/esp-idf/components/log/include -I/home/hu/esp/esp-idf/components/soc/esp32/include -I/home/hu/esp/esp-idf/components/soc/include -I/home/hu/esp/esp-idf/components/esp_rom/include -I/home/hu/esp/esp-idf/components/esp_common/include -I/home/hu/esp/esp-idf/components/xtensa/include -I/home/hu/esp/esp-idf/components/xtensa/esp32/include -I/home/hu/esp/esp-idf/components/esp32/include -I/home/hu/esp/esp-idf/components/driver/include -I/home/hu/esp/esp-idf/components/esp_ringbuf/include -I/home/hu/esp/esp-idf/components/esp_event/include -I/home/hu/esp/esp-idf/components/tcpip_adapter/include -I/home/hu/esp/esp-idf/components/lwip/include/apps -I/home/hu/esp/esp-idf/components/lwip/include/apps/sntp -I/home/hu/esp/esp-idf/components/lwip/lwip/src/include -I/home/hu/esp/esp-idf/components/lwip/port/esp32/include -I/home/hu/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/hu/esp/esp-idf/components/vfs/include -I/home/hu/esp/esp-idf/components/esp_wifi/include -I/home/hu/esp/esp-idf/components/esp_wifi/esp32/include -I/home/hu/esp/esp-idf/components/ethernet/include -I/home/hu/esp/esp-idf/components/efuse/include -I/home/hu/esp/esp-idf/components/efuse/esp32/include -I/home/hu/esp/esp-idf/components/app_trace/include  -Wall -Wsign-compare -Wuninitialized -Werror   -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0-dev-1018-gbbafd5066\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -o CMakeFiles/websockets.dir/lib/core/alloc.c.obj   -c /home/hu/Projects/lws-test/main/lib/libwebsockets/lib/core/alloc.c
In file included from /home/hu/Projects/lws-test/main/lib/libwebsockets/include/libwebsockets.h:334,
                 from /home/hu/Projects/lws-test/main/lib/libwebsockets/lib/core/private.h:126,
                 from /home/hu/Projects/lws-test/main/lib/libwebsockets/lib/core/alloc.c:1:
/home/hu/Projects/lws-test/main/lib/libwebsockets/include/libwebsockets/lws-esp32.h:64:10: fatal error: nvs.h: No such file or directory
 #include "nvs.h"
          ^~~~~~~
compilation terminated.
make[2]: *** [esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/build.make:63: esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/lib/core/alloc.c.obj] Error 1
make[2]: Leaving directory '/home/hu/Projects/lws-test/build'
make[1]: *** [CMakeFiles/Makefile2:6866: esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/all] Error 2
make[1]: Leaving directory '/home/hu/Projects/lws-test/build'
make: *** [Makefile:152: all] Error 2

```

### Steps to reproduce

1. Clone my test project: https://github.com/huming2207/lws-test
2. `cd` into my project
3. `mkdir -p build`
4. `cmake ../ -G 'Unix Makefiles'`
5. `make VERBOSE=1`
6. Check the log

### Code to reproduce this issue

Ref: https://github.com/huming2207/lws-test

## Debug Logs

By `make VERBOSE=1`:

```
/usr/bin/cmake -S/home/hu/Projects/lws-test -B/home/hu/Projects/lws-test/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /home/hu/Projects/lws-test/build/CMakeFiles /home/hu/Projects/lws-test/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/home/hu/Projects/lws-test/build'
make -f esp-idf/esp32/CMakeFiles/esp32_linker_script.dir/build.make esp-idf/esp32/CMakeFiles/esp32_linker_script.dir/depend
make[2]: Entering directory '/home/hu/Projects/lws-test/build'
cd /home/hu/Projects/lws-test/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/hu/Projects/lws-test /home/hu/esp/esp-idf/components/esp32 /home/hu/Projects/lws-test/build /home/hu/Projects/lws-test/build/esp-idf/esp32 /home/hu/Projects/lws-test/build/esp-idf/esp32/CMakeFiles/esp32_linker_script.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/hu/Projects/lws-test/build'
make -f esp-idf/esp32/CMakeFiles/esp32_linker_script.dir/build.make esp-idf/esp32/CMakeFiles/esp32_linker_script.dir/build
[... 289 lines cut ...]
make[2]: Nothing to be done for 'esp-idf/esp_ringbuf/CMakeFiles/__idf_esp_ringbuf.dir/build'.
make[2]: Leaving directory '/home/hu/Projects/lws-test/build'
[ 53%] Built target __idf_esp_ringbuf
make -f esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/build.make esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/depend
make[2]: Entering directory '/home/hu/Projects/lws-test/build'
cd /home/hu/Projects/lws-test/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/hu/Projects/lws-test /home/hu/Projects/lws-test/main/lib/libwebsockets /home/hu/Projects/lws-test/build /home/hu/Projects/lws-test/build/esp-idf/main/lib/libwebsockets /home/hu/Projects/lws-test/build/esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/home/hu/Projects/lws-test/build'
make -f esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/build.make esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/build
make[2]: Entering directory '/home/hu/Projects/lws-test/build'
[ 53%] Building C object esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/lib/core/alloc.c.obj
cd /home/hu/Projects/lws-test/build/esp-idf/main/lib/libwebsockets && ccache /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -I/home/hu/Projects/lws-test/build/config -I/home/hu/Projects/lws-test/main/lib/libwebsockets/include -I/home/hu/Projects/lws-test/main/lib/libwebsockets/plugins -I/home/hu/Projects/lws-test/build/esp-idf/main/lib/libwebsockets -I/home/hu/Projects/lws-test/main/lib/libwebsockets/lib -I/home/hu/Projects/lws-test/main/lib/libwebsockets/lib/tls/mbedtls/wrapper/include -I/home/hu/Projects/lws-test/main/lib/libwebsockets/lib/tls/mbedtls/wrapper/include/platform -I/home/hu/Projects/lws-test/main/lib/libweb [... cut ...]
 #include "nvs.h"
          ^~~~~~~
compilation terminated.
make[2]: *** [esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/build.make:63: esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/lib/core/alloc.c.obj] Error 1
make[2]: Leaving directory '/home/hu/Projects/lws-test/build'
make[1]: *** [CMakeFiles/Makefile2:6866: esp-idf/main/lib/libwebsockets/CMakeFiles/websockets.dir/all] Error 2
make[1]: Leaving directory '/home/hu/Projects/lws-test/build'
make: *** [Makefile:152: all] Error 2

```

huming2207, 2019-06-30:
For more information, please refer to this PR: https://github.com/warmcat/libwebsockets/pull/1621

Sorry if I misunderstood or forgot something in the IDF programming guide wiki. But it seems to be a problem caused by the IDF's CMake build system, as it doesn't pass in the correct include directories to the sub project.

renzbagaporo, 2019-06-30:
Hi @huming2207,

First of all,  https://github.com/warmcat/libwebsockets/pull/1621/files#diff-af3b638bc2a3e6c650974192a53c7291R226 will be addressed by a fix in our branch. The issue tracking it is https://github.com/espressif/esp-idf/issues/3611.

So Make build works because all include directories known are passed to every source build. With CMake , on the other hand, we subscribe to the idea of keeping propagation of build specification only limited based on dependencies established with `target_link_libraries`.  The `REQUIRES`, `PRIV_REQUIRES` arguments only applies to the library built by IDF, not those added by a subproject. To potentially solve this problem, you have to link nvs_flash to websockets. First we need to query the internal library name of `nvs_flash`

```
idf_component_get_property(nvs_flash_lib nvs_flash COMPONENT_LIB)
```

The problem is that `target_link_libraries` is only able to link to targets created in the current directory (at least prior to [v3.13](https://cmake.org/cmake/help/v3.15/policy/CMP0079.html#policy:CMP0079) ). You can potentially work around this using `LINK_LIBRARIES` and `INTERFACE_LINK_LIBRARIES` property (whichever is applicable, or both).

```
set_property(TARGET websockets APPEND PROPERTY LINK_LIBRARIES ${nvs_flash_lib})
set_property(TARGET websockets APPEND PROPERTY INTERFACE_LINK_LIBRARIES ${nvs_flash_lib})
```

Let me know if this works for you.

huming2207, 2019-06-30:
Hello @renzbagaporo ,

Great thanks for your help! Now NVS issue is solved. But I still would like to ask you one more questions.

I'm adding a temporary workaround for libwebsocket's generated configuration header files. Please refer to: https://github.com/huming2207/lws-test/commit/b815f0ef8f13bce49f583eb3c5fcdc560bb7f08c

If I don't have the `../build/esp-idf/main/lib/libwebsockets` addition to `idf_component_register`, the compiler will show me this error

```
[751/882] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj 
ccache /home/hu/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../main/lib/libwebsockets/include -I/home/hu/esp/esp-idf/components/newlib/platform_include -I/home/hu/esp/esp-idf/components/freertos/include -I/home/hu/esp/esp-idf/components/heap/include -I/home/hu/esp/esp-idf/components/log/include -I/home/hu/esp/esp-idf/components/soc/esp32/include -I/home/hu/esp/esp-idf/components/soc/include -I/home/hu/esp/esp-idf/components/esp_rom/include -I/home/hu/esp/esp-idf/components/esp_common/include -I/home/hu/esp/esp-idf/components/xtensa/include -I/home/hu/esp/esp-idf/components/xtensa/esp32/include -I/home/hu/esp/esp-idf/components/esp32/include -I/home/hu/esp/esp-idf/components/driver/include -I/home/hu/esp/esp-idf/components/esp_ringbuf/include -I/home/hu/esp/esp-idf/components/esp_event/include -I/home/hu/esp/esp-idf/components/tcpip_adapter/include -I/home/hu/esp/esp-idf/components/lwip/include/apps -I/home/hu/esp/esp-idf/components/lwip/include/apps/sntp -I/home/hu/esp/esp-idf/components/lwip/lwip/src/include -I/home/hu/esp/esp-idf/components/lwip/port/esp32/include -I/home/hu/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/hu/esp/esp-idf/components/vfs/include -I/home/hu/esp/esp-idf/components/esp_wifi/include -I/home/hu/esp/esp-idf/components/esp_wifi/esp32/include -I/home/hu/esp/esp-idf/components/ethernet/include -I/home/hu/esp/esp-idf/components/efuse/include -I/home/hu/esp/esp-idf/components/efuse/esp32/include -I/home/hu/esp/esp-idf/components/app_trace/include -I/home/hu/esp/esp-idf/components/nvs_flash/include -I/home/hu/esp/esp-idf/components/spi_flash/include -I/home/hu/esp/esp-idf/components/mbedtls/port/include -I/home/hu/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/hu/esp/esp-idf/components/mdns/include -I/home/hu/esp/esp-idf/components/console/. -I/home/hu/esp/esp-idf/components/nghttp/port/include -I/home/hu/esp/esp-idf/components/nghttp/nghttp2/lib/includes -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.0-dev-1018-gbbafd5066\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj -MF esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj.d -o esp-idf/main/CMakeFiles/__idf_main.dir/hello_world_main.c.obj   -c ../main/hello_world_main.c
In file included from ../main/hello_world_main.c:14:
../main/lib/libwebsockets/include/libwebsockets.h:39:10: fatal error: lws_config.h: No such file or directory
 #include "lws_config.h"
          ^~~~~~~~~~~~~~
``` 

So I would like to know that is there any better way to solve this by adding something into libwebsocket's CMakeList instead of adding workarounds into the main project's CMakeFile?

Thanks again!
Jackson

renzbagaporo, 2019-07-01:
I see you're mostly using the directory-specific command `include_directories`? This could be solved by transitioning to using `target_include_directories` and by having the directories you need as public directories. Then, **https://github.com/huming2207/lws-test/commit/b815f0ef8f13bce49f583eb3c5fcdc560bb7f08c#diff-87ba3e8d21d5a545894ecb3a61cea663R14** should propagate it to other components that might need it in the build.

huming2207, 2019-07-01:
@renzbagaporo oh yea I forgot that. I will try again tonight.

huming2207, 2019-07-01:
Hi @renzbagaporo 

I think I'd choose to wrap the `libwebsocket` into a component later. So I will leave the `include_directories` in the `libwebsockets` CMake project as-is to prevent any incompatibility issues.

I will close this issue for now. If there is anything else need to fix, please re-open it. Thanks!

Regards,
Jackson

closed 2019-07-01 by huming2207