espressif/esp-idf #3715: No IMCP answers on ethernet iface(LAN8720)

no1seman, 2019-06-30:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1065-g39a5fbdfe
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
  xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:     Linux
- Power Supply:         USB|external 5V&Battery

## Problem Description
After commit d7d91225d3ac852f3df54dd6d844474af48488c9 can't get ICMP anwers on Ethernet iface (LAN8720), http server works fine and on WiFi face ICMP answers present.

### Expected Behavior
ICMP answers must be pesent
### Actual Behavior
No ICMP answers
### Steps to repropduce

Use default "esp-idf/examples/protocols/http_server/simple/" example to reproduce.

suda-morris, 2019-07-01:
Hi @no1seman 
Can you push your sdkconfig file because I can't reproduce your issue.
* How your LAN8720 is connected to ESP32?
* What's the RMII clock mode? 

Since the Ethernet component has been refactored in 4.0, there will be a a little different when doing menuconfig. I would suggest you to look at this [README](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/README.md) when you're using Ethernet.

no1seman, 2019-07-01:
@suda-morris,

Here is my sdkconfig: 
[sdkconfig.zip](https://github.com/espressif/esp-idf/files/3344303/sdkconfig.zip)

The problem seems to be be more complicated. I have a network 192.168.3.0/24 with common gateway for all devices: 192.168.3.1 and there is a several devices: laptops, android phones with different operating systems. So, I can get ping answers from esp32 with test app based on  refactored Ethernet component only from ubuntu (tested 3 different hosts) and from win10, win 8.1 macOS 10.14, android 9.0 I can't see any answers in console. Here is some logs:

ESP32:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6484
load:0x40078000,len:11004
load:0x40080400,len:6644
entry 0x4008076c
I (29) boot: ESP-IDF v4.0-dev-1071-g70474ae84 2nd stage bootloader
I (29) boot: compile time 09:30:24
I (38) boot: Enabling RNG early entropy source...
I (38) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 8MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0f798 ( 63384) map
I (116) esp_image: segment 1: paddr=0x0001f7c0 vaddr=0x3ffb0000 size=0x00850 (  2128) load
I (117) esp_image: segment 2: paddr=0x00020018 vaddr=0x400d0018 size=0x3711c (225564) map
0x400d0018: _stext at ??:?

I (203) esp_image: segment 3: paddr=0x0005713c vaddr=0x3ffb0850 size=0x01800 (  6144) load
I (206) esp_image: segment 4: paddr=0x00058944 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/user/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1778

I (210) esp_image: segment 5: paddr=0x00058d4c vaddr=0x40080400 size=0x09f20 ( 40736) load
I (242) boot: Loaded app from partition at offset 0x10000
I (242) boot: Disabling RNG early entropy source...
I (243) cpu_start: Pro cpu up.
I (246) cpu_start: Application information:
I (251) cpu_start: Project name:     simple
I (256) cpu_start: App version:      1
I (261) cpu_start: Compile time:     Jul  1 2019 09:30:22
I (267) cpu_start: ELF file SHA256:  20f1fd9c913f591c...
I (273) cpu_start: ESP-IDF:          v4.0-dev-1071-g70474ae84
I (279) cpu_start: Starting app cpu, entry point is 0x4008109c
0x4008109c: call_start_cpu1 at /home/user/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:279

I (0) cpu_start: App cpu up.
I (290) heap_init: Initializing. RAM available for dynamic allocation:
I (296) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (302) heap_init: At 3FFB40F0 len 0002BF10 (175 KiB): DRAM
I (309) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (315) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (321) heap_init: At 4008A320 len 00015CE0 (87 KiB): IRAM
I (328) cpu_start: Pro cpu start user code
I (346) spi_flash: detected chip: generic
I (347) spi_flash: flash io: dio
I (347) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (431) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (5881) example_connect: Ethernet Link Up
I (9921) tcpip_adapter: eth ip: 192.168.3.222, mask: 255.255.255.0, gw: 192.168.3.1
I (9921) example_connect: Connected to Ethernet
I (9921) example_connect: IPv4 address: 192.168.3.222
I (9921) example_connect: IPv6 address: fe80:0000:0000:0000:b6e6:2dff:feee:7e78
I (9931) example: Starting server on port: '80'
[... 136 lines cut ...]
  status: inactive
awdl0: flags=8943<UP,BROADCAST,RUNNING,PROMISC,SIMPLEX,MULTICAST> mtu 1484
  ether 8e:7e:48:a1:60:4c
  inet6 fe80::8c7e:48ff:fea1:604c%awdl0 prefixlen 64 scopeid 0x7
  nd6 options=201<PERFORMNUD,DAD>
  media: autoselect
  status: active
utun0: flags=8051<UP,POINTOPOINT,RUNNING,MULTICAST> mtu 2000
  inet6 fe80::8694:ea53:54a2:4e1c%utun0 prefixlen 64 scopeid 0x8
  nd6 options=201<PERFORMNUD,DAD>
MacBook-Wise:~ wiseman$ curl http://192.168.3.222/hello
Hello World!
MacBook-Wise:~ wiseman$ ping 192.168.3.222
PING 192.168.3.222 (192.168.3.222): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
^C
--- 192.168.3.222 ping statistics ---
5 packets transmitted, 0 packets received, 100.0% packet loss

On windows 10/8.1 and android 9.0 theres no answers same as on MacOS.

Before refactoring everything worked fine.

suda-morris, 2019-07-01:
Oh no, @no1seman thanks for reporting this bug, it's really a bug! I think it's due the checksum error. I test the basic ethenret example, and ping from a raspberrypi (sorry I don't have windows and macos), and I will get `64 bytes from 192.168.2.151: icmp_seq=1 ttl=255 time=0.530 ms (BAD CHECKSUM!)`. We will look into this issue and fix it ASAP.

suda-morris, 2019-07-01:
@no1seman Can you change [this line](https://github.com/espressif/esp-idf/blob/master/components/soc/esp32/emac_hal.c#L167) to `hal->tx_desc[i].TDES0.ChecksumInsertControl = EMAC_DMATXDESC_CHECKSUM_BYPASS;`  To see if this can solve the problem?

no1seman, 2019-07-01:
@suda-morris,
 yes, now ping works, but as I undestood that option turns off checksum checking for all types of protocols: TCP/UDP/ICMP? If so it's not a good idea to skip Ethernet frame checksum checking?

suda-morris, 2019-07-01:
> @suda-morris,
> yes, now ping works, but as I undestood that option turns off checksum checking for all types of protocols: TCP/UDP/ICMP? If so it's not a good idea to skip TCP/UDP/ICMP packet checksum checking?

Sorry, ESP32 doesn't support hardware checksum insertion on the TX path, but does check the checksum on the RX path. It's my mistake that I tried to enable a feature that doesn't support by ESP32 (and most of the MCU on the market doesn't have this feature too).

no1seman, 2019-07-01:
@suda-morris, Ok, thank you for quick solution, wish you will commit bugfix soon.

tomseago, 2019-07-04:
@suda-morris I can confirm everything @no1seman said. I'm running into the same bug and would love for the fix to be committed soon. Your suggestion about setting ChecksumInsertControl to EMAC_DMATXDESC_CHECKSUM_BYPASS fixes the issue for me as well. Does it help if I just create a PR for this directly?

suda-morris, 2019-07-04:
Thanks @tomseago :+1: 

The fix is will be pushed to github in the next few days.

closed 2019-07-05 by igrr