espressif/esp-idf #3716: ESP-IDF make flash conf problem

AviraMatti, 2019-07-01:
Hi

I have the problem to continue next situation.
Make modeconfig works but I gave next problems.
Forward is also PATH lines.  I have follow ESPRESSIF page installation for Ubuntu.
 
atti@matti-Latitude-E6420:~/esp/hello_world$ make flash
Toolchain path: /home/matti/esp/xtensa-esp32-elf/bin/sh/xtensa-esp32-elf-gcc
Toolchain version: esp32-2019r1
Compiler version: 8.2.0
Python requirements from /home/matti/esp/esp-idf/requirements.txt are satisfied.
CC build/bootloader/bootloader_support/src/bootloader_flash.o
xtensa-esp32-elf-gcc: error trying to exec 'cc1': execvp: No such file or directory
/home/matti/esp/esp-idf/make/component_wrapper.mk:289: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 1
/home/matti/esp/esp-idf/make/project.mk:582: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/home/matti/esp/esp-idf/components/bootloader/Makefile.projbuild:41: recipe for target '/home/matti/esp/hello_world/build/bootloader/bootloader.bin' failed
make: *** [/home/matti/esp/hello_world/build/bootloader/bootloader.bin] Error 2

PATH info.
I have putting and try quite many different possibility...
This environment don't generate any .o files preset folders...??

matti@matti-Latitude-E6420:~/esp/hello_world$ echo $PATH
:/home/matti/esp/esp-idf/components/esptool_py/esptool:/home/matti/esp/esp-idf/components/espcoredump:/home/matti/esp/esp-idf/components/partition_table/:/home/matti/esp/esp-idf/tools/:/home/matti/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/matti/xtensa-esp32-elf/xtensa-esp32-elf/bin/sh:/home/matti/snap/bin:/home/matti/home/matti/xtensa-esp32-elf/xtensa-esp32-elf/bin/sh:/home/matti/home/matti/bin/sh:/home/matti/bin/sh:/home/matti/esp/xtensa-esp32-elf/bin/sh:/home/matti/usr/bin/gcc:/home/matti/build/bootloader/bootloader_support/src:/home/matti/build/bootloader/bootloader_support/src:/home/matti/home/matti/esp/esp-idf:/home/matti/build/bootloader/bootloader_support/src:/home/matti/home/matti/esp/esp-idf/make:/home/matti/home/matti/esp/esp-idf/make:/home/matti/eso/esp32-ogn-tracker/build

Some reason cc1 don't find.

matti@matti-Latitude-E6420:~/esp/hello_world$ which gcc
/usr/bin/gcc
matti@matti-Latitude-E6420:~/esp/hello_world$ which cc1
matti@matti-Latitude-E6420:~/esp/hello_world$ 

BR
Matti

nopnop2002, 2019-07-01:
Hello.
I think that Your path is not current.

	Toolchain path: /home/matti/esp/xtensa-esp32-elf/bin/sh/xtensa-esp32-elf-gcc

It's mine.

	Toolchain path: /home/nop/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc

AviraMatti, 2019-07-02:
Thanks a lot nopnop2002....it seems it was path problem...I had errors in paths. System generate now right flashing files.
I use Minicom for /dev/ttyUSB0 setup and it looks like it locks that port....don't find anymore when connect ESP32 development board to it. This is quite different issue and I have to check it next...

Alvin1Zhang, 2019-07-03:
@AviraMatti Thanks for reporting the issue. Feel free to reopen if your issue happens again. Thanks.

closed 2019-07-03 by Alvin1Zhang