espressif/esp-idf #3718: LEFT_SHIFT_KEY_MASK modificator in ble_hid_demo project not working

OPyshkin, 2019-07-01:
I'm writing a converter from ascii to keyboard scan code using ble_hid library (https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_hid_device_demo) . To make difference between big and small letters I'm trying to use LEFT_SHIFT_KEY_MASK modificator, but the letters are still small. Here's the snippet from my code 
'''
len = uart_read_bytes(UART_NUM_1, data, BUF_SIZE, 20 / portTICK_RATE_MS);
        // Write data back to the UART
        
        uint8_t bytestosend[1];
        for (int i=0;i<len;i++)
        {
            bytestosend[0] = data[i]; 
            uint8_t key=0;
            key_mask_t modifier;
            if (data[i]>=97 && data[i]<=122) 
            {
                key = data[i]-93;
                modifier = 0;
            }
            else if (data[i]>=65 && data[i]<=90)
            {
                key = data[i]-61;
                modifier = LEFT_SHIFT_KEY_MASK;
            }
            else if (data[i]>=49 && data[i]<=57)
            {
                key = data[i]-19;
                modifier = 0;
            }
            else if (data[i]==48)
            {
                key = data[i]-9;
                modifier = 0;
            }
            
            
            
            esp_hidd_send_keyboard_value( hid_conn_id, modifier, &key, 1 ) ;
            
            esp_hidd_send_keyboard_value( hid_conn_id, modifier, &key, 0 ) ;
            uart_write_bytes(UART_NUM_1, (const char *) bytestosend, 1);
        }
'''

I also tried running an example from ble_hid_device_demo and it still gave no result

OPyshkin, 2019-07-02:
Finally, I found the solution. Instead of esp_hidd_send_keyboard_value( hid_conn_id, LEFT_SHIFT_KEY_MASK, &key, 1 ); I used esp_hidd_send_keyboard_value( hid_conn_id, 2, &key, 1 ); . I guess that will work with other modificators

closed 2019-07-02 by OPyshkin

dankokp, 2019-11-17:
Hey, have the same issue:
https://github.com/espressif/esp-idf/issues/4353