espressif/esp-idf #3719: ESP32 Classic Bluetooth - RFCOMM connection in state 1 closed: Peer connection failed (res: 16)

MarWoz442, 2019-07-01:
## Environment

-ESP32-PICOD4 - chip
- IDF version v3.2-dirty
- Compiler version xtensa-esp32-elf-gcc version 1.22.0-80-g6c4433a5
- Operating System: Windows
- Power Supply:  USB/Battery

## Problem Description

I try to connect to the other BT  (SPP) device by using bt_spp_initiator example. Unfortunately I get some error from BT stack.
When I calling esp_spp_connect then I got ESP_SPP_CL_INIT_EVT event then I receive error:

## Debug Logs
```
BT_RFCOMM: port_rfc_closed RFCOMM connection in state 1 closed: Peer connection failed (res: 16)
```

and then I got ESP_SPP_CLOSE_EVT witout any more warrings.

## Debug Logs
```
I (4301) SPP_INITIATOR_DEMO: ESP_SPP_CL_INIT_EVT
I (4301) SPP_INITIATOR_DEMO: 00 13 43 3f 4c b0
D (4311) BT_APPL: BTA got event 0x10d
D (4311) BT_APPL: bta_dm_sm_execute event:0xd
D (4551) BT_RFCOMM: rfc_mx_sm_state_wait_conn_cnf - evt:14
D (4551) BT_RFCOMM: PORT_CloseInd
D (4551) BT_RFCOMM: port_rfc_closed state:1 sending events:0
D (4551) BT_APPL: bta_jv_port_mgmt_cl_cback:code:16, port_handle2
D (4561) BT_APPL: bta_jv_port_mgmt_cl_cback code=16 port_handle:2 handle:129
I (4561) BT_RFCOMM: PORT_CheckConnection() handle:2
W (4571) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 1 closed: Peer co
nnection failed (res: 16)
D (4581) BT_RFCOMM: port_release_port, p_port:0x3ffc73a8
D (4581) BT_RFCOMM: rfc_port_closed DONE
D (4591) BT_RFCOMM: rfc_timer_start - timeout:2
D (4591) BT_RFCOMM: rfc_port_timer_stop
D (4601) BT_RFCOMM: port_release_port:p_port->keep_port_handle:0
D (4601) BT_RFCOMM: port_release_port:Clean-up handle:2
D (4611) BT_RFCOMM: rfc_port_timer_stop
D (4611) BT_RFCOMM: rfc_timer_free
D (4621) BT_APPL: BTA got event 0x104
D (4621) BT_APPL: bta_dm_sm_execute event:0x4
D (4621) BT_APPL: BTA got event 0x104
D (4631) BT_APPL: bta_dm_sm_execute event:0x4
I (4631) SPP_INITIATOR_DEMO: ESP_SPP_CLOSE_EVT
```

blueMoodBHD, 2019-07-03:
Hi, could you please provide more informations,
    1. The commit id of `esp-idf`.
    2. What is the other BT (SPP) device?
    3. How did you use the `bt_spp_initiator` example? Just changed the `remote_device_name`, or changed something else, either?

MarWoz442, 2019-07-03:
Hi, 
Thank you for your interest and replay.
add.1 286202caa
add.2 The device uses BT module mnf by Panasonic mod.no ENW89841A3KF, Bluetooth 2.1 + EDR with Secure Simple Paring
add.3 Yes, at first I changed only ```remote_device_name```

MarWoz442, 2019-08-28:
Hello after a long break ...
I had to finish other projects, but now the topic has returned ...
Unfortunately, the problem could not be resolved:
Below I attach logs.
I suspect that I should receive the ESP_BT_GAP_KEY_REQ_EVT or ESP_BT_GAP_CFM_REQ_EVT event, instead it will receive the ESP_SPP_CL_INIT_EVT event.
I set the IO capabilities as ESP_BT_IO_CAP_IO and pin_type = ESP_BT_PIN_TYPE_VARIABLE.
What should I call to receive a valid event?

## Debug Logs

```
 I (7362) bt_app_gap_cb: Services for device 00:13:43:xx:xx:xx found
I (7372) bt_app_gap_cb: --1101
I (7372) bt_app_gap_cb: --ffcacade-afde-cade-defa-cade00000000
I (7382) bt_app_gap_cb: --5aeaae0f-cc5a-2d5f-5aea-ae0fcc5a2d5f
I (7382) BT_APPL: BTA_JvStartDiscovery
D (7392) BT_APPL: BTA got event 0x1a04

D (7392) BT_APPL: bta_jv_start_discovery in, sdp_active:0
D (7402) BT_APPL: call SDP_InitDiscoveryDb, p_data->start_discovery.num_uuid:1
D (7542) BT_APPL: bta_jv_start_discovery_cback res: 0x0
D (7542) BT_APPL: bta_jv_cb.uuid: 00001101-0000-1000-8000-00805f9b34fb
D (7542) BT_APPL: in, uuid:: 00001101-0000-1000-8000-00805f9b34fb
D (7552) BT_APPL: uuid len:16
D (7552) BT_APPL: shorten to 16 bits uuid: 1101
D (7552) BT_APPL: shorten uuid:: 01110000-0000-0000-0000-000000000000
D (7562) BT_APPL: p_sdp_rec:0x3ffca908
D (7562) BT_APPL: p_sdp_rec:0x0
I (7572) esp_spp_cb: ESP_SPP_DISCOVERY_COMP_EVT bbb status=0 scn_num=1
I (7572) BT_APPL: BTA_JvRfcommConnect
D (7582) BT_APPL: BTA got event 0x1a0d

D (7582) BT_RFCOMM: port_find_mcb: not found, bd_addr:00:13:43:xx:xx:xx
I (7592) BT_RFCOMM: RFCOMM_CreateConnection()  BDA: 00-13-43-xx-xx-xx
I (7602) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:0 mtu:990, p_mcb:0x0
D (7612) BT_RFCOMM: rfc_cb.port.port[2]:0x3ffc74e8 allocated, last_port:2
D (7612) BT_RFCOMM: port_allocate_port:bd_addr:00:13:43:xx:xx:xx
I (7622) BT_RFCOMM: RFCOMM_CreateConnection(): scn:1, dlci:2, is_server:0 mtu:990, p_mcb:0x0, p_port:0x3ffc74e8
D (7632) BT_RFCOMM: RFCOMM_CreateConnection dlci:2 signal state:0xb
D (7642) BT_RFCOMM: port_open_continue, p_port:0x3ffc74e8
D (7642) BT_RFCOMM: rfc_alloc_multiplexer_channel: bd_addr:00:13:43:xx:xx:xx
D (7652) BT_RFCOMM: rfc_alloc_multiplexer_channel:is_initiator:1
D (7652) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[0].state:0
D (7662) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
D (7672) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[1].state:0
D (7682) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
D (7682) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[2].state:0
D (7692) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
D (7702) BT_RFCOMM: rfc_alloc_multiplexer_channel rfc_cb.port.rfc_mcb[3].state:0
D (7702) BT_RFCOMM: (rfc_cb.port.rfc_mcb[i].bd_addr:00:00:00:00:00:00
D (7712) BT_RFCOMM: rfc_timer_free
D (7712) BT_RFCOMM: rfc_alloc_multiplexer_channel:is_initiator:1, create new p_mcb:0x3ffc7de0, index:0
D (7722) BT_RFCOMM: rfc_timer_start - timeout:60
D (7732) BT_RFCOMM: port_select_mtu application selected 990
D (7732) BT_RFCOMM: port_select_mtu credit_rx_max 10, credit_rx_low 4, rx_buf_critical 15
D (7742) BT_RFCOMM: rfc_mx_sm_state_idle - evt:6
D (7752) BT_APPL: bta_jv_alloc_rfc_cb port_handle:3 handle:0x82
I (7752) BT_RFCOMM: PORT_SetEventCallback() handle:3
I (7762) BT_RFCOMM: PORT_SetEventMask() handle:3 mask:0x30005
I (7762) BT_RFCOMM: PORT_GetState() handle:3
I (7772) BT_RFCOMM: PORT_SetState() handle:3
I (7772) BT_RFCOMM: PORT_SetState() handle:3 FC_TYPE:0xc
I (7782) esp_spp_cb: ESP_SPP_CL_INIT_EVT
D (7812) BT_RFCOMM: rfc_mx_sm_state_wait_conn_cnf - evt:9
D (7812) BT_RFCOMM: PORT_StartCnf result:5
W (7812) BT_RFCOMM: PORT_StartCnf failed result:5
D (7822) BT_RFCOMM: rfc_timer_free
D (7822) BT_APPL: bta_jv_port_mgmt_cl_cback:code:12, port_handle3
D (7832) BT_APPL: bta_jv_port_mgmt_cl_cback code=12 port_handle:3 handle:130
I (7842) BT_RFCOMM: PORT_CheckConnection() handle:3
D (7842) BT_RFCOMM: port_release_port, p_port:0x3ffc74e8
D (7852) BT_RFCOMM: rfc_port_closed DONE
D (7852) BT_RFCOMM: rfc_port_timer_stop
D (7852) BT_RFCOMM: port_release_port:p_port->keep_port_handle:0
D (7862) BT_RFCOMM: port_release_port:Clean-up handle:3
D (7872) BT_RFCOMM: rfc_port_timer_stop
I (7872) esp_spp_cb: ESP_SPP_CLOSE_EVT
D (7882) BT_APPL: BTA got event 0x104

D (7882) BT_APPL: bta_dm_sm_execute event:0x4
D (7882) BT_APPL: BTA got event 0x104

D (7892) BT_APPL: bta_dm_sm_execute event:0x4

```

MarWoz442, 2019-08-31:
Hi, 

I analyzed the problem further.
The reason for the problem is the security procedure (L2CAP layer)
The function  
```cpp
void l2c_csm_execute function (tL2C_CCB * p_ccb, UINT16 event, void * p_data);
```
is called with the CST_ORIG_W4_SEC_COMP event (/ * Originator waits security clearence * /)
Then the function 
```cpp
void l2c_csm_orig_w4_sec_comp (tL2C_CCB * p_ccb, UINT16 event, void * p_data)
```
is called with the event L2CEVT_SEC_COMP_NEG (/ * Security procedure failed * / or / * something is really bad with security * /)

Logs below

## Debug Logs

```
I (6719) bt_app_gap_cb: --1101
I (6719) bt_app_gap_cb: --ffcacade-afde-cade-defa-cade00000000
I (6729) bt_app_gap_cb: --5aeaae0f-cc5a-2d5f-5aea-ae0fcc5a2d5f
I (6739) BT_L2CAP: L2CA_ErtmConnectReq()  PSM: 0x0001  BDA: 001343xxxxxx  p_ertm
_info: 0x0 allowed:0x0 preferred:0
D (6749) BT_L2CAP: l2cu_allocate_ccb: cid 0x0000
D (6749) BT_L2CAP: l2cu_enqueue_ccb CID: 0x0042  priority: 2
D (6759) BT_L2CAP: l2c_link_adjust_chnl_allocation
D (6759) BT_L2CAP: CID:0x0042 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Qu
ota:200
D (6769) BT_L2CAP: L2CAP - st: CLOSED evt: 21
D (6779) BT_L2CAP: l2c_link_sec_comp: 17, 0x3ffc4600
D (6779) BT_L2CAP: ccb timer ticks: 0
D (6779) BT_L2CAP: L2CAP - st: CLOSED evt: 7
D (6789) BT_L2CAP: l2c_link_check_send_pkts
D (6789) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (6799) BT_L2CAP: l2c_link_send_to_lower
D (6799) BT_L2CAP: TotalWin=8,Hndl=0x81,Quota=9,Unack=1,RRQuota=0,RRUnack=0
D (6809) BT_L2CAP: RR scan pri=2, lcid=0x0042, q_cout=0
I (6819) BT_L2CAP: L2CAP - L2CA_conn_req(psm: 0x0001) returned CID: 0x0042
D (6819) BT_L2CAP: l2c_link_check_send_pkts
D (6829) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (6829) BT_L2CAP: RR scan pri=2, lcid=0x0042, q_cout=0
D (6839) BT_L2CAP: TotalWin=9,LinkUnack(0x81)=0,RRCheck=0,RRUnack=0

D (6849) BT_L2CAP: L2CAP - rcv_cid CID: 0x0001

D (6849) BT_L2CAP: cmd_code: 3, id:9, cmd_len:8
D (6859) BT_L2CAP: L2CAP - st: W4_L2CAP_CON_RSP evt: 12
D (6859) BT_L2CAP: L2CAP - rcv_cid CID: 0x0001

D (6869) BT_L2CAP: cmd_code: 3, id:9, cmd_len:8
D (6869) BT_L2CAP: L2CAP - st: W4_L2CAP_CON_RSP evt: 11
I (6879) BT_L2CAP: L2CAP - Calling Connect_Cfm_Cb(), CID: 0x0042, Success
I (6879) BT_L2CAP: L2CA_ConfigReq()  CID 0x0042: fcr_present:0 (mode 0) mtu_pres
ent:1 (672)
D (6889) BT_L2CAP: L2CAP - st: CONFIG evt: 24
D (6899) BT_L2CAP: l2c_link_check_send_pkts
D (6899) BT_L2CAP: partial_segment_being_sent=0,link_state=4,power_mode=0
D (6909) BT_L2CAP: l2c_link_send_to_lower
D (6909) BT_L2CAP: TotalWin=8,Hndl=0x81,Quota=9,Unack=1,RRQuota=0,RRUnack=0
D (6919) BT_L2CAP: RR scan pri=2, lcid=0x0042, q_cout=0
D (6919) BT_L2CAP: L2CAP - rcv_cid CID: 0x0001

D (6929) BT_L2CAP: cmd_code: 4, id:25, cmd_len:8
D (6929) BT_L2CAP: L2CAP - st: CONFIG evt: 14
D (6939) BT_L2CAP: l2c_fcr_process_peer_cfg_req() CFG fcr_present:0 fcr.mode:0 C
CB FCR mode:0 preferred: 0 allowed:1
D (6949) BT_L2CAP: l2cu_adjust_out_mps use 0   Based on peer_cfg.fcr.mps: 0  pac
ket_size: 1011
[... 97 lines cut ...]
I (7419) BT_L2CAP: L2CAP - Calling DisconnectCfm_Cb(), CID: 0x0042
I (7429) esp_spp_cb: ESP_SPP_DISCOVERY_COMP_EVT status=0 scn_num=1
I (7429) BT_L2CAP: L2CA_ErtmConnectReq()  PSM: 0x0003  BDA: 0013433f4cb0  p_ertm
_info: 0x0 allowed:0x0 preferred:0
D (7439) BT_L2CAP: l2cu_allocate_ccb: cid 0x0000
D (7449) BT_L2CAP: l2cu_enqueue_ccb CID: 0x0043  priority: 2
D (7449) BT_L2CAP: l2c_link_adjust_chnl_allocation
D (7459) BT_L2CAP: CID:0x0043 FCR Mode:0 Priority:2 TxDataRate:1 RxDataRate:1 Qu
ota:200
D (7469) BT_L2CAP: L2CAP - st: CLOSED evt: 21
I (7469) BT_L2CAP: L2CAP - L2CA_conn_req(psm: 0x0003) returned CID: 0x0043
I (7479) esp_spp_cb: ESP_SPP_CL_INIT_EVT
D (7509) BT_L2CAP: l2c_link_sec_comp: 10, 0x3ffc476c
D (7509) BT_L2CAP: L2CAP - st: ORIG_W4_SEC_COMP evt: 8
I (7509) BT_L2CAP: L2CAP - Calling ConnectCfm_Cb(), HCI_ERR_AUTH_FAILURE, CID: 0
x0043  Status: 5
D (7519) BT_L2CAP: l2cu_release_ccb: cid 0x0043  in_use: 1
D (7529) BT_L2CAP: l2cu_dequeue_ccb  CID: 0x0043
D (7529) BT_L2CAP: l2cu_no_dynamic_ccbs() IDLE timer 0, disconnecting link
D (7539) BT_L2CAP: l2cu_no_dynamic_ccbs() starting IDLE timeout: 30
W (7539) BT_RFCOMM: PORT_StartCnf failed result:5
I (7549) esp_spp_cb: ESP_SPP_CLOSE_EVT
I (7619) BT_L2CAP: L2CA_SetDesireRole() new:x1, disallow_switch:0

```

closed 2019-10-01 by MarWoz442

niemeyer, 2021-01-01:
@MarWoz442 Have you ever found more about the underlying cause for this issue, or found a reasonable workaround? I seem to be observing the same issue here.

abdulredwan, 2021-03-29:
Hi @MarWoz442, Did you resolve this issue? We are facing similar issue, your help will be much appreciated

Thanks
Redwan