espressif/esp-idf #3720: CMake build system: PROJECT_SOURCE_DIR for third-party project pointed to wrong place

huming2207, 2019-07-01:
## Environment

- Development Kit:      Homemade
- Kit version (for WroverKit/PicoKit/DevKitC):  N/A
- Module or chip used:  ESP32-WROVER
- IDF version (run ``git describe --tags`` to find it): v4.0-dev-1018-gbbafd5066
- Build System:    CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1-rc2) 8.2.0
- Operating System:  Linux
- Power Supply:  USB

## Problem Description

Hello again @renzbagaporo , 

It seems that there's an issue with `PROJECT_SOURCE_DIR` and `PROJECT_BINARY_DIR` variable with third-party projects.  These two variables have been pointed to the top directory of the main project source directory and top directory of main project's build directory.

For example, in my Pull Request of https://github.com/warmcat/libwebsockets/pull/1621, I did a workaround for this issue to prevent this issue affects the third-party sub-project. But I think this may be not a good idea as it changed too much too the sub-project itself.

For my workaround, please refer to here: https://github.com/warmcat/libwebsockets/blob/master/CMakeLists.txt#L274

So I would like to know that, did I misunderstood something and it supposed to be like this?  Or is it a bug?

I have provided the details listed below:

### Expected Behavior

- `PROJECT_SOURCE_DIR` in the third-party subproject's CMakeFile should be pointed to its own directory;
- `PROJECT_BINARY_DIR` in the third-party subproject's CMakeFile should be pointed to its own build directory.

Assume my project is in `/home/user/Project/esp-project` and my third-party project is placed in to `/home/user/Project/esp-project/main/lib/third-party`, then `PROJECT_SOURCE_DIR` should be pointed to `/home/user/Project/esp-project/main/lib/third-party` instead of `/home/user/Project/esp-project`.
 
### Actual Behavior

- `PROJECT_SOURCE_DIR` in the third-party subproject's CMakeFile was pointed to the top directory of the main project;
- `PROJECT_BINARY_DIR` in the third-party subproject's CMakeFile was pointed to the

### Steps to reproduce

1. Try compiling this project: https://github.com/huming2207/lws-test
2. Remove my workaround and change all `LWS_SRC_DIR` back to `PROJECT_SOURCE_DIR`, and `LWS_BIN_DIR` to `PROJECT_BINARY_DIR`: https://github.com/warmcat/libwebsockets/blob/master/CMakeLists.txt#L274
3. Try compile it again

huming2207, 2019-07-01:
Looks like you've done some other changes on CMake in the recent one day or two...I will test again tonight with the latest changes.

renzbagaporo, 2019-07-02:
Noticed that you setting stuff is before the `project` call? That might be it?

huming2207, 2019-07-02:
Hi @renzbagaporo ,

Yep thanks, problem solved. Sorry I didn't realise there's a `project()` beneath it... Normally I would leave it on top of each CMakeList.

Regards,
Jackson

closed 2019-07-02 by huming2207