espressif/esp-idf #3722: i2c command link malloc error (v3.2.2)

kvbulusu, 2019-07-02:
Environment
Development Kit: ESP32-LyraT
Module or chip used: ESP32-WROOM-32
IDF version (run git describe --tags to find it): v3.2.2
Build System: Make
Compiler version (run xtensa-esp32-elf-gcc --version to find it): 5.20
Operating System: Linux
Power Supply: USB

Problem Description
I'm seeing i2c errors randomly . I couldnt determine any pattern. Same code sometimes works and sometimes gives below error. Turning on debug doesnt give any further info as well. Any suggestions? Please let me know if you need any other info. 

Debug Logs
I (11905) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:3 
E (11915) i2c: i2c command link malloc error

koobest, 2019-07-02:
Did you call `i2c_cmd_link_delete` after each transfer?

kvbulusu, 2019-07-02:
Yes...Below is the code snippet:
	i2c_cmd_handle_t cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (OPT_ADDRESS << 1) | I2C_MASTER_WRITE, 1 );
	i2c_master_write_byte(cmd, reg, 1);
	i2c_master_stop(cmd);
	i2c_master_cmd_begin(I2C_PORT_NUM , cmd, 1000/portTICK_PERIOD_MS);
	i2c_cmd_link_delete(cmd);

koobest, 2019-07-15:
Hi,
Yes, your use is correct. Is there another task that uses too much memory? According to your description, we can't find the root cause. Can you provide a test code that can reproduce this issue?

thanks!!

kvbulusu, 2019-07-15:
Thanks @koobest . I havent see task mem consumption as a problem till now, but will keep that mind. Since this is bit random, I will try to refactor code and see . I'll close the issue and will reopen if needed.

closed 2019-07-15 by kvbulusu

varunkudalkar, 2020-01-30:
@kvbulusu Did you get any solutions for this issue, I might be having same issue.

kvbulusu, 2020-02-05:
> @kvbulusu Did you get any solutions for this issue, I might be having same issue.
No I didnt ..but refactored my code and I havent seen it ...check task mem consumption as suggested earlier ...