espressif/esp-idf #3723: error in https://github.com/espressif/esp-idf/tree/master/examples/protocols/README.md

nopnop2002, 2019-07-02:
There is some error in https://github.com/espressif/esp-idf/tree/master/examples/protocols/README.md.

# Error
https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples/common/connect.c

# Correct
https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/connect.c

# Error
https://github.com/espressif/esp-idf/blob/master/examples/examples/wifi/getting_started

# Correct
https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started

# Error
https://github.com/espressif/esp-idf/blob/master/examples/examples/ethernet/ethernet

# Correct
https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic

# Error
https://github.com/espressif/esp-idf/blob/master/examples/examples/ethernet/ethernet/README.md

# Correct
https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/README.md

krzychb, 2019-07-02:
HI @nopnop2002 thank you for taking time to document this issue and preparing detailed report.

nopnop2002, 2019-07-03:
Hello.
When you fixed README.md, please close this issues.

closed 2019-07-03 by igrr