espressif/esp-idf #3724: rt-thread 系统支持

xiaodingding, 2019-07-02:
你好，请问，贵公司有考虑对rtthread 嵌入式系统的支持吗？

Alvin1Zhang, 2019-07-02:
@xiaodingding Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

negativekelvin, 2019-07-03:
https://github.com/RT-Thread/rt-thread/issues/1610

BernardXiong, 2019-07-05:
esp这边修改了gcc toolchains，导致都不是标准的gcc语法，太不标准了。rt-thread最初有过一个版本，但是跟得太累

suda-morris, 2019-07-05:
您好，@BernardXiong 
1. 请问能否告知移植 RT-Thread 时遇到了哪些困难，需要我们提供哪些支持？

2. ESP32 和ARM Cortex M系列的芯片确实有差别（比如支持多种存储器，包括IRAM、Flash等等）

3. ESP-IDF最新使用的gcc8工具链已经集成了libc的代码（之前gcc5版本还需要依赖IDF中的newlib组件），这或许可以减轻移植esp32到RT-Thread的困难

BernardXiong, 2019-07-05:
我的仓库中有个版本：

https://github.com/BernardXiong/rtthread-esp-idf

可以clone下来，以scons + 你们原来工具链的方式试试（Windows/Linux环境都可以）。整体系统，组件，wifi网络都是可以正常工作的。然后你们可以换换不同的工具链（代码不改动），然后就编译错误了

igrr, 2019-08-15:
When building with GCC 8.2, we no longer use the newlib header files from $IDF_PATH/components/newlib/include, instead the version of newlib bundled with the toolchain is used. I think you need to remove this one line from Sconscript_idf:
https://github.com/BernardXiong/rtthread-esp-idf/blob/c67b3ef4f0b57f8e885bd92043e6240042caafe6/esp-idf-port/SConscript_idf#L135

There are also a few changes to the way the libraries are linked, please check the relevant cmakelists (https://github.com/espressif/esp-idf/blob/39f090a4f1dee4e325f8109d880bf3627034d839/components/newlib/CMakeLists.txt#L19-L28, https://github.com/espressif/esp-idf/blob/39f090a4f1dee4e325f8109d880bf3627034d839/components/newlib/CMakeLists.txt#L54-L58) or GNU Make equivalents (https://github.com/espressif/esp-idf/blob/39f090a4f1dee4e325f8109d880bf3627034d839/components/newlib/component.mk#L3-L23).

BernardXiong, 2019-08-16:
@igrr 
Thank you for your reply. We will try it with latest IDF and RT-Thread.

igrr, 2019-10-07:
I'm going to close this since there isn't anything actionable from IDF project side left. @BernardXiong If you run into any other problem, please open a new issue and we will look into it. Thanks!

closed 2019-10-07 by igrr