espressif/esp-idf #3725: [Wifi] can't associate as station after commit c139683

enricop, 2019-07-02:
## Environment

- Development Kit:   ESP32-WRover-B
- IDF version: v4.0-dev-1095-g16014079f
- Build System: Make
- Compiler version: 8.2.0
- Operating System: Linux

## Problem Description

After this commit
https://github.com/espressif/esp-idf/commit/c1396830243b4c8fbea255a04d8a04ed2cb9f3d7 
the example 
https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station
does not work anymore.

The commit above updates the submodule below to the latest commit on 28th of June:
https://github.com/espressif/esp32-wifi-lib/commit/6579ef9a7ebdfa4196398056ee5337c300ca74e3

(ssid and password are correct)

## Debug Logs

I (0) cpu_start: App cpu up.
I (376) heap_init: Initializing. RAM available for dynamic allocation:
I (383) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (389) heap_init: At 3FFB9000 len 00027000 (156 KiB): DRAM
I (395) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (401) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (408) heap_init: At 40090450 len 0000FBB0 (62 KiB): IRAM
I (414) cpu_start: Pro cpu start user code
I (432) spi_flash: detected chip: generic
I (433) spi_flash: flash io: dio
I (433) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (539) wifi station: ESP_WIFI_MODE_STA
I (549) wifi: wifi driver task: 3ffc0b64, prio:23, stack:3584, core=0
I (549) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (549) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (579) wifi: wifi firmware version: 9cff65e
I (579) wifi: config NVS flash: enabled
I (579) wifi: config nano formating: disabled
I (579) wifi: Init dynamic tx buffer num: 32
I (579) wifi: Init data frame dynamic rx buffer num: 32
I (589) wifi: Init management frame dynamic rx buffer num: 32
I (589) wifi: Init management short buffer num: 32
I (599) wifi: Init static rx buffer size: 1600
I (599) wifi: Init static rx buffer num: 10
I (609) wifi: Init dynamic rx buffer num: 32
I (699) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (699) wifi: mode : sta (30:ae:a4:cc:3e:c8)
I (699) wifi station: wifi_init_sta finished.
I (709) wifi station: connect to ap SSID:***** password:***********
I (829) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (829) wifi: state: init -> auth (b0)
I (829) wifi: state: auth -> assoc (0)
I (839) wifi: state: assoc -> run (10)
I (4839) wifi: state: run -> init (2c0)
I (4849) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (4849) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (4849) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (4859) wifi: state: init -> auth (b0)
I (4869) wifi: state: auth -> assoc (0)
I (4879) wifi: state: assoc -> run (10)
I (8879) wifi: state: run -> init (2c0)
I (8879) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (8879) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (8889) wifi station: retry to connect to the AP
I (8889) wifi station: connect to the AP fail
I (9489) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (9489) wifi: state: init -> auth (b0)
I (9499) wifi: state: auth -> assoc (0)
I (9509) wifi: state: assoc -> run (10)
I (13509) wifi: state: run -> init (2c0)
I (13509) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (13509) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (13519) wifi station: retry to connect to the AP
I (13519) wifi station: connect to the AP fail
I (14849) wifi: new:<11,0>, old:<6,0>, ap:<255,255>, sta:<11,0>, prof:1
I (14849) wifi: state: init -> auth (b0)
I (14859) wifi: state: auth -> assoc (0)
I (14869) wifi: state: assoc -> run (10)
I (18879) wifi: state: run -> init (2c0)
I (18879) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (18879) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (18889) wifi station: retry to connect to the AP
I (18889) wifi station: connect to the AP fail
I (20939) wifi station: retry to connect to the AP
I (20939) wifi station: connect to the AP fail
I (21059) wifi: new:<1,0>, old:<11,0>, ap:<255,255>, sta:<1,0>, prof:1
I (21059) wifi: state: init -> auth (b0)
I (21069) wifi: state: auth -> assoc (0)
I (21079) wifi: state: assoc -> run (10)
I (25089) wifi: state: run -> init (2c0)
I (25089) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (25089) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (25089) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (25099) wifi: state: init -> auth (b0)
I (25109) wifi: state: auth -> assoc (0)
I (25119) wifi: state: assoc -> run (10)
I (29119) wifi: state: run -> init (2c0)
I (29119) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (29129) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (29129) wifi station: retry to connect to the AP
I (29139) wifi station: connect to the AP fail
I (29739) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (29739) wifi: state: init -> auth (b0)
I (29739) wifi: state: auth -> assoc (0)
I (29749) wifi: state: assoc -> run (10)
I (33759) wifi: state: run -> init (2c0)
I (33759) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (33759) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (33769) wifi station: connect to the AP fail

liuzfesp, 2019-07-03:
HI @enricop, how about commit IDF 39a5fbd?

enricop, 2019-07-03:
== AP Configuration == 
Protocol:	Wi-Fi 5 (802.11ac)
Security:	WPA2-Personal
Band:	5 GHz
Channel:	36
with DHCP

sorry, I can't understand what your commit is about.

liuzfesp, 2019-07-03:
@enricop, I guess the the band of AP is 2.4G because ESP32 doesn't support 5G yet, right? BTW, what's the brand of the AP? I suspect this issue only exists on some specific APs.

The commit 39a5fbd is the commit before c139683, because I want to know whether this issue is introduced by c139683.

enricop, 2019-07-03:
With the previous commit 39a5fbdfeebd1826f7be3928bd23447a15c260f1 it works with the same AP.

==== DEBUG LOGS ====
I (0) cpu_start: App cpu up.
I (375) heap_init: Initializing. RAM available for dynamic allocation:
I (382) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (388) heap_init: At 3FFB9010 len 00026FF0 (155 KiB): DRAM
I (395) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (401) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (407) heap_init: At 40090458 len 0000FBA8 (62 KiB): IRAM
I (414) cpu_start: Pro cpu start user code
I (432) spi_flash: detected chip: generic
I (432) spi_flash: flash io: dio
I (433) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (539) wifi station: ESP_WIFI_MODE_STA
I (549) wifi: wifi driver task: 3ffc0b70, prio:23, stack:3584, core=0
I (549) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (549) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (579) wifi: wifi firmware version: ec61a20
I (579) wifi: config NVS flash: enabled
I (579) wifi: config nano formating: disabled
I (579) wifi: Init dynamic tx buffer num: 32
I (579) wifi: Init data frame dynamic rx buffer num: 32
I (589) wifi: Init management frame dynamic rx buffer num: 32
I (589) wifi: Init management short buffer num: 32
I (599) wifi: Init static rx buffer size: 1600
I (599) wifi: Init static rx buffer num: 10
I (609) wifi: Init dynamic rx buffer num: 32
I (699) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (699) wifi: mode : sta (30:ae:a4:cc:3e:c8)
I (699) wifi station: wifi_init_sta finished.
I (709) wifi station: connect to ap SSID:****** password:***********
I (829) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (829) wifi: state: init -> auth (b0)
I (829) wifi: state: auth -> assoc (0)
I (839) wifi: state: assoc -> run (10)
I (859) wifi: connected with RE:Lab, channel 1, bssid = b4:fb:e4:77:da:db
I (859) wifi: pm start, type: 1

I (1539) tcpip_adapter: sta ip: 192.168.3.60, mask: 255.255.0.0, gw: 192.168.1.254
I (1539) wifi station: got ip:192.168.3.60

enricop, 2019-07-03:
hi @liuzfesp the AP has both 2.4G and 5G and we were correctly associating with w-room and w-rover devices with it before latest changes. The AP specs I've put before was taken from my laptop connected to the same AP. sorry for that. I don't have access to the AP directly.

liuzfesp, 2019-07-03:
@enricop I just can't reproduce this issue with my APs, could you let me know your AP type?

enricop, 2019-07-03:
Hi @liuzfesp : I was able to connect to another AP using the new code. I don't know why using the old code it works also with the first AP.
The two APs are the same physical device. The problem I see with the old code and the first AP is an association problem, so the part handled by the wpa_supplicant. It is not a DHCP problem.

liuzfesp, 2019-07-04:
@enricop we passed WiFi association regression test before merge commit c139683. If the ESP32 can't associate the 1st AP, it should be related to wpa_supplicant. I'm going to replicate this issue myself, but I just can't reproduce it. It would be appreciate if you can provide the type of your AP and I can do the test with the same AP as yours.

xiruilin, 2019-07-04:
I got the same issue.
With commit [c139683](https://github.com/espressif/esp-idf/commit/c1396830243b4c8fbea255a04d8a04ed2cb9f3d7), test three APs:

1st AP, TP-LINK wr800N, it's good, log below:

I (1807) WIFI_SERV: Connect to stored Wi-Fi SSID:********
I (1901) wifi: mode : sta (b4:e6:2d:a4:55:01)
I (1907) WIFI_SERV: WIFI_SERV_CMD_CONNECT
W (1911) WIFI_SERV: STATE type:1, pdata:0x0, len:0

I (2031) wifi: new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (2710) wifi: state: init -> auth (b0)
I (2713) wifi: state: auth -> assoc (0)
I (2719) wifi: state: assoc -> run (10)
I (3237) wifi: connected with ********, channel 1, bssid = 50:fa:84:96:8b:b1
I (3240) wifi: pm start, type: 1

I (5428) tcpip_adapter: sta ip: 192.168.1.101, mask: 255.255.255.0, gw: 192.168.1.1
I (5428) WIFI_SERV: Got ip:192.168.1.101
W (5430) WIFI_SERV: STATE type:2, pdata:0x0, len:0

2nd AP, 小米路由器3C, it's failed, log below:

I (1799) WIFI_SERV: Connect to stored Wi-Fi SSID:********
I (1903) wifi: mode : sta (b4:e6:2d:a4:55:01)
I (1908) WIFI_SERV: WIFI_SERV_CMD_CONNECT
W (1912) WIFI_SERV: STATE type:1, pdata:0x0, len:0

I (3120) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (3121) wifi: state: init -> auth (b0)
I (3158) wifi: state: auth -> assoc (0)
I (3199) wifi: state: assoc -> run (10)
I (9323) wifi: state: run -> init (fc0)
I (9324) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (9324) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (9329) WIFI_SERV: STA Auth Error
W (9333) WIFI_SERV: STATE type:3, pdata:0x0, len:0
W (9338) WIFI_SERV: Disconnect reason 1
I (11338) WIFI_SERV: WIFI_SERV_CMD_CONNECT
I (12548) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (12549) wifi: state: init -> auth (b0)
I (12586) wifi: state: auth -> assoc (0)
I (12626) wifi: state: assoc -> run (10)
I (18743) wifi: state: run -> init (fc0)
I (18744) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (18744) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (18749) WIFI_SERV: STA Auth Error
W (18753) WIFI_SERV: STATE type:3, pdata:0x0, len:0
W (18758) WIFI_SERV: Disconnect reason 1

3rd AP, Android Mobile WLAN hotspot, it's failed too, log below:

I (1807) WIFI_SERV: Connect to stored Wi-Fi SSID:********
I (1903) wifi: mode : sta (b4:e6:2d:a4:55:01)
I (1905) WIFI_SERV: WIFI_SERV_CMD_CONNECT
W (1912) WIFI_SERV: STATE type:1, pdata:0x0, len:0

I (2516) wifi: new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2516) wifi: state: init -> auth (b0)
I (2520) wifi: state: auth -> assoc (0)
I (2536) wifi: state: assoc -> run (10)
I (6557) wifi: state: run -> init (3c0)
I (6557) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (6558) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (6562) WIFI_SERV: STA Auth Error
W (6566) WIFI_SERV: STATE type:3, pdata:0x0, len:0
W (6571) WIFI_SERV: Disconnect reason 1
I (8571) WIFI_SERV: WIFI_SERV_CMD_CONNECT
I (9177) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (9177) wifi: state: init -> auth (b0)
I (9187) wifi: state: auth -> assoc (0)
I (9203) wifi: state: assoc -> run (10)
I (13212) wifi: state: run -> init (3c0)
I (13212) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (13212) wifi: new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (13217) WIFI_SERV: STA Auth Error
W (13221) WIFI_SERV: STATE type:3, pdata:0x0, len:0
W (13226) WIFI_SERV: Disconnect reason 1

Roll-back to commit [72a5762](https://github.com/espressif/esp-idf/commit/72a5762525d6201599d993a65e4d6e51eca125e7), all APs work fine.

enricop, 2019-07-04:
> @enricop we passed WiFi association regression test before merge commit [c139683](https://github.com/espressif/esp-idf/commit/c1396830243b4c8fbea255a04d8a04ed2cb9f3d7). If the ESP32 can't associate [...]

the model of the AP is : Ubiquity UniFi UAP-AC-LR

sagb2015, 2019-07-05:
@enricop : Please refer to the last few comments in the [issue](https://github.com/espressif/esp-idf/issues/3726). Can you try with the same AP with which it was failing? It should work fine afterwards.

liuzfesp, 2019-07-08:
Hi @enricop @xiruilin,  could you help to check whether it can fix your issue or NOT: add two lines after https://github.com/espressif/esp-idf/blob/138c941fad431d3ef2ccbe3eaabf08b96cdc4d0b/components/wpa_supplicant/src/rsn_supp/wpa.c#L1785 and before https://github.com/espressif/esp-idf/blob/138c941fad431d3ef2ccbe3eaabf08b96cdc4d0b/components/wpa_supplicant/src/rsn_supp/wpa.c#L1786:

`extern void esp_wifi_sta_update_ap_info_internal(void);`
`esp_wifi_sta_update_ap_info_internal();`

enricop, 2019-07-08:
Hi, adding those two lines in wpa.c does not fix the issue with our first AP.

xiruilin, 2019-07-09:
Hi, @liuzfesp，thanks for you work.
I add two lines, but the issue still exists, whether it's Xiao AP 3C or Android hotspot.

liuzfesp, 2019-07-09:
@xiruilin could you take a picture about your Xiaomi AP 3C? I think I need to buy the same AP as yours and then replicate this issue...

xiruilin, 2019-07-09:
@liuzfesp now it's upgraded to 4C. [Xiaomi AP 4C](https://detail.tmall.com/item.htm?spm=a1z10.1-b-s.w15914064-15567552165.154.67b05f3bWZegRe&id=574686451093&scene=taobao_shop&skuId=3761176249258)

liuzfesp, 2019-07-09:
HI @xiruilin today, I tried more 7 different Android hotspot and also tried Xiaomi 3C AP, but still failed to reproduce this issue. 

Could you give me your demo code that can generate this issue?

xiruilin, 2019-07-11:
Hi, @liuzfesp
This issue can be solved by erasing flash.

`
make erase_flash
`

Demo code [[esp-idf examples]](https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station)

1. use commit 72a5762 build it, it's run good, log below:
`
I (676) wifi: mode : sta (24:0a:c4:08:69:00)
I (686) wifi station: wifi_init_sta finished.
I (686) wifi station: connect to ap SSID:****** password:********
I (806) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (1816) wifi: state: init -> auth (b0)
I (1856) wifi: state: auth -> assoc (0)
I (1896) wifi: state: assoc -> run (10)
I (2036) wifi: connected with ******, channel 11, bssid = ****************
I (2046) wifi: pm start, type: 1
I (3006) tcpip_adapter: sta ip: 192.168.1.117, mask: 255.255.255.0, gw: 192.168.1.1
I (3006) wifi station: got ip:192.168.1.117
`
2. use commit c139683 build it, it's run failed, reason=15, log below:
I (674) wifi: mode : sta (24:0a:c4:08:69:00)
I (674) wifi station: wifi_init_sta finished.
I (684) wifi station: connect to ap SSID:****** password:********
I (804) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (804) wifi: state: init -> auth (b0)
I (844) wifi: state: auth -> assoc (0)
I (884) wifi: state: assoc -> run (10)
I (6994) wifi: state: run -> init (fc0)
I (6994) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (6994) wifi: new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
I (7004) wifi station: retry to connect to the AP
I (7004) wifi station: connect to the AP fail

3. Erase flash and use commit c139683 build it, it's run good.
I (674) wifi: mode : sta (24:0a:c4:08:69:00)
I (684) wifi station: wifi_init_sta finished.
I (684) wifi station: connect to ap SSID:****** password:********
I (804) wifi: new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1
I (1784) wifi: state: init -> auth (b0)
I (1824) wifi: state: auth -> assoc (0)
I (1864) wifi: state: assoc -> run (10)
I (2004) wifi: connected with ******, channel 11, HT20, bssid = ****************
I (2014) wifi: pm start, type: 1
I (3004) tcpip_adapter: sta ip: 192.168.1.117, mask: 255.255.255.0, gw: 192.168.1.1
I (3004) wifi station: got ip:192.168.1.117

enricop, 2019-07-11:
Hi, we noticed indeed that the issue is somehow related to NVS. I confirm that erasing default "nvs" partition could change if the esp32 connects or not to an AP.

liuzfesp, 2019-07-11:
Thanks @xiruilin @enricop,  the root cause has been found, we need some time to do review/regression test. We will deliver the fix next week.

liuzfesp, 2019-07-24:
Hi @enricop , the issue is fixed by f3f08fa7

Alvin1Zhang, 2019-07-26:
@enricop Thanks for reporting the issue, the fixes has already been synchronized into GitHub, feel free to reopen the ticket if your issue still exists. Thanks.

closed 2019-07-26 by Alvin1Zhang