espressif/esp-idf #3726: [WiFi] ieee80211_hostap.c 280 error

no1seman, 2019-07-02:
## Environment

- Development Kit:      ESP32-Wrover-Kit + LAN8720
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used:  ESP32-WROVER-B
- IDF version:  v4.0-dev-1095-g16014079f
- Build System:  CMake
- Compiler version:  xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System:   Linux
- Power Supply:   USB|external 5V&Battery

## Problem Description
Tried to look at esp-idf/examples/ethernet/eth2ap and got the following error:
E (6102) wifi: ieee80211_hostap.c 280

### Expected Behavior
No error
### Actual Behavior
Got error
### Steps to repropduce
Just build, flash and run esp-idf/examples/ethernet/eth2ap example with default settings.

### Full log:
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6488
load:0x40078000,len:10696
load:0x40080400,len:6560
entry 0x40080768
I (28) boot: ESP-IDF v4.0-dev-1095-g16014079f 2nd stage bootloader
I (28) boot: compile time 20:01:09
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 8MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x138b0 ( 80048) map
I (122) esp_image: segment 1: paddr=0x000238d8 vaddr=0x3ffb0000 size=0x03100 ( 12544) load
I (127) esp_image: segment 2: paddr=0x000269e0 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/user/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1778

I (129) esp_image: segment 3: paddr=0x00026de8 vaddr=0x40080400 size=0x09228 ( 37416) load
I (154) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x5bc1c (375836) map
0x400d0018: _stext at ??:?

I (288) esp_image: segment 5: paddr=0x0008bc3c vaddr=0x40089628 size=0x08b70 ( 35696) load
0x40089628: udp_sendto at /home/user/esp/esp-mdf/esp-idf/components/lwip/lwip/src/core/udp.c:588

I (314) boot: Loaded app from partition at offset 0x10000
I (315) boot: Disabling RNG early entropy source...
I (315) cpu_start: Pro cpu up.
I (319) cpu_start: Application information:
I (323) cpu_start: Project name:     eth2ap
I (328) cpu_start: App version:      1
I (333) cpu_start: Compile time:     Jul  2 2019 20:01:03
I (339) cpu_start: ELF file SHA256:  43bf69882d6684b8...
I (345) cpu_start: ESP-IDF:          v4.0-dev-1095-g16014079f
I (352) cpu_start: Starting app cpu, entry point is 0x40081220
0x40081220: call_start_cpu1 at /home/user/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:279

I (0) cpu_start: App cpu up.
I (362) heap_init: Initializing. RAM available for dynamic allocation:
I (369) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (375) heap_init: At 3FFB84F8 len 00027B08 (158 KiB): DRAM
I (381) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (388) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (394) heap_init: At 40092198 len 0000DE68 (55 KiB): IRAM
I (400) cpu_start: Pro cpu start user code
I (418) spi_flash: detected chip: generic
I (419) spi_flash: flash io: dio
I (419) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (492) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (512) eth_example: Ethernet Started
I (522) wifi: wifi driver task: 3ffb9094, prio:23, stack:3584, core=0
I (522) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (522) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (552) wifi: wifi firmware version: 9cff65e
I (552) wifi: config NVS flash: enabled
I (552) wifi: config nano formating: disabled
I (552) wifi: Init dynamic tx buffer num: 32
I (552) wifi: Init data frame dynamic rx buffer num: 32
I (562) wifi: Init management frame dynamic rx buffer num: 32
I (562) wifi: Init management short buffer num: 32
I (572) wifi: Init static rx buffer size: 1600
I (572) wifi: Init static rx buffer num: 10
I (582) wifi: Init dynamic rx buffer num: 32
I (6012) eth_example: Ethernet Link Up
I (6102) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (6102) wifi: mode : softAP (b4:e6:2d:ee:7e:78)
E (6102) wifi: ieee80211_hostap.c 280

liuzfesp, 2019-07-02:
@no1seman, it's our bug, we will look into it. Please @zhangyanjiaoesp take a look at this issue.

suda-morris, 2019-07-03:
Sorry, since we have moved the wpa_supplicant to the idf, and updated the `esp_wifi_init`, so the `eth2ap` won't work on master now. We're fixing this bug these days, hope will push to github ASAP. 

For now, you can hack the issue by remove [Line102~105](https://github.com/espressif/esp-idf/blame/master/components/esp_wifi/src/wifi_init.c#L102) and change [this line](https://github.com/espressif/esp-idf/blob/master/examples/ethernet/eth2ap/main/ethernet_example_main.c#L169) to `ESP_ERROR_CHECK(esp_wifi_init(&cfg));`.

tomseago, 2019-07-04:
This change to the wpa_supplicant, or some similar change, seems to have also broken the `examples/wifi/getting_started/station` example as well.

tomseago, 2019-07-04:
To be more specific about my last comment, it appears that a board which was previously working and had been flashed with a build of the station example from an IDF build prior to the inclusion of the supplicant move will no longer function with a build made after the supplicant move unless a nvs_flash_erase is done at say line 96 of station_example_main.c - that is, before the nvs_flash_init call happens. I haven't narrowed down the exact commit which causes the need for this "erase before init" but it seems like nvs_flash_init() needs to be returning ESP_ERR_NVS_NEW_VERSION_FOUND because of recent changes related to the supplicant move and it is not.

Hope this helps. Obviously this is a very problematic bug because boards are no longer able to connect to Wifi at all as a result.

sagb2015, 2019-07-04:
ESP_ERR_NVS_NEW_VERSION_FOUND reason serves the purpose of indicating incompatibility between NVS format found the flash and NVS format expected by the application (if compiled with older IDF version).  If you are testing with builds with just commit difference, there has not been any changes in NVS format. The cause for this might be changes in WiFi attributes stored on the NVS.

sagb2015, 2019-07-05:
This was indeed a change in WiFi attribute interpretation (PMK to be specific) due to changed code flow. As a workaround, you can make the if block present at https://github.com/espressif/esp-idf/blob/master/components/wpa_supplicant/src/rsn_supp/wpa.c#L1778 unconditional and then you could seamlessly switch between the commits.

liuzfesp, 2019-07-08:
HI @tomseago, could you help to check whether it can fix your issue or NOT: add two lines after https://github.com/espressif/esp-idf/blob/138c941fad431d3ef2ccbe3eaabf08b96cdc4d0b/components/wpa_supplicant/src/rsn_supp/wpa.c#L1785 and before https://github.com/espressif/esp-idf/blob/138c941fad431d3ef2ccbe3eaabf08b96cdc4d0b/components/wpa_supplicant/src/rsn_supp/wpa.c#L1786:

`extern void esp_wifi_sta_update_ap_info_internal(void);`
`esp_wifi_sta_update_ap_info_internal();`

enricop, 2019-07-08:
Hi,
adding those two lines in wpa.c does not fix the issue with our first AP.

closed 2019-07-15 by igrr