espressif/esp-idf #3727: Extract binary file content from HTTP body of post request

carlessole, 2019-07-02:
Hello,
I'm using the stable release of ESP-IDF: [v3.2.2](https://github.com/espressif/esp-idf/releases/tag/v3.2.2) and I'm trying to do a simple http_server and send a file to it.

When I do the HTTP POST (using for example [postman](https://www.getpostman.com/) or  [insomnia](https://insomnia.rest/)) it looks something like that (using wireshark):
[image]
The "blue marked" bytes are the content of my binary file (demo binary file 0x00,0x01...0x3F).

I was able to get from ESP32 side all the body of the POST request (see console logs):
[image]

My problem is that I don't know ho to get just the bytes that corresponds to the binary file.

Is there an standard way to achieve that? Or I need to implement a custom parser for that task?

Thanks

anurag-kar, 2019-07-02:
Hi @carlessole

I believe pos

carlessole, 2019-07-03:
Thanks @anurag-kar.
And regarding to the "multipart/form-data" I do it on purpose because is the way that an HTML form will submit the file, right?

Sending just the binary file is ok, and it is easy to parse because body just contains the file content:
[image]

But I wanted to know how to parse the file content in a "multipart/form-data". As you said, it is not supported yet, right? So I will follow the example you provide me to make my web to send just the raw binary file.

Thank you!

anurag-kar, 2019-07-03:
@carlessole Glad that it works for you.

> As you said, it is not supported yet, right?

That's right. We do have plans to be able to support  "multipart/form-data" and "chunked" in the future, though presently it's not high priority.

I am going to close this issue now. If the need arises, feel free to reopen.

closed 2019-07-03 by anurag-kar

creatormir, 2023-07-09:
@anurag-kar, Is there support for x-www-form-urlencoded and form-data ?