espressif/esp-idf #3728: esp-idf/examples/ethernet/eth2ap become watchdog fail.

nopnop2002, 2019-07-03:
I updated esp-idf to this:

## Environment

	freeRTOS version:V8.2.0
	NEWLIB version:2.2.0
	lwIP version:2-0-3-255
	ESP-IDF version:v4.0-dev-1095-g16014079f

## Problem Description

esp-idf/examples/ethernet/eth2ap become watchdog fail.

[image]

### Steps to repropduce
[image]

[image]

[image]

[image]

[image]

[image]

suda-morris, 2019-07-03:
Same issue to [3726](https://github.com/espressif/esp-idf/issues/3726)

nopnop2002, 2019-07-03:
Hello.
I'm waiting your fix.

closed 2019-07-15 by igrr

nopnop2002, 2019-07-15:
I tried esp-idf/examples/protocols/sntp using eth2ap.
Is this correct?

[image]

suda-morris, 2019-07-16:
Theoretically it can work. From your log I can see your station has got an IP address via the wifi provided by another esp32 board, right? But for the tcpip_adapter error info, I don't know the reason. You might need to monitor the log from both the two esp32 board, to see what happened on eth2ap example when you got this tcpip_adapter error.

nopnop2002, 2019-07-16:
Hello.
This is the log from both the two esp32 board.

eth2ap side:

	I (0) cpu_start: App cpu up.
	I (386) heap_init: Initializing. RAM available for dynamic allocation:
	I (393) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (399) heap_init: At 3FFB88C0 len 00027740 (157 KiB): DRAM
	I (406) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (412) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (418) heap_init: At 40090468 len 0000FB98 (62 KiB): IRAM
	I (425) cpu_start: Pro cpu start user code
	I (443) spi_flash: detected chip: generic
	I (443) spi_flash: flash io: dio
	I (443) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (514) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (524) eth_example: Ethernet Started
	I (534) wifi: wifi driver task: 3ffb94fc, prio:23, stack:3584, core=0
	I (534) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (534) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (564) wifi: wifi firmware version: 82c4fce
	I (564) wifi: config NVS flash: enabled
	I (564) wifi: config nano formating: disabled
	I (564) wifi: Init dynamic tx buffer num: 32
	I (564) wifi: Init data frame dynamic rx buffer num: 32
	I (574) wifi: Init management frame dynamic rx buffer num: 32
	I (574) wifi: Init management short buffer num: 32
	I (584) wifi: Init static rx buffer size: 1600
	I (584) wifi: Init static rx buffer num: 10
	I (594) wifi: Init dynamic rx buffer num: 32
	I (6094) eth_example: Ethernet Link Up
	I (6204) phy: phy_version: 4101, 100880e, Jul 12 2019, 19:32:06, 0, 0
	I (6204) wifi: mode : softAP (30:ae:a4:ca:e1:43)
	I (6204) wifi: Total power save buffer number: 16
	I (6204) wifi: Init max length of beacon: 752/752
	I (6214) wifi: Init max length of beacon: 752/752
	I (47044) wifi: new:<1,1>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
	I (47044) wifi: station: 3c:71:bf:4f:c1:a0 join, AID=1, bgn, 40U
	I (47064) eth_example: Wi-Fi AP got a station connected
	I (54754) wifi: station: 3c:71:bf:4f:c1:a0 leave, AID = 1, bss_flags is 134243, bss:0x3ffc8460
	I (54754) wifi: new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
	I (54764) eth_example: Wi-Fi AP got a station disconnected

sntp sido:

	I (0) cpu_start: Starting scheduler on APP CPU.
	I (468) example: Boot count: 1
	I (468) example: Time is not set yet. Connecting to WiFi and getting time over NTP.
	I (528) wifi: wifi driver task: 3ffc0d08, prio:23, stack:3584, core=0
	I (528) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (528) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
	I (558) wifi: wifi firmware version: 82c4fce
	I (558) wifi: config NVS flash: enabled
	I (558) wifi: config nano formating: disabled
	I (558) wifi: Init dynamic tx buffer num: 32
	I (558) wifi: Init data frame dynamic rx buffer num: 32
	I (568) wifi: Init management frame dynamic rx buffer num: 32
	I (568) wifi: Init management short buffer num: 32
	I (578) wifi: Init static rx buffer size: 1600
	I (578) wifi: Init static rx buffer num: 10
	I (588) wifi: Init dynamic rx buffer num: 32
	I (588) example_connect: Connecting to eth2ap...
	I (698) phy: phy_version: 4101, 100880e, Jul 12 2019, 19:32:06, 0, 0
	I (708) wifi: mode : sta (3c:71:bf:4f:c1:a0)
	I (828) wifi: new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
	I (1808) wifi: state: init -> auth (b0)
	I (1818) wifi: state: auth -> assoc (0)
	I (1828) wifi: state: assoc -> run (10)
	I (1848) wifi: connected with eth2ap, channel 1, 40U, bssid = 30:ae:a4:ca:e1:43
	I (1848) wifi: pm start, type: 1

	I (3008) tcpip_adapter: sta ip: 192.168.10.121, mask: 255.255.255.0, gw: 192.168.10.1
	I (3508) example_connect: Connected to eth2ap
	I (3508) example_connect: IPv4 address: 192.168.10.121
	I (3508) example_connect: IPv6 address: fe80:0000:0000:0000:3e71:bfff:fe4f:c1a0
	I (3518) example: Initializing SNTP
	I (3518) example: Waiting for system time to be set... (1/10)
	I (5528) example: Waiting for system time to be set... (2/10)
	I (7528) example: Waiting for system time to be set... (3/10)
	I (8138) example: Notification of a time synchronization event
	I (9528) wifi: state: run -> init (0)
	I (9528) wifi: pm stop, total sleep time: 5239617 us / 7677520 us

	I (9528) wifi: new:<1,0>, old:<1,1>, ap:<255,255>, sta:<1,1>, prof:1
	E (9528) tcpip_adapter: handle_sta_disconnected 197 esp_wifi_internal_reg_rxcb ret=0x3014
	I (9568) wifi: flush txq
	I (9568) wifi: stop sw txq
	I (9568) wifi: lmac stop hw txq
	I (9568) wifi: Deinit lldesc rx mblock:10
	I (9578) example_connect: Disconnected from eth2ap
	I (9578) example: The current date/time in New York is: Tue Jul 16 05:48:33 2019
	I (9578) example: The current date/time in Shanghai is: Tue Jul 16 17:48:33 2019
	I (9588) example: Entering deep sleep for 10 seconds

suda-morris, 2019-07-16:
Does sntp work like this when you connect to a real router not the eth2ap?

nopnop2002, 2019-07-16:
I connected to a real router not the eth2ap.

This is not due to eth2ap.

	I (0) cpu_start: App cpu up.
	I (403) heap_init: Initializing. RAM available for dynamic allocation:
	I (410) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
	I (416) heap_init: At 3FFB9210 len 00026DF0 (155 KiB): DRAM
	I (422) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
	I (428) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
	I (435) heap_init: At 40090C20 len 0000F3E0 (60 KiB): IRAM
	I (441) cpu_start: Pro cpu start user code
	I (459) spi_flash: detected chip: generic
	I (460) spi_flash: flash io: dio
	I (460) cpu_start: Starting scheduler on PRO CPU.
	I (0) cpu_start: Starting scheduler on APP CPU.
	I (468) example: Boot count: 3
	I (468) example: The current date/time in New York is: Tue Jul 16 10:01:14 2019
	I (478) example: The current date/time in Shanghai is: Tue Jul 16 22:01:14 2019
	I (488) example: Entering deep sleep for 10 seconds

[image]

igrr, 2019-07-16:
Regarding the  `esp_wifi_internal_reg_rxcb` error, it should was reported already and the fix should be in master soon (https://github.com/espressif/esp-idf/issues/3587#issuecomment-511185777)

nopnop2002, 2019-07-16:
OK.
I'm waiting for master fix.