espressif/esp-idf #3729: [newlib] gettimeofday/settimeofday doesn't return/use actual timezone

no1seman, 2019-07-03:
## Environment

- Development Kit:  ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v4.1
- Module or chip used: ESP32-WROVER-B
- IDF version: v4.0-dev-1095-g16014079f
- Build System: CMake
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp32-2019r1) 8.2.0
- Operating System: Linux
- Power Supply:  USB|external 5V&Battery

## Problem Description
1. gettimeofday() doesn't return actual timezone, tz_minuteswest always == 0
2. settimeofday() doesn't use timezone structure data to set local time

### Expected Behavior
1. gettimeofday() must return actual data in timezone struct
2. settimeofday() must use timezone struct data when setting local date/time, for example, if I pass GMT time in timeval and set timezone.tz_minuteswest = 0 (GMT) and local timezone != GMT  settimeofday() must calculate time offset and set local time according to local timezone 
### Actual Behavior
1. gettimeofday() always return timezone.tz_minuteswest == 0
2. settimeofday() skipping timezone.tz_minuteswest and setting localtime asis in timeval

### Steps to repropduce
See code below

### Code to reproduce this issue
#include <stdbool.h>
#include <stdio.h>
#include <string.h>
#include "sys/time.h"
#include "freertos/FreeRTOS.h"

#define DATE_TIME_FORMAT "%a %d %b %Y %H:%M:%S %Z"

void set_local_time_zone(void)
{
    setenv("TZ", "MSK-3", 1);
    tzset();
}

//Printing SYSTEM date/time using local timezone
void print_local_date_time(void)
{
    struct tm timeinfo = {0};
    time_t now = 0;
    char strftime_buf[32];
    
    time(&now);
    localtime_r(&now, &timeinfo);
    strftime(strftime_buf, sizeof(strftime_buf), DATE_TIME_FORMAT, &timeinfo);
    printf("SYSTEM date/time: %s\n", strftime_buf);
}

void app_main(void)
{
    struct timeval tv = {0};
    struct timezone tz = {0};
    
    printf("Before setting timezone:\n");
    print_local_date_time();
    gettimeofday(&tv, &tz);
    printf("'tz_dsttime': %d, 'tz_minuteswest': %d\n", tz.tz_dsttime, tz.tz_minuteswest);

    set_local_time_zone();

    printf("After setting timezone:\n");
    print_local_date_time();
    gettimeofday(&tv, &tz);
    printf("'tz_dsttime': %d, 'tz_minuteswest': %d\n", tz.tz_dsttime, tz.tz_minuteswest);
}

## Debug Logs

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6488
load:0x40078000,len:10696
load:0x40080400,len:6560
entry 0x40080768
I (29) boot: ESP-IDF v4.0-dev-1095-g16014079f 2nd stage bootloader
I (29) boot: compile time 14:44:47
I (39) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x05834 ( 22580) map
I (102) esp_image: segment 1: paddr=0x0001585c vaddr=0x3ffb0000 size=0x01ff4 (  8180) load
I (106) esp_image: segment 2: paddr=0x00017858 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /home/user/esp/esp-mdf/esp-idf/components/freertos/xtensa_vectors.S:1778

I (112) esp_image: segment 3: paddr=0x00017c60 vaddr=0x40080400 size=0x083b0 ( 33712) load
I (135) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x12c78 ( 76920) map
0x400d0018: _stext at ??:?

I (162) esp_image: segment 5: paddr=0x00032c98 vaddr=0x400887b0 size=0x00a98 (  2712) load
0x400887b0: xTaskCheckForTimeOut at /home/user/esp/esp-mdf/esp-idf/components/freertos/tasks.c:3231 (discriminator 1)

I (170) boot: Loaded app from partition at offset 0x10000
I (170) boot: Disabling RNG early entropy source...
I (172) cpu_start: Pro cpu up.
I (175) cpu_start: Application information:
I (180) cpu_start: Project name:     timezone
I (185) cpu_start: App version:      1
I (190) cpu_start: Compile time:     Jul  3 2019 14:44:41
I (196) cpu_start: ELF file SHA256:  811ba2eb6e14ffbb...
I (202) cpu_start: ESP-IDF:          v4.0-dev-1095-g16014079f
I (209) cpu_start: Starting app cpu, entry point is 0x40080f38
0x40080f38: call_start_cpu1 at /home/user/esp/esp-mdf/esp-idf/components/esp32/cpu_start.c:279

I (0) cpu_start: App cpu up.
I (219) heap_init: Initializing. RAM available for dynamic allocation:
I (226) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (232) heap_init: At 3FFB3000 len 0002D000 (180 KiB): DRAM
I (238) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (245) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (251) heap_init: At 40089248 len 00016DB8 (91 KiB): IRAM
I (257) cpu_start: Pro cpu start user code
I (276) spi_flash: detected chip: generic
I (276) spi_flash: flash io: dio
I (276) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Before setting timezone:
SYSTEM date/time: Thu 01 Jan 1970 00:00:00 GMT
'tz_dsttime': 0, 'tz_minuteswest': 0
After setting timezone:
SYSTEM date/time: Thu 01 Jan 1970 03:00:00 MSK
'tz_dsttime': 0, 'tz_minuteswest': 0

negativekelvin, 2019-07-03:
> The struct timezone type is obsolete and should never be used. 

https://www.gnu.org/software/libc/manual/html_node/High_002dResolution-Calendar.html

no1seman, 2019-07-03:
Ok, thanks for answer

closed 2019-07-03 by no1seman