espressif/esp-idf #3731: Installing IDF_TOOL_SETUP Cyrillic in the file path.

ArTyGill, 2019-07-03:
When trying to configure vitualenv, a path mismatch error occurs for UTF-8

CMD.exe output

```
C:\Docs\espressif\esp-idf>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, n
inja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0 (already installed)
Skipping ccache@3.7 (already installed)
Setting up Python environment
Creating a new Python environment in C:\Docs\espressif\.espressif\python_env\idf
4.0_py3.7_env
Installing virtualenv
Collecting virtualenv
  Using cached https://files.pythonhosted.org/packages/c4/9a/a3f62ac5122a65dec34
ad4b5ed8d802633dae4bc06a0fc62e55fe3e96fe1/virtualenv-16.6.1-py2.py3-none-any.whl

Installing collected packages: virtualenv
  WARNING: The script virtualenv.exe is installed in 'C:\Users\Артем\AppData\Roa
ming\Python\Python37\Scripts' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warn
ing, use --no-warn-script-location.
Successfully installed virtualenv-16.6.1
Using base prefix 'C:\\Program Files\\Python37'
New python executable in C:\Docs\espressif\.espressif\python_env\idf4.0_py3.7_en
v\Scripts\python.exe
Installing setuptools, pip, wheel...
done.
Traceback (most recent call last):
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 931, in call_subprocess
    line = line.decode(encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc0 in position 27: invalid start byte

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Program Files\Python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Program Files\Python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 2611, in <module>
    main()
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 862, in main
    symlink=options.symlink,
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 1154, in create_environment
    install_wheel(to_install, py_executable, search_dirs, download=download)
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 1004, in install_wheel
    _install_wheel_with_search_dir(download, project_names, py_executable, search_dirs)
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 1091, in _install_wheel_with_
search_dir
    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=script)
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 933, in call_subprocess
    line = line.decode(fs_encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc0 in position 27: invalid start byte
Traceback (most recent call last):
  File "C:\Docs\espressif\esp-idf\tools\idf_tools.py", line 1249, in <module>
    main(sys.argv[1:])
  File "C:\Docs\espressif\esp-idf\tools\idf_tools.py", line 1245, in main
    action_func(args)
  File "C:\Docs\espressif\esp-idf\tools\idf_tools.py", line 1069, in action_install_python_env
    stdout=sys.stdout, stderr=sys.stderr)
  File "C:\Program Files\Python37\lib\subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\Program Files\\Python37\\python.exe', '-m', 'virtualenv', '--no-site-pack
ages', 'C:\\Docs\\espressif\\.espressif\\python_env\\idf4.0_py3.7_env']' returned non-zero exit status 1.
All done! You can now run:
   export.bat
C:\Docs\espressif\esp-idf>
```

ArTyGill, 2019-07-03:
How can I fix the problem locally, or should I wait for bug fixes?

As you can see, the problem is that the username is given by the Cyrillic and this can only be corrected by major changes to the system.

```
Traceback (most recent call last):
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 931, in call_subprocess
    line = line.decode(encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc0 in position 27: invalid start byte
```

igrr, 2019-07-03:
Unfortunately this seems to be a limitation of virtualenv. While I couldn't reproduce the exact error you have posted, I did run into another one described here: https://github.com/pypa/virtualenv/issues/1204. This happens even without IDF, just by doing `virtualenv.exe <dirname>` in a directory like `C:\ascii-name`, as long as user name has non-ASCII characters in it. I will look for some workarounds, but a quick search hasn't revealed any so far.

ArTyGill, 2019-07-03:
@iggr the path must be utf-8, there are no Russian characters in the ascii table .

ArTyGill, 2019-07-03:
@igrr I managed to solve the problem, but I had to change the file virtualenv.py (Change the encoding to cp2112).
I did not understand why the system does not read utf-8

> 925 encoding` = "cp1251" # sys.getdefaultencoding()

ArTyGill, 2019-07-03:
When trying to change the path in utf-8, other errors occur.  for example, there is no pip module installed.

ArTyGill, 2019-07-04:
Led the string that swears virtualenv.

```
Installing setuptools, pip, wheel...
b'Looking in links: C:\\Users\\\xc0\xf0\xf2\xe5\xec\\AppData\\Roaming\\Python\\Python37\\site-packages\\virtualenv_support\r\n'
done.
Traceback (most recent call last):
  File "C:\Users\Артем\AppData\Roaming\Python\Python37\site-packages\virtualenv.py", line 932, in call_subprocess
    line = line.decode(encoding)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc0 in position 27: invalid start byte
```

ArTyGill, 2019-07-04:
@igrr 
I understood what the problem was.
In windows, paths are encoded in UTF-16 and windows-1251 (according to data from wikipedia).
In virtualenv.py, decoding by encoding occurs first using the file encoding (UTF-8), and then failing, the file system encoding must be windows-1251, but sys.getfilesystemencoding() returns utf-8 again.

If we replace line 933 in virtualenv.py  with an exception (listed below), then everything works fine.

```
try: 
    line = line.decode(fs_encoding)
except UnicodeDecodeError:
    line = line.decode('cp1251')
```

I think the developers of the SYS module or the developers of VIRTUALENV need to add processing locale and language installed on the computer.

tom-borcin, 2021-02-25:
Hello @ArTyGill,

thank you for reporting this issue. At Espressif we really appreciate it when our users report issues as it helps us make our products better. Unfortunately there hasn't been any progress with this issue for almost 2 years and we are closing it to keep our backlog manageable.

If you don't agree with this decision, please feel free to reopen the issue or contact me directly.

Regards,

Tomas

closed 2021-02-25 by tom-borcin

georgik, 2021-02-25:
@ArTyGill Please try to enable " Use Unicode UTF-8 for worldwide language support" in your Windows as described here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#esp-idf-tools-installer

"The installation path of Python or ESP-IDF should not contain special characters (non-ASCII) unless the operating system is configured with “Unicode UTF-8” support. System Administrator can enable the support via Control Panel - Change date, time, or number formats - Administrative tab - Change system locale - check the option “Beta: Use Unicode UTF-8 for worldwide language support” - Ok and reboot the computer."

Please try installation with newer 2.5 installer which fixes several issues with virtualenv:
- Online installer (25 MB) : https://dl.espressif.com/dl/esp-idf-tools-setup-online-2.5.exe
- Offline installer - contains v4.2 and v4.1.1 (810 MB): https://dl.espressif.com/dl/esp-idf-tools-setup-offline-2.5.exe