espressif/esp-idf #3732: Several bootloaders on esp32 and update bootloaders.

ArTyGill, 2019-07-03:
I can create custom bootloader for load two secondary bootloaders(to choose), for OTA1 and OTA2?

I would like the bootloaders to be updated along with the app.

Alvin1Zhang, 2019-07-03:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

negativekelvin, 2019-07-03:
You could possibly create 1 custom 2nd stage bootloader and two custom 3rd stage bootloaders, but it is not supported.  Why can't you just put the custom code in the app?  What does your bootloader do?

ArTyGill, 2019-07-03:
I want update esp-idf with ota.

negativekelvin, 2019-07-03:
But why do you want to update the bootloader and not just the app?

ArTyGill, 2019-07-03:
I want to update with ota the bootloader and the application together

negativekelvin, 2019-07-03:
But what is your reason?

ArTyGill, 2019-07-03:
@negativekelvin New features and stability enhancement, bug fixes, OTA updates.

ArTyGill, 2019-07-03:
and do not think about what esp-idf bootloader is installed before the OTA.

negativekelvin, 2019-07-03:
Bootloader is designed to be minimal, backwards compatible, and never updated.  So unless you need to do something special, it's probably a waste of time.  I don't think there will be an official support for it, but try it yourself if you want.  You may as well make partition table upgradable too.

ArTyGill, 2019-07-04:
Can I run the standard esp-idf bootloader using my bootloader?

negativekelvin, 2019-07-04:
Yes, it would be smart to make it work that way

ArTyGill, 2019-07-05:
I will try to run.

closed 2019-07-05 by ArTyGill