espressif/esp-idf #3733: Title: Pin interrupts are constantly triggered

tutuwin, 2019-07-03:
----------------------------- Delete below -----------------------------

Text: Set a pin bit falling edge interrupt, it will trigger continuously after one trigger, can't stop, but the pin voltage measurement has returned to high level, using pin io26.

Alvin1Zhang, 2019-07-03:
Thanks for reporting the issue. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/blob/master/docs/issue_template.md)? Thanks.

tutuwin, 2019-07-03:
> Thanks for reporting the issue. Would you please help provide more details as suggested in the issue template? Thanks.
   
我使用xpt2046的irq引脚作为中断引脚，当有下降沿时会触发中断事件，之后我用事件组传出此事件，在另一个任务等待此事件然后读取坐标，结果时触发一次后，此引脚持续中断。请问gpio中断需要清除标志位吗？如果有怎么清除

          I use the IRQ pin of xpt2046 as the interrupt pin. When there is a drop edge, the interrupt event will be triggered. Then I use the event group to send out the event. I wait for the event in another task and read the coordinates. When the result is triggered once, the pin will continue to interrupt. Does the GPIO interrupt need to be cleared? If so, how to clean it up

negativekelvin, 2019-07-03:
Did you use gpio_install_isr_service or what code are you using?

tutuwin, 2019-07-04:
> Did you use gpio_install_isr_service or what code are you using?
code:void irq_gpio_init()
{
    gpio_config_t io_conf;

    //disable interrupt

    io_conf.intr_type = GPIO_PIN_INTR_DISABLE;

    //set as output mode

    io_conf.mode = GPIO_MODE_INPUT;

    //bit mask of the pins that you want to set,e.g.GPIO

    io_conf.pin_bit_mask = (1<<pen_irq);

    //disable pull-down mode

    io_conf.pull_down_en = 0;

   
    //enable pull-up mode

    io_conf.pull_up_en = 1;

   
    //interrupt of faling edge

    io_conf.intr_type = GPIO_INTR_NEGEDGE;

    //bit mask of the pins, use GPIO here

    gpio_config(&io_conf);

    //change gpio intrrupt type for one pin

    gpio_set_intr_type(pen_irq, GPIO_INTR_NEGEDGE);

    //install gpio isr service

   esp_err_t re =  gpio_install_isr_service(0);

    printf("esp_err_t:%d\r\n",re);

    //hook isr handler for specific gpio pin

    gpio_isr_handler_add(pen_irq, gpio_isr_handler, (void*) pen_irq);

}

static  void IRAM_ATTR gpio_isr_handler(void* arg) 
{  
    #if 1
    uint32_t gpio_num = (uint32_t) arg;
    xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
    #endif
    //printf("tp is running\r\n");
cnt++;
  //  BaseType_t xHigherPriorityTaskWoken;
  //  xEventGroupSetBitsFromISR(eg,TASK_1_BIT, &xHigherPriorityTaskWoken);

    //remove isr handler for gpio number.
  //  gpio_isr_handler_remove(pen_irq);
 //gpio_intr_disable(pen_irq);
    //hook isr handler for specific gpio pin again
    //gpio_isr_handler_add(pen_irq, gpio_isr_handler, (void*) pen_irq);
  
} 
position read_point(void* arg)
{

  bool pressed;
  static uint32_t io_num;
  while(1)
{if(xQueueReceive(gpio_evt_queue, &io_num, portMAX_DELAY)) 
  {
if(io_num == pen_irq )
{
    printf("cnt:%d\r\n",cnt);
  //  EventBits_t xbit = xEventGroupWaitBits(eg, TASK_1_BIT, pdTRUE, pdTRUE, portMAX_DELAY);
 
   // irq_gpio_init();
  //  printf("%02x\r\n",xbit);
  //  if((xbit & TASK_1_BIT) == TASK_1_BIT )
//if(gpio_get_level(pen_irq)==0 )
 //  {
     
   
    position samples[XPT2046_SMPSIZE];
    position distances[XPT2046_SMPSIZE];
    pressed = true;

    int aveX = 0;
    int aveY = 0;

    for (int i = 0; i < XPT2046_SMPSIZE; i++) {
        samples[i].x = getSample(TOUCH_CMD_X);
        samples[i].y = getSample(TOUCH_CMD_Y);
     
        if (samples[i].x == 0 || samples[i].x == 4095 || samples[i].y == 0 || samples[i].y == 4095){
            pressed = false;
        }
        aveX += samples[i].x;
        aveY += samples[i].y;
    }

    aveX /= XPT2046_SMPSIZE;
    aveY /= XPT2046_SMPSIZE;
    for (int i = 0; i < XPT2046_SMPSIZE; i++) {
        distances[i].x = i;
        distances[i].y = ((aveX - samples[i].x) * (aveX - samples[i].x)) + ((aveY - samples[i].y) * (aveY - samples[i].y));
    }

    for (int i = 0; i < XPT2046_SMPSIZE-1; i++) {
        for (int j = 0; j < XPT2046_SMPSIZE-1; j++) {
            if (samples[j].y > samples[j+1].y) {
                int yy = samples[j+1].y;
                samples[j+1].y = samples[j].y;
                samples[j].y = yy;
                int xx = samples[j+1].x;
                samples[j+1].x = samples[j].x;
                samples[j].x = xx;
            }
        }
    }

    int tx = 0;
    int ty = 0;
    int tx1 = 0;
    int ty1 = 0;
    for (int i = 0; i < (XPT2046_SMPSIZE >> 1); i++) {
        tx += samples[distances[i].x].x;
        ty += samples[distances[i].x].y;
    }

    tx1 = tx / (XPT2046_SMPSIZE >> 1);
    ty1 = ty / (XPT2046_SMPSIZE >> 1);
    tx=-0.06793*(tx1)+258.4;
    ty=-0.09142*(ty1)+354.7;  
    tp_point.press = pressed;

   if(tx < 0)
	  {
               tx = 0;
          } 
       if(ty < 0)
          {
	       ty = 0;
          }
       tp_point.x = tx;
       tp_point.y = ty;
       if(tp_point.x>=240)
             tp_point.x = 239;
       if( tp_point.y >= 320)
             tp_point.y = 319;    
       
  

    printf("tp_point.x:%d\r\n",tp_point.x);
    printf("tp_point.y:%d\r\n",tp_point.y);
    printf("pressed:%d\r\n",tp_point.press);
    
    vTaskDelay(1000 / portTICK_RATE_MS);

   }
  }

}
}
   
//}

koobest, 2019-07-09:
Hi, @tutuwin
Have you solved your problem? I tried your test code, but did not reproduce your issue. The interrupt works fine.

thanks !!

Alvin1Zhang, 2019-07-19:
@tutuwin Thanks for reporting the issue, sorry that we tried but didn't reproduce your issue. Would you help share if any updates from your side? Has your issue been resolved? Thanks.

Alvin1Zhang, 2020-01-15:
@tutuwin Thanks for reporting. Will close due to short of feedback, feel free to reopen if the issue still happens. Thanks.

closed 2020-01-15 by Alvin1Zhang