espressif/esp-idf #3735: flash read err, 1000 on Sparkfun Thing Plus

LtMerlin, 2019-07-03:
Hi, 

I flashed a brand new [Sparkfun ESP32 Thing Plus](https://www.sparkfun.com/products/14689) with a recent MicroPython firmware binary (from https://micropython.org/download/#esp32). (first erased properly and then flashing with the esptool as described below)

```bash
esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 460800 --before default_reset --after hard_reset erase_flash
```

```bash
esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 460800 --before default_reset --after hard_reset erase_flash
```

Everything works fine except at startup I see the error message "**flash read err, 1000
ets_main.c 371**" like you can see below at the first few lines:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4928
ho 0 tail 12 room 4
load:0x40078000,len:9332
load:0x40080400,len:6216
entry 0x400806e8
I (433) cpu_start: Pro cpu up.
I (433) cpu_start: Application information:
I (433) cpu_start: Compile time:     07:44:16
I (435) cpu_start: Compile date:     May 29 2019
I (441) cpu_start: ESP-IDF:          v3.3-beta1-268-g5c88c5996
I (447) cpu_start: Starting app cpu, entry point is 0x4008294c
I (0) cpu_start: App cpu up.
I (458) heap_init: Initializing. RAM available for dynamic allocation:
I (465) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (471) heap_init: At 3FFB9B88 len 00026478 (153 KiB): DRAM
I (477) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (483) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (490) heap_init: At 40093398 len 0000CC68 (51 KiB): IRAM
I (496) cpu_start: Pro cpu start user code
I (67) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32
Type "help()" for more information.
>>>
```
Using other firmware versions does not help to resolve it...

When I run the following commands to get the filesystem sizes:
```python
import os
fs_stat = os.statvfs('/')
fs_size = fs_stat[0] * fs_stat[2]
fs_free = fs_stat[0] * fs_stat[3]
print("File system size {:,} - Free space {:,}".format(fs_size, fs_free))
```
Returns:
```bash
File system size 2,072,576 - Free space 2,068,480
```

I only get 2MB of availble or free space while I should get 16MB right?

Running:
```python
import esp
esp.flash_size()
```
Returns:
```bash
16777216
```

I get **16777216**, which is the correct 16MB... I would like to write files to the filesystem (which SHOULD have 16MB of space) but I only can write max 2MB to the filesystem. Any ideas to resolve this flash memory issue? Or what am I doing wrong?

chegewara, 2019-07-03:
Is it correct that you are using twice the same command with `erase_flash`, which you pasted 2 times?
```
esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 460800 --before default_reset --after hard_reset erase_flash
```
Second time should be only `flash` not `erase_flash`.

LtMerlin, 2019-07-04:
Oops pasted twice indeed. Mistake while posting this. The second command I perform is:

esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 460800 write_flash -z --flash_freq 40m --flash_mode dio --flash_size=16MB 0x1000 esp32-20190529-v1.11.bin

LtMerlin, 2019-07-04:
> Is it correct that you are using twice the same command with `erase_flash`, which you pasted 2 times?
> 
> ```
> esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 460800 --before default_reset --after hard_reset erase_flash
> ```
> 
> Second time should be only `flash` not `erase_flash`.

Oops pasted it twice indeed. Mistake while posting this. The second command I perform is:

`esptool.py --chip esp32 --port /dev/tty.SLAB_USBtoUART --baud 460800 write_flash -z --flash_freq 40m --flash_mode dio --flash_size=16MB 0x1000 esp32-20190529-v1.11.bin`

I updated the original post with this.

chegewara, 2019-07-04:
Hi,
then its typo, again, or error. You cant flash app binary in address 0x1000, its reserved for bootloader.
Im guessing it should be 0x10000, but to confirm that we need your partition table csv.

LtMerlin, 2019-07-04:
> Hi,
> then its typo, again, or error. You cant flash app binary in address 0x1000, its reserved for bootloader.
> Im guessing it should be 0x10000, but to confirm that we need your partition table csv.

@chegewara I am new to the ESP32 so I followed the guide at [https://docs.micropython.org/en/latest/esp32/tutorial/intro.html#getting-the-firmware](https://docs.micropython.org/en/latest/esp32/tutorial/intro.html#getting-the-firmware) to get started and there is no partition table mentioned. Can you point me into the right direction to get Micropython on this board and where to start on writing the right partition table csv?

chegewara, 2019-07-04:
Ok, then all is ok. Micropython is running:
```
MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32
Type "help()" for more information.
>>>
```

LtMerlin, 2019-07-04:
> Ok, then all is ok. Micropython is running:
> 
> ```
> MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32
> Type "help()" for more information.
> >>>
> ```

Yes Micropython is running, but is says:
```
flash read err, 1000
ets_main.c 371 
```
And the Micropython filesystem only has 2MB available, while on this board there should be 16MB...

costaud, 2019-07-18:
There is an passage in ECO documentation about this.

"3.8 Due to the flash start-up time, a spurious watchdog reset occurs when ESP32 is powered up or wakes up from Deepsleep. "

"If the ESP32 reads from the flash chip before it is ready, invalid data can cause booting to
fail until a Watchdog Timer reset occurs. This can occur on power-on and on wake from
Deep-sleep, if the ESP32 VDD_SDIO is used to power the flash chip. "

This would only happen on the first time the module is powered, due to a slower init time of the flash chip. However, it would affect your firmware after it is running.

costaud, 2020-05-29:
I'll close this issue for now, feel free to reopen if you have further questions.

closed 2020-05-29 by costaud