espressif/esp-idf #3736: esp_http_client_open() doesn't send POST Content

paloky, 2019-07-04:
I try to send a HTTP post data using the examples of "examples/esp_http_client.c", but doesn't work.
The "esp_http_client_open", only send the Header of HTML, but not the Content.
I using this code.

```c
        char *buffer = malloc(MAX_HTTP_RECV_BUFFER + 1);
	if (buffer == NULL)	{
		ESP_LOGE(TAG, "Cannot malloc http receive buffer");
		return;
	}	

       esp_http_client_config_t config = {
            .host = "httpbin.org",
            .path = "/post",
            .transport_type = HTTP_TRANSPORT_OVER_TCP,
             .event_handler = _http_event_handler,
	};

      esp_http_client_handle_t client = esp_http_client_init(&config);

    // POST DEMO2
    const char *post_data = "{\"field1\":\"value1\",\"field2\":\"value2\"}";
    esp_http_client_set_method(client, HTTP_METHOD_POST);
    esp_http_client_set_post_field(client, post_data, strlen(post_data));
    esp_http_client_set_header(client, "Content-Type", "application/json");

     esp_err_t err;
     if ((err = esp_http_client_open(client, 0)) != ESP_OK)  	{
        ESP_LOGE(TAG, "Failed to open HTTP connection: %s", esp_err_to_name(err));
        free(buffer);
        return;
	}	

       int content_length =  esp_http_client_fetch_headers(client);   
	int total_read_len = 0, read_len;
	if (total_read_len < content_length && content_length <= MAX_HTTP_RECV_BUFFER) {
		read_len = esp_http_client_read(client, buffer, content_length);
		if (read_len <= 0){
			ESP_LOGE(TAG, "Error read data");
		}
		buffer[read_len] = 0;
		ESP_LOGD(TAG, "read_len = %d", read_len);
	}
		
	ESP_LOGI(TAG, "HTTP Stream reader Status = %d, content_length = %d",  esp_http_client_get_status_code(client), esp_http_client_get_content_length(client));	
		
	printf("BUFFER: \r\n %s", buffer);

    esp_http_client_close(client);
    esp_http_client_cleanup(client);
    free(buffer);
```

I try putting the Content Size in the "esp_http_client_open(client, strlen(post_data)))" function, but doesn't work too.

The response result of this code was:
```
 {
  "args": {},
  "data": "",
  "files": {},
  "form": {},
  "headers": {
    "Content-Length": "0",
    "Content-Type": "application/json",
    "Host": "httpbin.org",
    "User-Agent": "ESP32 HTTP Client/1.0"
  },
  "json": null,
  "origin": "46.27.106.247, 46.27.106.247",
  "url": "https://httpbin.org/post"
}
```

Using https://apitester.com/, to test the POST, the Rigth response include the POST Content that I send in "form":
```
{
  "args": {}, 
  "data": "", 
  "files": {}, 
  "form": {
    "{\\\"field1\\\":\\\"value1\\\",\\\"field2\\\":\\\"value2\\\"}": ""
  }, 
  "headers": {
    "Accept": "*/*", 
    "Content-Length": "45", 
    "Content-Type": "application/x-www-form-urlencoded", 
    "Host": "www.httpbin.org", 
    "User-Agent": "Mozilla/5.0 (compatible; Rigor/1.0.0; http://rigor.com)"
  }, 
  "json": null, 
  "origin": "50.16.249.180, 50.16.249.180", 
  "url": "https://www.httpbin.org/post"
}
```

ganeshlandge, 2019-07-26:
@paloky esp_http_client_open() is not sending POST content because you are passing 0 as second argument. Instead try strlen(post_data) i.e. length of POST content as second argument to esp_http_client_open(), it will work.

paloky, 2019-07-30:
Hi ganeshlandge.
If you read all my issue, in the midle I say:

> I try putting the Content Size in the "esp_http_client_open(client, strlen(post_data)))" function, but doesn't work too.

Using strlen(post_data) doesnt works... neither.
Regards.

ganeshlandge, 2019-08-06:
Hey @paloky,
You are right, actually only adding strlen(post_data) won't work. The esp_http_client_open() API does not call the esp_http_client_send_post_data(esp_http_client_handle_t client) API, so the post data is not sent to the server.
In addition to **strlen(post_data)** do the following changes in **esp_http_client_open(esp_http_client_handle_t client, int write_len)** API 
~~~
--- a/components/esp_http_client/esp_http_client.c
+++ b/components/esp_http_client/esp_http_client.c
@@ -1135,6 +1135,9 @@ esp_err_t esp_http_client_open(esp_http_client_handle_t client, int write_len)
     if ((err = esp_http_client_request_send(client, write_len)) != ESP_OK) {
         return err;
     }
+    if ((err = esp_http_client_send_post_data(client)) != ESP_OK) {
+        return err;
+    }
     return ESP_OK;
 }
~~~
Above is result for git diff. Add this changes and let me know if it works.

closed 2019-12-27 by Alvin1Zhang