espressif/esp-idf #3737: export.sh fails for custom tools path

jamieparkinson, 2019-07-04:
## Problem Description
Using the new install scripts with IDF_TOOLS_PATH to install the toolchain to a custom location, sourcing `export.sh` fails with errors of the type `tool x has no installed versions`. This is due to `idf_tools.py` being run in a subshell which doesn't have access to the parent envvars.

### Expected Behavior
Setting the same `IDF_TOOLS_PATH` for `install.sh` and `export.sh` results in a successful installation, with the toolchain in the specified directory.

### Actual Behavior
`install.sh` runs successfully but `export.sh` fails with errors from `idf_tools.py` claiming that the tools are not installed.

igrr, 2019-07-08:
@jamieparkinson Just to clarify one thing, the documentation asks to set an *environment variable* IDF_TOOLS_PATH here: https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/index.html#customizing-the-tools-installation-path.

Did you set it as `export IDF_TOOLS_PATH=/path` (i.e. an environment variable) or just `IDF_TOOLS_PATH=/path` (i.e. a shell variable) before running `export.sh`?

jamieparkinson, 2019-07-08:
Ah, you're right - thank you! Perhaps this distinction could be made explicit in the docs to avoid this mistake (I'm happy to PR that if you'd like)

closed 2019-07-08 by jamieparkinson

igrr, 2019-07-08:
Thanks for the confirmation! I am preparing some fixes for these tools and will update the getting started guide to explicitly mention `export`. Not to discourage you, future PRs are of course welcome :)