espressif/esp-idf #3740: install.bat & export.bat fail

axbusch, 2019-07-04:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version  v4.0-dev-1136-g28f1cdf5e
- Build System:         [CMake]
- Compiler version 
C:\Users\funky\esp\tools>idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp32-2019r1-8.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51.20170517 (recommended, installed)
* cmake: CMake build system
  - 3.13.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20190313 (recommended, installed)
* mconf: menuconfig tool
  - v4.6.0.0-idf-20190628 (recommended, installed)
* ninja: Ninja build system
  - 1.9.0 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
WARNING: tool idf-exe version 1.0 is installed, but the tool failed to run
  - 1.0 (recommended)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
- Operating System:     [Windows 7]
- Power Supply:         [USB]

## Problem Description
idf.py.exe does not run wenn called by install.bat
manually i can execute the program

### Expected Behavior

### Actual Behavior
C:\Users\funky\esp>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@esp32-2019r1-8.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
WARNING: tool idf-exe version 1.0 is installed, but the tool failed to run
Installing idf-exe@1.0
file idf-exe-v1.0.zip is already downloaded
WARNING: destination path already exists, removing
Traceback (most recent call last):
  File "C:\Users\funky\esp\tools\idf_tools.py", line 1249, in <module>
    main(sys.argv[1:])
  File "C:\Users\funky\esp\tools\idf_tools.py", line 1245, in main
    action_func(args)
  File "C:\Users\funky\esp\tools\idf_tools.py", line 1048, in action_install
    tool_obj.install(tool_version)
  File "C:\Users\funky\esp\tools\idf_tools.py", line 550, in install
    shutil.rmtree(dest_dir)
  File "C:\python3\lib\shutil.py", line 513, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "C:\python3\lib\shutil.py", line 397, in _rmtree_unsafe
    onerror(os.unlink, fullname, sys.exc_info())
  File "C:\python3\lib\shutil.py", line 395, in _rmtree_unsafe
    os.unlink(fullname)
PermissionError: [WinError 5] Zugriff verweigert: 'C:\\Users\\funky\\.espressif\\tools\\idf-exe\\1.0\\idf.py.exe'
Setting up Python environment
Installing Python packages from C:\Users\funky\esp\requirements.txt
Requirement already satisfied: setuptools in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from -r C:\Users\funky\esp\requirements.txt (line 4)) (41.0.1)
Requirement already satisfied: click>=5.0 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from -r C:\Users\funky\esp\requirements.txt (line 8)) (7.0)
Requirement already satisfied: pyserial>=3.0 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from -r C:\Users\funky\esp\requirements.txt (line 9)) (3.4)
Requirement already satisfied: future>=0.15.2 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from -r C:\Users\funky\esp\requirements.txt (line 10)) (0.17.1)
Requirement already satisfied: cryptography>=2.1.4 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from -r C:\Users\funky\esp\requirements.txt (line 11)) (2.7)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from -r C:\Users\funky\esp\requirements.txt (line 12)) (2.3.1)
Requirement already satisfied: pyelftools>=0.22 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from -r C:\Users\funky\esp\requirements.txt (line 13)) (0.25)
Requirement already satisfied: asn1crypto>=0.21.0 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\funky\esp\requirements.txt (line 11)) (0.24.0)
Requirement already satisfied: cffi!=1.11.3,>=1.8 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\funky\esp\requirements.txt (line 11)) (1.12.3)
Requirement already satisfied: six>=1.4.1 in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from cryptography>=2.1.4->-r C:\Users\funky\esp\requirements.txt (line 11)) (1.12.0)
Requirement already satisfied: pycparser in c:\users\funky\.espressif\python_env\idf4.0_py3.7_env\lib\site-packages (from cffi!=1.11.3,>=1.8->cryptography>=2.1.4->-r C:\Users\funky\esp\requirements.txt (line 11)) (2.19)
All done! You can now run:
   export.bat

C:\Users\funky\esp>export.bat
Setting IDF_PATH: C:\Users\funky\esp

Adding ESP-IDF tools to PATH...
WARNING: tool idf-exe version 1.0 is installed, but the tool failed to run
ERROR: tool idf-exe has no installed versions. Please run 'C:\Users\funky\esp\install.bat' to install it.

hcarlott, 2019-07-06:
I'm seeing the same thing.
Same environment / Windows 7
Seems related to idf.py.exe sending version  (--version) to Stdout.

igrr, 2019-07-09:
Thank you for the reports. We have reproduced the issue and the fix is coming soon.

igrr, 2019-07-09:
In the meantime, you can download the fixed version from https://dl.espressif.com/dl/idf-exe-v1.0-fix.zip, extract it, and replace C:\Users\<username>\.espressif\tools\idf-exe\1.0\idf.py.exe. After this, export.bat and the "ESP-IDF Command Prompt" shortcut should work.

Yuvaraj4041, 2019-07-11:
Hi @igrr 

i downloaded and installed esp-idf-tools-setup-2.0.exe.
i am facing same problem axbusch reported.
Then i checked .espressif folder, there is no folder named tools. So i dont know where to replace idf.py exe file. Can you help me.

closed 2019-07-15 by igrr

mazorvanunu, 2020-09-13:
haha lol those errors... did install.bat then export.bat showed everything ok... still no idf.py in path, people in espressif just code smth and print it out, no testing no nothing... whats the point???

wachidsusilo, 2024-04-10:
did this but The PATH still not added to the environment variables. any idea how to solve this?

```
PS C:\Espressif\esp-idf> ./install.ps1
Installing ESP-IDF tools
Updating C:\Users\wachid\.espressif\idf-env.json
Selected targets are: esp32, esp32s2, esp32s3, esp32c6, esp32c61, esp32c3, esp32h2, esp32c2, esp32p4, esp32c5
Current system platform: win64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, cmake, openocd-esp32, ninja, idf-exe, ccache, dfu-util, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@12.1_20231023 (already installed)
Skipping riscv32-esp-elf-gdb@12.1_20231023 (already installed)
Skipping xtensa-esp-elf@esp-13.2.0_20240305 (already installed)
Skipping riscv32-esp-elf@esp-13.2.0_20240305 (already installed)
Skipping esp32ulp-elf@2.38_20240113 (already installed)
Skipping cmake@3.24.0 (already installed)
Skipping openocd-esp32@v0.12.0-esp32-20240318 (already installed)
Skipping ninja@1.11.1 (already installed)
Skipping idf-exe@1.0.3 (already installed)
Skipping ccache@4.8 (already installed)
Skipping dfu-util@0.11 (already installed)
Skipping esp-rom-elfs@20240305 (already installed)
Setting up Python environment
Python 3.11.5
pip 24.0 from C:\Users\wachid\.espressif\python_env\idf5.3_py3.11_env\Lib\site-packages\pip (python 3.11)
Upgrading pip and setuptools...
Requirement already satisfied: pip in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (24.0)
Requirement already satisfied: setuptools in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (69.2.0)
Skipping the download of C:\Users\wachid\.espressif\espidf.constraints.v5.3.txt because it was downloaded recently.
Installing Python packages
 Constraint file: C:\Users\wachid\.espressif\espidf.constraints.v5.3.txt
 Requirement files:
  - C:\Espressif\esp-idf\tools\requirements\requirements.core.txt
Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Ignoring importlib_metadata: markers 'python_version < "3.8"' don't match your environment
Requirement already satisfied: setuptools in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 3)) (69.2.0)
Requirement already satisfied: packaging in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 4)) (24.0)
Requirement already satisfied: click in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 7)) (8.1.7)
Requirement already satisfied: pyserial in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 8)) (3.5)
Requirement already satisfied: cryptography in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 9)) (42.0.5)
Requirement already satisfied: pyparsing in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 10)) (3.1.2)
Requirement already satisfied: pyelftools in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 11)) (0.31)
Requirement already satisfied: idf-component-manager in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 12)) (1.5.2)
Requirement already satisfied: esp-coredump in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 13)) (1.10.0)
Requirement already satisfied: esptool in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 14)) (4.8.dev3)
Requirement already satisfied: esp-idf-kconfig in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 15)) (2.2.0)
Requirement already satisfied: esp-idf-monitor in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 16)) (1.4.0)
Requirement already satisfied: esp-idf-nvs-partition-gen in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 17)) (0.1.2)
Requirement already satisfied: esp-idf-size in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 18)) (1.2.0)
Requirement already satisfied: esp-idf-panic-decoder in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 19)) (1.0.1)
Requirement already satisfied: pyclang in c:\users\wachid\.espressif\python_env\idf5.3_py3.11_env\lib\site-packages (from -r C:\Espressif\esp-idf\tools\requirements\requirements.core.txt (line 20)) (0.4.2)
[... cut ...]
C:\Users\wachid\.espressif\python_env\idf5.3_py3.11_env\Scripts
Checking if Python packages are up to date...
Constraint file: C:\Users\wachid\.espressif\espidf.constraints.v5.3.txt
Requirement files:
 - C:\Espressif\esp-idf\tools\requirements\requirements.core.txt
Python being checked: C:\Users\wachid\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

```