espressif/esp-idf #3743: Provisioning over advertising channel

jkockx, 2019-07-05:
I'm using the example project esp-idf/examples/bluetooth/esp_ble_mesh/ble_mesh_provisioner/:
I didn't succeed in provisioning over PB-ADV only by changing the initiailization parameters. Backtracing my problem, I started thinking this is even not possible at all. Though, as a certified ble mesh api such functionality will probably be possible. I've probably missed something trying to understand the api as I'm quite new to embedded progranmming. Could someone help me on my way?

WCCWCC, 2019-07-16:
1. I need to know how you actually operate.It is best to provide some log information. 

2. My understanding is that you only open the PA-ADV channel for communication. I will reply to you after I verify.

Alvin1Zhang, 2019-07-19:
@jkockx Thanks for reporting the issue, would you help share any logs for the 1st question by @WCCWCC ? Thanks.

jkockx, 2019-07-21:
Thanks for your reply. I succeeded in provisioning over PB-ADV by changing some parameters in the menuconfig as described in https://www.esp32.com/viewtopic.php?t=11332

Alvin1Zhang, 2019-10-11:
@jkockx Glad that you succeeded in provisioning over PB-ADV. Feel free to reopen if you still have the issue, thanks.

closed 2019-10-11 by Alvin1Zhang