espressif/esp-idf #3744: The path to СMake was not automatically added to the environment variables.

ArTyGill, 2019-07-05:
When installing esp-idf-tools-setup-2.0.exe, the path to it is not added to the environment in the ".espressif" folder.

I had to add to the manual.
Same problem with IDF_PATH.

igrr, 2019-07-05:
Hi @ArTyGill, can you please post the output you get when running `export.bat` from esp-idf directory in a new cmd.exe window? And also show what `echo %PATH%` prints afterwards. Thanks.

ArTyGill, 2019-07-05:
1-st terminal

```
C:\Docs\espressif\esp-idf>export.bat
Setting IDF_PATH: C:\Docs\espressif\esp-idf

Adding ESP-IDF tools to PATH...
    C:\Docs\espressif\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin
    C:\Docs\espressif\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\Docs\espressif\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\Docs\espressif\.espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\Docs\espressif\.espressif\tools\ninja\1.9.0\
    C:\Docs\espressif\.espressif\tools\idf-exe\1.0\
    C:\Docs\espressif\.espressif\tools\ccache\3.7\
    C:\Docs\espressif\.espressif\python_env\idf4.0_py3.7_env\Scripts

Checking if Python packages are up to date...
Python requirements from C:\Docs\espressif\esp-idf\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Docs\espressif\esp-idf>@echo %PATH%
C:\Docs\espressif\.espressif\tools\xtensa-esp32-elf\esp32-2019r1-8.2.0\xtensa-esp32-elf\bin;C:\Docs\espressif\.espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin;C:\Docs\espressif\.espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin;C:\Docs\espressif\.espressif\tools\mconf\v4.6.0.0-idf-20190628\;C:\Docs\espressif\.espressif\tools\ninja\1.9.0\;C:\Docs\espressif\.espressif\tools\idf-exe\1.0\;C:\Docs\espressif\.espressif\tools\ccache\3.7\;C:\Docs\espressif\.espressif\python_env\idf4.0_py3.7_env\Scripts;C:\Python37\Scripts\;C:\Python37\;C:\Program Files (x86)\PC Connectivity Solution\;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files (x86)\AMD\ATI.ACE\Core-Static;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\PuTTY\;C:\Program Files (x86)\GitExtensions\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\Intel\Platform Flash Tool Lite;C:\Docs\espressif\.espressif\tools\cmake\3.13.4\bin;C:\Docs\espressif\esp-idf\tools;C:\Program Files\Git\cmd;C:\Docs\espressif\.espressif\tools\cmake\3.13.4\;C:\Cadence\SPB_16.6\tools\bin;C:\Cadence\SPB_16.6\tools\specctra\bin;C:\Program Files\Microsoft VS Code\bin;C:\Users\Артем\AppData\Local\Microsoft\WindowsApps;C:\Users\Артем\AppData\Local\Programs\Microsoft VS Code\bin;
```
2-nd Terminal

```
C:\Users\Артем>@echo %PATH%
C:\Python37\Scripts\;C:\Python37\;C:\Program Files (x86)\PC Connectivity Solution\;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\ATI Technologies\ATI.ACE\Core-Static;C:\Program Files (x86)\Windows Live\Shared;C:\Program Files\Microsoft SQL Server\110\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files (x86)\AMD\ATI.ACE\Core-Static;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\PuTTY\;C:\Program Files (x86)\GitExtensions\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files (x86)\Intel\Platform Flash Tool Lite;C:\Docs\espressif\.espressif\tools\cmake\3.13.4\bin;C:\Docs\espressif\esp-idf\tools;C:\Program Files\Git\cmd;C:\Docs\espressif\.espressif\tools\cmake\3.13.4\;C:\Cadence\SPB_16.6\tools\bin;C:\Cadence\SPB_16.6\tools\specctra\bin;C:\Program Files\Microsoft VS Code\bin;C:\Users\Артем\AppData\Local\Microsoft\WindowsApps;C:\Users\Артем\AppData\Local\Programs\Microsoft VS Code\bin;
```

igrr, 2019-07-05:
That looks correct! Do you observe any issues running `idf.py build` in the Shell where you have executed `export.bat`?

Note that `export.bat` and IDF Tools Installer v2 do not set any persistent environment variables, aside from `IDF_TOOLS_PATH`. This is intentional and allows for having multiple IDF versions installed side by side, each having correct tools versions by means of export.bat.

IDF Tools Installer also creates an "ESP-IDF Command Prompt (cmd.exe)" shortcut in the Start menu, which opens cmd.exe and runs `export.bat`.

ArTyGill, 2019-07-05:
Well, I figured out how to make it work with VS Code?

ArTyGill, 2019-07-05:
@igrr 
I will try to add the export.bat launch in the VS Code terminal call.

igrr, 2019-07-05:
For VS Code, currently the easiest option is to start it from cmd.exe where paths have already been set up. Or, you can create a 2-line batch file: the first line executes export.bat, the 2nd starts VS Code. Then start VS code through this batch file.

We are also planning to release an extension for VS Code which will be aware of IDF_TOOLS_PATH installation method. Please follow esp32.com for announcements.

igrr, 2019-07-05:
For those who find this issue through search, here are the sections of the getting started guide which explain export.bat and the "ESP-IDF command prompt":

https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/index.html#step-4-set-up-the-environment-variables

https://docs.espressif.com/projects/esp-idf/en/latest/get-started-cmake/windows-setup.html#using-the-command-prompt

closed 2019-07-05 by ArTyGill