espressif/esp-idf #3746: Disable EFUSE option causing compiler error

KaeLL, 2019-07-05:
If you enable `CONFIG_ESP32_DISABLE_BASIC_ROM_CONSOLE`, on the current state of the master branch, you get
```
/home/nvmd/Documents/VSCode/esp-idf/components/esp32/cpu_start.c: In function 'start_cpu0_default':
/home/nvmd/Documents/VSCode/esp-idf/components/esp32/cpu_start.c:361:5: error: implicit declaration of function 'esp_efuse_disable_basic_rom_console' [-Werror=implicit-function-declaration]
     esp_efuse_disable_basic_rom_console();
```

One `#include` away from paradise :P

KonstantinKondrashov, 2019-07-08:
Hi @KaeLL, thanks for your report, it was fixed recently by [commit 09ee721](https://github.com/espressif/esp-idf/commit/09ee7219322c127aaa102a4809d8f29e9bf67cad)

Alvin1Zhang, 2019-07-08:
@KaeLL Thanks for reporting the issue. Feel free to reopen the ticket if the issue still exists. Thanks.

closed 2019-07-08 by Alvin1Zhang